gotthardp/lorawan-server #1: I need to build your Server Code and want to use

mruthk, 2016-08-23:
Dear Lorawan Server,

Could you please let me know how I can build the lorawan-server code and use for my testing of the mote..?

Could you please help on this.?

Regards,
Mruthyunjaya(MJ)

gotthardp, 2016-08-23:
Hi Mruthyunjaya,
Thank you for your interest in the lorawan-server. I'm glad to help you. May I ask you to follow the [Installation Instructions](https://github.com/gotthardp/lorawan-server#installation) and let me know what needs to be better described or at what step did you experience troubles?

Regards,
Petr

gotthardp, 2016-08-23:
Hello again. I created a binary release and improved the [README.md](https://github.com/gotthardp/lorawan-server/blob/master/README.md), so everything may be a bit easier now.

mruthk, 2016-08-25:
Hi Petr,

Thanks for your prompt reply for my query.

Could you please let me know the compiling(build) instructions if I have to compile and generate a Binary. 

As somehow there is some problem in accessing your binary because of the network policy on my network.

Thanks and regards,
Mruthyunjaya(MJ)

gotthardp, 2016-08-25:
What OS do you have? Windows, Linux or something else?

I wrote some instructions [here](https://github.com/gotthardp/lorawan-server#build-instructions). Does it work?

glederer007, 2016-09-08:
Hello Petr,

I would like to try out your lorawan-server, but I have Ubuntu 16.4. Could you give some instructions on how to install the binaries on Ubuntu (amd64)? Should I compile and build the binaries for my machine?

Thanks,

Gabor

gotthardp, 2016-09-08:
Hello Gabor,
the Linux related instructions should work with any Ubuntu. You even don't need to build it and use the pre-build binaries. Please let me know if it works.

Petr

glederer007, 2016-09-08:
Hello Petr,

I installed erlang with "apt-get install erlang".

After running I am getting the following error:

"bin/lorawan-server: line 47: /usr/lib/erlang/erts-7.2/bin/erl: No such file or directory"

It seems a version issue.

I have tried to modify the "7.2" entry in the lorawan-server file, but then I get the following error:

Exec: /usr/lib/erlang/erts-7.3/bin/erlexec -noshell -noinput +Bd -boot /home/xx/lorawan-server/releases/0.1.0/start -mode embedded -config /home/xx/lorawan-server/releases/0.1.0/sys.config -boot_var ERTS_LIB_DIR /usr/lib/erlang/erts-7.3/../lib -args_file /home/xx/lorawan-server/releases/0.1.0/vm.args -- foreground
Root: /home/xx/lorawan-server
{"init terminating in do_boot",{'cannot load',error_handler,get_files}}

Note: xx is my home directory.

Regards,

Gabor

gotthardp, 2016-09-08:
Oops. The binary release is not portable. I am sorry for that. I will fix that soon.

Would you mind trying to build the server on your own? It should not be that difficult.

glederer007, 2016-09-08:
Hello Petr,

Of course. I will try it on the weekend.

Regards,

Gabor

glederer007, 2016-09-09:
Hello Petr,

I was able to build the application, it starts. I didn't have the time to try it out with my LORANK8 gateway, but I plan to do it on the weekend. However there is already a change I suggest to do: when I type 192.168.0.21:8080 in the browser (that is the IP address assigned to my virtual Linux machine), the application web interface doesn't start. If I type 192.168.0.21:8080/admin it starts. Perhaps it would be better to let the program jump to /admin if the x.x.x.x/8080 address or x.x.x.x is typed in. Additionally I would suggest to put in the comments that the package needs to be started with sudo in order to run correctly. Notwithstanding, great work, congratulations!

Regards,

Gabor

PS.: I will let you know my experiences with running the server together with my LORANK8 and Microchip's DM164138 LoraWAN Mote.

gotthardp, 2016-09-09:
Hello Gabor,
thank you very much for trying the server and providing me with a feedback. If anything is not as you'd expect, please let me know. I will try to address your comments as fast as possible.

Kind Regards,
Petr

glederer007, 2016-09-09:
Hello Petr,

Probably I need some more instructions on how to proceed.

1) I started the program.
2) I added my gateway MAC: it is in the web interface (pop up shows "element succesfully..."), but on the linux terminal is writes "15:39:37.007 [error] ERROR: {unknown_mac,<<104,201,11,255,255,236,224,134>>}
" (the packet forwarder is already running at this point and sending UDP data to the lorawan-server, maybe that is the problem).
3)  I added my Motes" DevEUI: it is in the web interface (pop up shows "element succesfully...")
4) I switched on my Mote and pushed OTAA, linux terminal writes: "15:43:15.967 [info] JOIN REQUEST <<112,179,213,126,208,0,11,69>> <<0,4,163,11,0,27,69,48>> -> <<6,34,183,140>>"
5) I send an UPLINK unconfirmed message and i get some errors:

"15:44:07.529 [error] gen_server lorawan_iface_forwarder terminated with reason: no function clause matching lorawan_application_mote:handle(<<6,34,183,140>>, <<"loramote">>, undefined, 146, <<"00826027">>) line 18
15:44:07.530 [error] CRASH REPORT Process <0.577.0> with 0 neighbours exited with reason: no function clause matching lorawan_application_mote:handle(<<6,34,183,140>>, <<"loramote">>, undefined, 146, <<"00826027">>) line 18 in gen_server:terminate/7 line 826
15:44:07.530 [error] Supervisor lorawan_sup had child packet_forwarder started with lorawan_iface_forwarder:start_link(1680) at <0.577.0> exit with reason no function clause matching lorawan_application_mote:handle(<<6,34,183,140>>, <<"loramote">>, undefined, 146, <<"00826027">>) line 18 in context child_terminated
15:44:07.532 [debug] Supervisor lorawan_sup started lorawan_iface_forwarder:start_link(1680) at pid <0.600.0>"

Of course application "loramote" is not defined. But how do I define it?

Thanks,

Gabor

gotthardp, 2016-09-10:
Hi Gabor,
despite all the crashes and errors it worked quite well. There is a loramote application (see lorawan_application_mote.erl), which works with the mote I have. My mote sends data on LoRaWAN port "2" (see the "2" in lorawan_application_mote:handle, line 18), however your mote sends (some other?) data on port 146. Hence the crash.
I will modify the app to log unexpected data instead of crashing. But if your mote is something what others could also purchase, it would be much better to correctly parse the data sent just like I do for port 2.

glederer007, 2016-09-10:
Hello Petr,

I am using the following mote:
http://www.microchip.com/Developmenttools/ProductDetails.aspx?PartNO=dm164138

I have both the legacy (OLED) and the new (LCD) versions, but for your server application I am using the legacy (OLED) version.

The Mote is sending eight bytes payload (three digits temperature, five digits illumination) on a randomly selected port. The uplink period can be set between five and thirty minutes and each uplink is sent on a different port.

Regards,

Gabor

gotthardp, 2016-09-10:
I made some updates:
 * A step towards binary portability. The server is now started by `bin/lorawan-server` (no parameters anymore).
 * Implemented an application for Microchip's Mote. You need to set `microchip-mote` as the application.
 
Please try it and let me know how it works.

The new start script uses a different database name, so it will purge the data you entered. You may want to backup your database by `dbexport` before updating and do `dbimport` after updating.

gotthardp, 2016-09-10:
Hi again, Gabor,
concerning your other two remarks:
 * I implemented the "jump to /admin if the x.x.x.x/8080 address or x.x.x.x is typed in" you suggested.
 * Also, I confirm the updated lorawan-server does *not* require root permissions and can run *without* sudo.

Regards,
Petr

glederer007, 2016-09-15:
Hello Petr,

I compiled the code, it works fine. However I still don't understand how can I get out the uplink data from the application (how can I display it)? The same is true for the downlink part, how can I send data to the Mote? Is there an application included in your code for this?

Regards,

Gabor

gotthardp, 2016-09-15:
Hi Gabor,
how you want to process your data? Do you want to process them inside the server, or you want to send the data out to some other server (e.g. via Web Sockets)?

Right now, the data are processed inside the server. Have a look at `lorawan_application_microchip_mote.erl`, there the following function:

```erlang
handle(DevAddr, _App, _AppID, _Port, <<Light:5/binary, Temp:3/binary>>) ->
    lager:debug("PUSH_DATA ~s ~s",[DevAddr, Light, Temp]),
    % display actual time
    {H, M, S} = time(),
    Time = lists:flatten(io_lib:format('~2..0b:~2..0b:~2..0b', [H, M, S])),
    {send, 2, list_to_binary(Time)};
```

This is all the processing. It parses the input by matching `<<Light:5/binary, Temp:3/binary>>` and logs the values. Then it obtains current time, converts it to a string and then sends it via the port 2 back.

gotthardp, 2016-11-13:
In addition to the above I just implemented a WebSocket interface, which shall further simplify the data access. I assume the issue is solved, so I close this report now; if you need further support, feel free to open a new issue.

closed 2016-11-13 by gotthardp