gotthardp/lorawan-server #2: Docker based install

ubergesundheit, 2016-08-25:
Hi,

at the bottom of this issue, you will find a working Dockerfile to run this project in a docker container. There is one issue though. For now its not possible to change the values of the `sys.config`. It would be best, if the config values could be changed through environment variables.

The workarounds for this would be mounting a config file into the container.
Or include the config at image build time

```
FROM solarisbank/erlang:19.0

RUN apk --no-cache add nodejs curl git
RUN curl -o /usr/bin/rebar3 "https://s3.amazonaws.com/rebar3/rebar3" && chmod +x /usr/bin/rebar3
RUN git clone https://github.com/gotthardp/lorawan-server.git && cd lorawan-server && rebar3 release

WORKDIR /lorawan-server/_build/default/rel/lorawan-server

CMD bin/lorawan-server foreground
```

gotthardp, 2016-08-25:
Hi. This is cool. Thanks for that,

What scenario you have in mind? In particular:
 * Do you want to modify some or all the values?
 * Do you want to modify them during the Docker build or during execution?

ubergesundheit, 2016-08-25:
It would be good if all values in the config file could be configured at container startup.

I think there are multiple ways to do this:
- executable parameters of the `bin/lorawan-server foreground`
- environment variables which are read at startup

gotthardp, 2016-08-25:
The configuration is included in the config file. I struggle why someone would need to have a configurable config file? :)

You can use the `RELX_CONFIG_PATH` environment variable to override the entire config file. Thus, one could define various .config files and then use the `RELX_CONFIG_PATH` to say which one should be used in a given case.

You could even define a custom template end generate (by a simple substitution) the .config file just before the server is started.

Would some of this meet your needs?

gotthardp, 2016-08-25:
I think I may now better understand your need: you want the server to be configurable via Docker's environment variables, right? (Docker Cloud has a GUI for that.)

I will consider this during the next refactoring. Some of the .config parameters (like the username/password) need a better configuration anyway.

ubergesundheit, 2016-08-25:
Yeah you're right. One of the ideas of docker is to be able to distribute
the image an be able to put in own configuration at runtime

gotthardp, 2016-09-04:
Hi again. In commit https://github.com/gotthardp/lorawan-server/commit/300e5e1b6b104941e08641d8cc2aa5b0acf74449 I enhanced the web-based configuration. The  .config file is now used only to
* Configure ports that are exposed in the Dockerfile
* Configure default values, which can be updated via the web-interface later

You can now distribute the image, connect the exposed ports via standard Docker interfaces and change everything else via the REST-API or web-admin of the server. Does this meet your needs?

ubergesundheit, 2016-09-04:
Looks really nice! Will try this with my gateways tomorrow!

gotthardp, 2016-09-11:
Any success? May I close the issue now?

ubergesundheit, 2016-09-11:
I had only time to test the docker build. But everything looks good

closed 2016-09-11 by ubergesundheit