gotthardp/lorawan-server #6: something wrong about 'application' when try to add node

chunis, 2016-10-09:
Hi, Developers:
when try to "add new device" or "create new link", I've encountered the same problem when try to fill the form about "Application". From the config I can see below lines:
    {plugins, [
        {<<"semtech-mote">>, lorawan_application_semtech_mote},
        {<<"microchip-mote">>, lorawan_application_microchip_mote}]},
So I think I should be able to fill with 'semtech-mote', but it always complained with "Invalid form". And although the form contains a down arrow, it shows no items after I click it.

So it looks like the applications are not enabled. I've tried both the binary and source code, but just the same issue. Then what may be wrong? Thanks.

gotthardp, 2016-10-10:
Hi. This is caused by a bug in ng-admin. I reverted back to an older version of ng-admin, which works. Please pull the latest updates of the lorawan-server and try again.

chunis, 2016-10-15:
The issue is fixed. Thanks.

closed 2016-10-15 by chunis