gotthardp/lorawan-server #7: unknown_mac and registering the gateway

ymkins, 2016-10-31:
I run lorawan-server in docker (Thank you for the Dockerfile). And I got an issue on registering the gateway (Kerlink, PROD_FW=wirmaV2_wirgrid_v2.2, LORABOARD_TYPE="868-27dBm").

As I understand, the MAC field in the Web Admin "Create new gateway" form is the MAC address of my LoRaWAN Gateway.
The Web form filters the MAC field with: 
```ng-pattern="[A-Za-z0-9]{16}" placeholder="e.g. 0123456789ABCDEF".```
So it doesn't allow enter my value ```02:4B:08:0E:0E:29```.
I've tried a few converted values (https://www.vultr.com/tools/mac-converter/) in REST API:
```
curl -H 'Authorization: Basic YWRtaW46YWRtaW4=' -H 'Content-Type: application/json' -d '{"mac":"02:4B:08:0E:0E:29","netid":"000000","gpspos":{"lat":1,"lon":1},"gpsalt":1}' http://lorawan-server:8080/gateways
curl -H 'Authorization: Basic YWRtaW46YWRtaW4=' -H 'Content-Type: application/json' -d '{"mac":"02:4b:08:0e:0e:29","netid":"000000","gpspos":{"lat":1,"lon":1},"gpsalt":1}' http://lorawan-server:8080/gateways
curl -H 'Authorization: Basic YWRtaW46YWRtaW4=' -H 'Content-Type: application/json' -d '{"mac":"024B080E0E29","netid":"000000","gpspos":{"lat":1,"lon":1},"gpsalt":1}' http://lorawan-server:8080/gateways
curl -H 'Authorization: Basic YWRtaW46YWRtaW4=' -H 'Content-Type: application/json' -d '{"mac":"2521280941609","netid":"000000","gpspos":{"lat":1,"lon":1},"gpsalt":1}' http://lorawan-server:8080/gateways
curl -H 'Authorization: Basic YWRtaW46YWRtaW4=' -H 'Content-Type: application/json' -d '{"mac":"02:4b:08:00:00:0e:0e:29","netid":"000000","gpspos":{"lat":1,"lon":1},"gpsalt":1}' http://lorawan-server:8080/gateways
curl -H 'Authorization: Basic YWRtaW46YWRtaW4=' -H 'Content-Type: application/json' -d '{"mac":"02:4b:08:ff:ff:0e:0e:29","netid":"000000","gpspos":{"lat":1,"lon":1},"gpsalt":1}' http://lorawan-server:8080/gateways
curl -H 'Authorization: Basic YWRtaW46YWRtaW4=' -H 'Content-Type: application/json' -d '{"mac":"02:4b:08:ff:fe:0e:0e:29","netid":"000000","gpspos":{"lat":1,"lon":1},"gpsalt":1}' http://lorawan-server:8080/gateways
curl -H 'Authorization: Basic YWRtaW46YWRtaW4=' -H 'Content-Type: application/json' -d '{"mac":"00:4b:08:ff:fe:0e:0e:29","netid":"000000","gpspos":{"lat":1,"lon":1},"gpsalt":1}' http://lorawan-server:8080/gateways

curl -H 'Authorization: Basic YWRtaW46YWRtaW4=' http://lorawan-server:8080/gateways

[{"gpsalt":1,"gpspos":{"lat":1,"lon":1},"netid":"000000","mac":"0234B3083FF3FE30E30E32"},{"gpsalt":1,"gpspos":{"lat":1,"lon":1},"netid":"000000","mac":"252128094160"},{"gpsalt":1,"gpspos":{"lat":1,"lon":1},"netid":"000000","mac":"024B080E0E29"},{"gpsalt":1,"gpspos":{"lat":1,"lon":1},"netid":"000000","mac":"0234B30830E30E32"},{"gpsalt":1,"gpspos":{"lat":1,"lon":1},"netid":"000000","mac":"0034B3083FF3FE30E30E32"},{"gpsalt":1,"gpspos":{"lat":1,"lon":1},"netid":"000000","mac":"0234B3083FF3FF30E30E32"},{"gpsalt":1,"gpspos":{"lat":1,"lon":1},"netid":"000000","mac":"0234B30830030030E30E32"}]
```

And all time I got one log message:
```
11:55:43.298 [error] ERROR: {unknown_mac,<<0,0,2,75,8,14,14,41>>}
```
As I understand, the server receives packets from my GW station but doesn't handle it. 

So, please clarify, how to register the GW.

gotthardp, 2016-10-31:
The Web form expects a MAC address without the colons. Please note that your `02:4B:08:0E:0E:29` is only 6-bytes long, whereas the MAC address shall be 8-bytes long.

The error message suggests your gateway uses MAC `00:00:02:4B:08:0E:0E:29`. Remove the colons and enter this value to the Web form, or via the `curl` just like you did and it shall work.

ymkins, 2016-10-31:
Thanks!
And now I see 
```
16:06:06.352 [error] gen_server lorawan_iface_forwarder terminated with reason: no function clause matching lorawan_mac:handle_rxpk({rxq,<<"2016-10-31T16:06:09.566237Z">>,2772238820,2,1,1,-58,6.8}, 4, <<0,107,243,113>>, <<"semtech-mote">>, undefined, 4, <<0,0,0,0,0,0,0>>) line 151
16:06:06.352 [error] CRASH REPORT Process <0.954.0> with 0 neighbours exited with reason: no function clause matching lorawan_mac:handle_rxpk({rxq,<<"2016-10-31T16:06:09.566237Z">>,2772238820,2,1,1,-58,6.8}, 4, <<0,107,243,113>>, <<"semtech-mote">>, undefined, 4, <<0,0,0,0,0,0,0>>) line 151 in gen_server:terminate/7 line 812
16:06:06.357 [error] Supervisor lorawan_sup had child packet_forwarder started with lorawan_iface_forwarder:start_link(1680) at <0.954.0> exit with reason no function clause matching lorawan_mac:handle_rxpk({rxq,<<"2016-10-31T16:06:09.566237Z">>,2772238820,2,1,1,-58,6.8}, 4, <<0,107,243,113>>, <<"semtech-mote">>, undefined, 4, <<0,0,0,0,0,0,0>>) line 151 in context child_terminated
16:06:06.360 [debug] Supervisor lorawan_sup started lorawan_iface_forwarder:start_link(1680) at pid <0.955.0>
```
So, it's time to implement handlers.

closed 2016-10-31 by ymkins

gotthardp, 2016-10-31:
Yes, the error you are now getting suggests that your gateway is using some unexpected protocol.

kazi93, 2017-05-31:
@gotthardp I failed to add the gateways in the web form: When I click the submit button it simply does not work. In the Dashboard and debug console I have the below :
Last Occurred	Entity	Eid	Text
2017-05-31 20:06:16	gateway	B827EBFFFF5E0AAE	unknown_mac
2017-05-31 20:06:04	gateway	B827EBFFFFC53182	unknown_mac
2017-05-31 20:05:46	gateway	B827EBFFFF5E0AAE	{uplinks_lost,1}
2017-05-31 20:05:04	gateway	B827EBFFFFC53182	{uplinks_lost,1}

root@vps416657:~# systemctl status lorawan-server -l
● lorawan-server.service - LoRaWAN Server
   Loaded: loaded (/lib/systemd/system/lorawan-server.service; disabled)
   Active: active (running) since Tue 2017-05-30 14:40:57 CEST; 1 day 5h ago
  Process: 17585 ExecStop=/usr/lib/lorawan-server/bin/lorawanctl stop (code=exited, status=1/FAILURE)
  Process: 17602 ExecStart=/usr/lib/lorawan-server/bin/lorawan-daemon (code=exited, status=0/SUCCESS)
 Main PID: 17603 (su)
   CGroup: /system.slice/lorawan-server.service
           ‣ 17603 su lorawan -s /bin/sh -c /usr/lib/lorawan-server/bin/lorawan-server

May 31 20:16:04 vps416657.ovh.net lorawan-daemon[17602]: 20:16:04.379 [warning] gateway B827EBFFFFC53182 {uplinks_lost,1}
May 31 20:16:04 vps416657.ovh.net lorawan-daemon[17602]: 20:16:04.379 [error] gateway B827EBFFFFC53182 unknown_mac
May 31 20:16:16 vps416657.ovh.net lorawan-daemon[17602]: 20:16:16.391 [error] gateway B827EBFFFF5E0AAE unknown_mac
May 31 20:16:20 vps416657.ovh.net lorawan-daemon[17602]: 20:16:20.474 [error] gateway B827EBFFFF5E0AAE unknown_mac
May 31 20:16:34 vps416657.ovh.net lorawan-daemon[17602]: 20:16:34.375 [error] gateway B827EBFFFFC53182 unknown_mac
May 31 20:16:34 vps416657.ovh.net lorawan-daemon[17602]: 20:16:34.377 [error] gateway B827EBFFFFC53182 unknown_mac
May 31 20:16:46 vps416657.ovh.net lorawan-daemon[17602]: 20:16:46.367 [warning] gateway B827EBFFFF5E0AAE {uplinks_lost,1}
May 31 20:16:46 vps416657.ovh.net lorawan-daemon[17602]: 20:16:46.368 [error] gateway B827EBFFFF5E0AAE unknown_mac
May 31 20:17:04 vps416657.ovh.net lorawan-daemon[17602]: 20:17:04.375 [error] gateway B827EBFFFFC53182 unknown_mac
May 31 20:17:04 vps416657.ovh.net lorawan-daemon[17602]: 20:17:04.378 [error] gateway B827EBFFFFC53182 unknown_mac