gotthardp/lorawan-server #11: Store gateway GPS position

ymkins, 2016-11-15:
Hi Petr.
I run the lorawan-server in docker.
I've configured 3 gateways (Kerlink, PROD_FW=wirmaV2_wirgrid_v2.2, LORABOARD_TYPE="868-27dBm") to this host and saw log messages like:
```
11:26:33.966 [error] ERROR: {unknown_mac,<<0,0,2,75,8,14,16,2>>}
11:26:34.682 [error] ERROR: {unknown_mac,<<0,0,2,75,8,14,14,41>>}
11:26:36.850 [error] ERROR: {unknown_mac,<<0,0,2,75,8,14,14,30>>}
``` 

Then I've registered gateways (with default location) and after some time I got log messages like:
```
12:00:37.356 [error] ERROR: {unknown_devaddr,<<0,0,80,151>>}
12:00:37.368 [error] ERROR: {unknown_devaddr,<<0,0,80,151>>}
12:00:37.369 [error] ERROR: {unknown_devaddr,<<0,0,80,151>>}
```
It's expected because I didn't registered the end device.

But I see that only one gateway got actual location.
```
$ curl -H 'Authorization: Basic YWRtaW46YWRtaW4=' http://lorawan-server/gateways
[{"gpsalt":_hide_,"gpspos":{"lat":_hide_,"lon":_hide_},"netid":"010203","mac":"0000024B080E0E29"},{"gpspos":{"lat":48.88,"lon":14.12},"netid":"010203","mac":"0000024B080E0E1E"},{"gpspos":{"lat":48.88,"lon":14.12},"netid":"010203","mac":"0000024B080E1002"}]
```

I expected that all 3 gateways obtained actual location.

gotthardp, 2016-11-15:
It should behave like you expect. Please check (somewhere in the gateway logs) if the other two gateways provide non-zero numeric values for all 3 components (lat, lon, alt).

ymkins, 2016-11-16:
I got actual location on the 2nd gateway a few hours later, will check packets.

ymkins, 2016-11-17:
Finally I've found that issue was caused by placement of devices and the availability of the GPS signal.
So, I close it. Thanks for clarification.

closed 2016-11-17 by ymkins