gotthardp/lorawan-server #12: the freq 868xxxMhz,if i want to change the freq ,How to do ??

lancezhangsf, 2016-11-17:
Hi gotthardp,

I use 433Mhz, some errors happens ,   i look the code ,i find that this is 868Mzh,   IF  I want to use 433Mh, how to do ??????

gotthardp, 2016-11-17:
Hello, I made some enhancements that *should* enable you to use the server for 433MHz, but I have no means to test it. Please get the latest sources and check the `rx2_rf` field in server config. It may be sufficient to comment the 869 line and uncomment the 434.

If this doesn't work, some more adjustments may be needed to the RF parameters. In such case, please tell me what error are you getting.

gotthardp, 2016-12-23:
In commit 890b04e I added some more documentation on how to enable the 433 band. Please let me know if anything is unclear.

closed 2016-12-23 by gotthardp