gotthardp/lorawan-server #15: error when trying to edit fcnt

mzemb, 2016-12-26:
Hi.

I'm experimenting with your lorawan-server (my complete setup includes two loramotes and one kerlink gateway).

I'm using a build from your latest commit (205c666b - 21dec2016).

I have encountered an issue when I try to edit the FCnt fields in the Links tabs from the http admin interface. (I understand that it should not be needed, but unfortunately the behavior of the loramotes seems to be to restart fcnt at 0 on reset).

Anyway, when I click on the "save changes" button, I get the following error:

> [error] CRASH REPORT Process <0.16296.0> with 0 neighbours exited with reason: bad argument in call to erlang:binary_to_integer(undefined, 2) in lorawan_admin:parse_adr/1 line 119
> [error] Lager event handler error_logger_lager_h exited with reason {'EXIT',{{case_clause,[http,<0.16284.0>,9,<0.16296.0>,badarg,[{erlang,binary_to_integer,[undefined,2],[]},{lorawan_admin,parse_adr,1,[{file,"/home/getraline/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_admin.erl"},{line,119}]},{lorawan_admin,'-parse_admin/1-fun-0-',1,[{file,"/home/getraline/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_admin.erl"},{line,83}]},{lists,map,2,[{file,"lists.erl"},{line,1238}]},{lists,map,2,[{file,"lists.erl"},{line,...}]},...]]},...}}

gotthardp, 2016-12-26:
Hi. Sorry for that. I thought I fixed that. Please check the latest commit.

mzemb, 2016-12-27:
Thank you for your quick answer.
Unfortunately, your commit 3169b5c2 doesn't totally fix the issue: I'm now able to change the Fcnt once, but if I try it a second time, I get the same error:

> [error] CRASH REPORT Process <0.342.0> with 0 neighbours exited with reason: bad argument in call to erlang:binary_to_integer(<<"undefined">>, 2) in lorawan_admin:parse_adr/1 line 121
> [error] Lager event handler error_logger_lager_h exited with reason {'EXIT',{{case_clause,[http,<0.286.0>,39,<0.342.0>,badarg,[{erlang,binary_to_integer,[<<"undefined">>,2],[]},{lorawan_admin,parse_adr,1,[{file,"/home/gl/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_admin.erl"},{line,121}]},{lorawan_admin,'-parse_admin/1-fun-0-',1,[{file,"/home/gl/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_admin.erl"},{line,83}]},{lists,map,2,[{file,"lists.erl"},{line,1238}]},{lists,map,2,[{file,"lists.erl"},...]},...]]},...}}

gotthardp, 2016-12-27:
Could you try it once more, please? :)

mzemb, 2016-12-28:
This time the issues is solved (with commit 6403edaa).
I've play a lot with the counters, setting high/low values, resetting my nodes, I haven't encountered any problem.
Thank you.

closed 2016-12-28 by mzemb