gotthardp/lorawan-server #16: unknown devaddr is an error ?

mzemb, 2016-12-28:
Hi.
I'm quite new to lorawan, but I wonder why an unknown devaddr is reported as an error.
From my understanding, the gateway job is not to filter anything, but to forward all packets to the server. If that's correct, why is an unknown node considered as an error ? I suppose it will be a common scenario on private networks that the gateway pickups other nodes (ie not the ones that I deployed).

Should the error message be changed to a simple debug or info, or I'm wrong in my assumptions ?

gotthardp, 2016-12-28:
Hi. I assumed that all devices we receive data from belong to our network. Hence that the unknown devaddr is an *error* in configuration because that unknown device should be correctly enlisted.

However, you are right this is a rather "soft error", so it could easily be just a warning. I still think it's more than a notice-- it's something the operator may need to pay attention to. Notices on the other hand are meant as indicators that the server behaves correctly, so the operator doesn't have to pay attention to it.

Does this make sense, or you see some "foreign" devices and the messages annoy you?

mzemb, 2016-12-29:
Well in my case the device did not belong to my network. So I'm annoyed by a message that is really not an error. But I understand your reluctance to have it as a notice only. I would say that a warning is a good middle ground.
I'm considering anyway to change it to a simple notice for me.
Or maybe this could be a server configuration ?

mzemb, 2016-12-29:
I just saw your commit, so I'll close this.

closed 2016-12-29 by mzemb

gotthardp, 2016-12-29:
I was considering to make some sort of an "ignore list"-- a list of devices we hear, but which belong to another network.

mzemb, 2016-12-30:
I like the idea.