gotthardp/lorawan-server #17: single channel gateway on US 902.3 MHz frequency using SF7

dlarue, 2017-01-03:
Hi, I'm trying to get your lorawan-server running in a docker contain and seem to have it running but can not see data getting into the lorawan-server from my single channel gateway.  What I've done is:

boot the container and manually executed the Dockerfile commands, edited the sys.config file to use the 902.300 frequency and then run bin/lorawan-server.

On my Single Channel Gateway I've set ports to 1680 and set the router address to 172.17.0.2.  I can see and use the lorawan-server admin console on port 8080 so I know the server works and the IP address is correct.  Next I added my gateway using the EUI specified from the single channel gateway web interface/page.  I just made up a NetID, 0123AB

Any information on what I might be doing wrong and where to debug next? Thank you.

gotthardp, 2017-01-03:
Hi. Are you able to see the log file written by the lorawan-server? The best way to start is to check the log for errors and warnings. The gateway and/or device registration may be wrong. If there is no error nor warning, the connectivity between your gateway and the server is the issue.

dlarue, 2017-01-03:
I've not found any logs but an empty crash.log file.  Starting lorawan-server on the commandline provides lots of debug output on that console and nothing indicates errors.  How might I enable logging?

gotthardp, 2017-01-03:
The console log contains all debug information available. So you have a gateway, some LoRa device communicating with it and the console log does not show anything except a long list of startup messages. Nothing appears in the long when the device sends a message, right?

dlarue, 2017-01-03:
correct, nothing shows up in the console/startup log but startup messages. I'm sure now that I have a routing issue with IP traffic since the lorawan-server is running in a docker container. I will fix that and see what happens after that. But it will be a few hours before I can get back on this.  I will post updates tonight. Thanks.

gotthardp, 2017-01-03:
Thanks. I just made a little update (commit 5ae2ce7), so you can see a log message when a gateway successfully connected to the lorawan-server. It's a useful information and it may help you debugging your connectivity issue.

dlarue, 2017-01-03:
It was my LAN routing issue. I used socat to redirect TCP/IP 8080 and TCP/UDP 1680 and saw this on the lorawan-server log:

21:38:03.807 [warning] Unknown DevAddr 26011668
21:38:03.807 [error] ERROR: {unknown_devaddr,<<38,1,22,104>>}

Now to figure out how to define ADB devices. Thanks.

gotthardp, 2017-01-03:
To add this device, just add a new "Link", copy-paste the DevAddr "26011668" and enter the keys.

dlarue, 2017-01-03:
Fantastic, seeing data now:

21:49:30.965 [error] ERROR: bad_mic
21:49:55.054 [debug] LinkADRReq {1,5,1}
21:49:55.054 [error] ERROR: {unexpected_data,{rxdata,1,<<3,3,3,28,220,148,65,139,114,203,66>>,false,true}}

It's a LMIC based ProMini node so I will probably have to figure out how to create a application filter. Using the default Semtech handler so probably why the error.

gotthardp, 2017-01-03:
The `unexpected_data` error is because you use some other (not yet supported) application. Some description of how to build the app handlers is [here](https://github.com/gotthardp/lorawan-server/blob/master/doc/Handlers.md).

I am more worried about the `bad_mic` error. It shouldn't happen if everything is well configured and it's not related to app handlers.

gotthardp, 2017-02-13:
Hello, does the problem persist?

gotthardp, 2017-02-16:
Support for US 902.3 MHz was introduced in the release 0.3.0.

closed 2017-02-16 by gotthardp

dlarue, 2017-02-16:
I've not had time to get back to this.  I build this into a bootable LiveCD for workshops I've been doing but out of 10 people nobody has shown interest in the Docker side so I've just got been back there. Soon though as I will be helping a school do some long range balloon tests and we'll be running local servers with a gateway in the chase vehicles.  I'll open any issues I run into.

gotthardp, 2017-02-16:
OK. If the problem persists, feel free to raise a new issue.