gotthardp/lorawan-server #18: Troubles with websockets

Marcze, 2017-01-10:
Please help me to debug websockets issue.
Cannot use ws.html page.

**Lora server:** Ubuntu 16.04 .Compiled from sources, Jan 5.
**Client:** Firefox, Ubuntu.
**Page:**  zzz.zz:8080/admin/ws.html

**Logs (F12 key in Firefox)**
Firefox can’t establish a connection to the server at ws://zzz.zz:8080/ws/00000000000000FF/raw

**Wireshark from client (network sniffer)**

_Frame 13: 653 bytes on wire (5224 bits), 653 bytes captured (5224 bits) on interface 0_
Hypertext Transfer Protocol
    GET /ws/00000000000000FF/raw HTTP/1.1\r\n
    Host: zzz.zz:8080\r\n
    User-Agent: Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:48.0) Gecko/20100101 Firefox/48.0\r\n
    Accept: text/html,application/xhtml+xml,application/xml;q=0.9,*/*;q=0.8\r\n
    Accept-Language: en-US,en;q=0.5\r\n
    Accept-Encoding: gzip, deflate\r\n
    Sec-WebSocket-Version: 13\r\n
    origin: http://zzz.zz:8080\r\n
    Sec-WebSocket-Extensions: permessage-deflate\r\n
    Sec-WebSocket-Key: cGa4Ye3GBYRDb+Ig15hMFQ==\r\n
    Authorization: Basic YWRtaW46dGFyYW50dWxhMTcq\r\n
    Connection: keep-alive, Upgrade\r\n
    Pragma: no-cache\r\n
    Cache-Control: no-cache\r\n
    Upgrade: websocket\r\n
    \r\n
    [Full request URI: http://zzz.zz:8080/ws/00000000000000FF/raw]
    [HTTP request 1/1]
    [Response in frame: 16]

_Frame_ 16: 196 bytes on wire (1568 bits), 196 bytes captured (1568 bits) on interface 0
Hypertext Transfer Protocol
    **HTTP/1.1 400 Bad Request\r\n**
    Content-Length: 0\r\n
    Date: Tue, 10 Jan 2017 08:24:03 GMT\r\n
    Server: Cowboy\r\n
    Keep-Alive: 60\r\n
    Via: HTTP/1.1 proxy10030\r\n
    \r\n
    [HTTP response 1/1]
    [Time since request: 1.237749933 seconds]
    [Request in frame: 13]

**Server log**
Nothing if correct DevEUI is specified.
If I enter wrong DevEUI, it shows;
 lorawan-server[1354]: 09:55:22.147 #033[1;33m[warning] WebSocket for unknown DevEUI: <<0,0,0,0,0,0,0,250>>#033[0m

gotthardp, 2017-01-10:
Hi. The very last line is important `WebSocket for unknown DevEUI: <<0,0,0,0,0,0,0,250>`
This suggests you have no *Device* with DevEUI `00000000000000FF` with Application set to WebSocket, or there is no *Link* for that device. Could this be the case?

Marcze, 2017-01-10:
Hi,

This is not the case.

I put wrong DevEUI only once, just for test, to see if connect button works. Server responds wrong DevEUI in log, as intended.

When I use correct DevEUI, there is no warning message in log but ws.html page still doesn't work.
I double checked, I have DevEUI device in Links and application is set to websocket.

Marcze, 2017-01-10:
I can be network security setting on my side. I Checked in different location and works OK.

closed 2017-01-10 by Marcze

gotthardp, 2017-01-10:
Nice to hear. Thanks for leting me know.

MoonBeast1994, 2017-02-07:
Hello gotthardp ! Nicely done with the private lorawan server !
I am currently running v0.2.0 precompiled release without any problems.

I just have 2 questions.

1) Where can i see the actual packet parameters such as payload,bw,sf,snr.. etc ?

i have successfully send packets from a node and i can see the received packets at the graphs at "link page".

2)Downlink for link registered device

Using admin/ws.html  there is option for DevEUI. Is there an option for using Devaddr ?

gotthardp, 2017-02-07:
Hello.

1) You can get the payload via websocket, but you cannot get the parameters. The physical/MAC layer parameters are kept inside the server. Do you really need them?

2) In 0.2.0 you'd need to create a fake device linked with the DevAddr you want to see. In the next release (current HEAD) you can simply use either DevEUI or DevAddr. The server distinguishes them based on the length.

MoonBeast1994, 2017-02-11:
Thanks for your reply ! I could use the physical/MAC layer parameters for testing the protocol. Would it be possible to extract them on a later revision ?