gotthardp/lorawan-server #19: Brocaar's Loraserver comparison

fifthaxe, 2017-01-10:
Hi Petr,

Trying to compare your work and Brocaar implementation (https://github.com/brocaar/loraserver), do you consider both projet beeing directly comparable ? I am wondering what was your motivation in starting this project from scratch whereas you may have forked an existing one.

By the way thank you very much for opensourcing such a great work !

gotthardp, 2017-01-10:
Hello. The feature sets are different, but this will evolve as both servers are under active development. The main difference (and the reason for building a second open-source loraserver) is the software architecture and the intended deployment.
Brocaar implementation is made from several servers, bridges and controllers, so I *assume* the intended deployment is a large distributed public network, operated by several collaborating entities. This architecture may become an overkill of you are a single entity operating a private network, either as an individual geek who wants to try a cool lora sensor, or as a commercial company who wants to collect data from the sensors deployed around their factory. My server aims at these private deployments and explicitly excludes the public distributed deployments. Everything runs in one "box", including the LoRa applications, so it *could* be easier to deploy, maintain and secure.
A second big difference is the language used. My server is written in Erlang (which I love), because it is very suitable for implementation of reliable near-realtime, message oriented protocols (just like LoRa). You may want to read [Some Thoughts on Go and Erlang](http://blog.erlware.org/some-thoughts-on-go-and-erlang). Erlang even supports scalability (clustering), so my server could (one day) support large private networks, but the way it achieves that is fundamentally different to the Brocaar implementation. I am not saying mine is better (or worse); it's just very different and designed to meet different needs. My goal is to complement rather than to compete-- to address different use-cases using a different architecture.
A third reason for building the server is personal-- I just wanted to learn LoRaWAN by building yet another server (because it's so easy to build servers in Erlang).

altishchenko, 2017-01-12:
High Five!

All of the reasons listed below (and Erlang most notably) were my guides to the choice of your server implementation instead of Brocaar’s one. I am working on a few sensors with Cisco’s gateway in between and your server is the best one I came across. And giving Erlang’s natural process distribution, in-flight upgrades and You Know, Oh! What a future!

With best regards,
Alexander Tishchenko
mob: +7 (916) 704-61-57
atishch@gmail.com

gotthardp, 2017-01-12:
Thank you for such a nice comment, Alexander. I am very glad the server meets your needs.

lsxredrain, 2017-01-13:
I have integrated into lorawan-server into emqtt  as a plugin,  it is excellent!

altishchenko, 2017-01-15:
Hello Jhon,,
Can you share you solution? May be in a form of markdown howto to this project?

With best regards,
Alexander Tishchenko
mob: +7 (916) 704-61-57
atishch@gmail.com

sigysmund, 2017-03-28:
I will also have a try with this one...
like was said, Erlang has a bit more tradition in messaging/broadcasting.

Thanks for a great work done!

altishchenko, 2017-03-29:
MQTT is on the way, RSN. There were serious changes by Petr in the code since the topic you are quoting was started and by far we are very close to a working interoperability with cisco now (they are not using semtech UDP PF, they have an Actility stack built-in).

Cheers,
Alexander Tishchenko
mob: +7 (916) 704-61-57
atishch@gmail.com

closed 2017-07-15 by gotthardp