gotthardp/lorawan-server #20: difference devices/links and websockets usage

mzemb, 2017-01-24:
Hi.

I'm not sure to have understood the difference between the devices and the links section. In the doc (
doc/Administration.md), I read that "Once the device joins the network, the Link field will contain a reference to the Links list."
Does that mean that the devices section is there for devices that will go through a join procedure in the OTAA phase ? If that's the case and if I want to use ABP, can I directly enter a link that will not have a corresponding device entry ?

From the official lorawan spec, section 6, "end device activation", I understand that I don't need (nor can) specify the DevEUI in the case of ABP. 
I'm trying to access data from a device activated through ABP through the websocket raw interface (http://<ip>:8082/admin/ws.html ), but I need the DevEUI for that, and when creating a link I enter only the DevAddr, not the DevEUI.

Did I miss something ?
(I'm using your 0.2.0 release).

Thanks,

Regards,

MichaEl

gotthardp, 2017-01-24:
Hello. Yes, the *Devices* list is for those that join via OTAA. If you want an ABP device, adding a *Link* is just enough. However, the web socket interface has a deficiency-- it cannot handle ABP devices. I will fix that later today.

mzemb, 2017-01-24:
Sweet, thanks.

gotthardp, 2017-01-24:
Should be fixed now (commit 8aa849a7d603ce3361c854e8446bca26d7bf0975). Please let me know if it works for you.

mzemb, 2017-01-26:
I confirm that commit 8aa849a closes the issue.
I'm now able to see the data of a ABP'ed device through the websocket client by entering the DevAddr instead of the DevEUI.
(To be perfect, you should change the text label to reflect the fact that you can also enter a devAddr).
Thank you.

closed 2017-01-26 by mzemb