gotthardp/lorawan-server #21: Start server on Windows 10 machine

jmpmscorp, 2017-02-02:
Firstly, thank you very much for your job and excuse my ignorance. I'm absolutely unfamiliar with erlang...

I'm trying to set up your lorawan-server on my Windows 10 machine for local testing puporses. I downloaded erlang 19.2 and I installed without problems. I could test it with erlide and works.

I have downloaded your release and unzipped. On created folder, I go to 'bin' folder and launch lorawan-server.bat. A quick CMD window opens but dissapears in a brief... Anything more happen.

I open my chrome browser and  I try to navigate to localhost:8082, localhost:8082/admin but no response... In sys.config port is set at 8082. Seems like any server is running...

Am I doing anything wrong?? Could you guide me??

Again, thank you very much. Regards.

gotthardp, 2017-02-02:
Hello. It *should* work. Could you please run CMD from the start menu, then change directory to where you put the release and then run `bin\lorawan-server.bat`. It likely will not work, but I hope it may print some error message, which will not disappear. Please let me know what did it print.

jmpmscorp, 2017-02-02:
Thank you very much for reply @gotthardp.

I have done that you told me and I have a problem with "erl" command. When I launch 'lorawan-server.bat' through cmd window an error like "erl is not known as intern or extern command..." is shown. In spit of that erlide works, seems that I have something wrong with erl path or command... I'll try to solve it and then, launch again your server.

If you don't mind, have you planned implement class C devices?

Thank again. Regards.

gotthardp, 2017-02-02:
I see. I have erl.exe in `C:\Program Files\erl7.1\bin`. Either add your erl.exe dir into PATH, or modify the 'lorawan-server.bat' file and put there a full path to erl.exe.

The implementation is "demand driven", i.e. I implement only the features someone needs. If you need some class C features, I can consider adding them to my TODO list, but I'd need to know some more details about the intended behavior.

jmpmscorp, 2017-02-03:
Hi again @gotthardp.

I could solve the problem. I added erl path to my enviroment and .bat is launched like a charm. Now, I can navigate through localhost, see webpages,...

Respect to class C devices, we are trying to build a real-time control system, where sometimes, randomly, we must on/off some devices. With class A downlinks are only sent when device sent previously uplink but we must sent commands to various devices without previous downlinks. We are studying Lora and LoraWan to set up this configuration, so we think we need class C implementation.

I'm trying to learn erlang and understand your code. Maybe, in the future I could use your code and add functionalitys.

Thank you very much. Best regards.

gotthardp, 2017-02-04:
Concerning the class C: I can quite easily (and quickly) add a functionality to send "immediate message", which will not wait for uplink. How will you interface the lorawan-server? Native erlang, websocket, or you'd prefer something else?

jmpmscorp, 2017-02-05:
We haven't decided yet how we interface but we are considering some options.

a. Use native erlang. This is the most improbable. I'm completely unfamiliar with erlang but perhaps, It's a good reason to learn.

b. Use MQTT.

c. Use websocket over socket.io. I'm familiar with NodeJS.

Which do you think would be the best option?

gotthardp, 2017-02-05:
There is no best option that matches all needs. It depends on your architecture and your application server. But MQTT seems to be used by some of the big players. This reminds me I should implement MQTT support.

altishchenko, 2017-02-05:
Or may be take the emqttd/emqttc (https://github.com/emqtt/emqttc) as a starting point. They seem to be very active in development and one of the guys here already managed to integrate the two products. Unfortunately he didn’t share his solution though… I started there myself, but had to postpone because of other necessities and websockets are quite enough for us at the moment.

With best regards,
Alexander Tishchenko
mob: +7 (916) 704-61-57
atishch@gmail.com

gotthardp, 2017-02-06:
👍  And this is exactly what I am going to use. There are not many open-source Erlang MQTT clients.

altishchenko, 2017-02-06:
Hello Petr,
Can you drop me an email to the address below in the signature? I am into it myself and we can keep in touch. (I am sorry, I don’t know how to send direct messages via github and raising issues is not a good idea.) Tried linked-in, but just to find out that access to it is blocked in our country %)

With best regards, 
Alexander Tishchenko
mob: +7 (916) 704-61-57
atishch@gmail.com

gotthardp, 2017-02-07:
The issue was fixed.

closed 2017-02-07 by gotthardp

gnob417, 2017-12-07:
@gotthardp 

Hi petr,
could I compile whole server source code in the erlide environment?
I use Windows 10 OS and Eclipse with erlide.
If  i could, what things should be installed to compile whole codes?

gotthardp, 2017-12-07:
Hi, @gnob417. This is in the Build Instructions described: https://github.com/gotthardp/lorawan-server/blob/master/doc/Installation.md#manual-installation

gnob417, 2017-12-08:
I mean how to use erlide & eclipse to compile codes with rebar3.

your suggestion is just using 'make' in window cmd?

gotthardp, 2017-12-08:
`make` won't work in windows. you should do `rebar3 release`
I have no experiences with erlide.