gotthardp/lorawan-server #22: Class C devices not supported

gotthardp, 2017-02-07:
Implement ability to send a message "immediately" and not wait for the RX windows. Requested by @jmpmscorp in issue #21.

Rapsoulhs, 2017-02-23:
Do you have a time frame of delivery for Class C support?

gotthardp, 2017-02-23:
It also depends on the user demand.
The 0.3.0 is about to be released. The 0.4.0 will focus on automatic ADR, which may take me 2-4 weeks to implement and test. If nothing more important pops up I could do Class C after that (March-April).

gotthardp, 2017-03-02:
Just so you know: I increased the priority of Class C implementation and will implement the application-initiated downlinks as well as multicast now. Should be ready next week.

gotthardp, 2017-03-04:
I believe I implemented a (basic) Class C support. @jmpmscorp or @Rapsoulhs, would you be able to test it, please?

jmpmscorp, 2017-03-04:
Hi @ghottardp. Thank you very much dor your work. It's marvelous. I can test your new implementation yet, I haven't any C class device... As soon as I could, I'll tell you but probably no until next month or more. I hope someone could.

I can't only say thank you.

batulzii, 2017-03-27:
Hi Petr, 
Please, set list where we can select Class C/A in the web admin.
Thanks

gotthardp, 2017-03-27:
Hi. You don't need to set this. Just use the `immediately` flag as described at https://github.com/gotthardp/lorawan-server/blob/master/doc/WebSockets.md

batulzii, 2017-03-27:
got it. will test it

batulzii, 2017-03-28:
ws.html sending but where can i set **immediately** ?

gotthardp, 2017-03-28:
Switch to JSON and then write something like `{"data":"00", "time":"immediately"}` to the *Hex String or JSON* field.

batulzii, 2017-03-28:
got problem, log is bellow:
2017-03-28 21:33:42.968 [error] <0.2119.0> CRASH REPORT Process <0.2119.0> with 0 neighbours exited with reason: call to undefined function lorawan_mac_commands:handle({link,<<1,165,0,169>>,<<"EU863-870">>,<<"websocket">>,undefined,undefined,<<11,22,213,130,141,172,...>>,...}, <<>>) in cowboy_clear:proc_lib_hack/5 line 34
2017-03-28 21:33:42.969 [error] <0.2119.0> Ranch listener http terminated with reason: call to undefined function lorawan_mac_commands:handle({link,<<1,165,0,169>>,<<"EU863-870">>,<<"websocket">>,undefined,undefined,<<11,22,213,130,141,172,...>>,...}, <<>>)

gotthardp, 2017-03-28:
Oh, this is a bug, sorry for that. I will fix that in few hours.

gotthardp, 2017-03-28:
Could you try again, please?

batulzii, 2017-03-28:
Thanks. Seems to be ok. but i need to debug my device, because it not shown as received command.
Can we do set immediately with raw data in ws.html?

gotthardp, 2017-03-28:
I was not able to test it as I don't have any class C device, so it may be my fault also. You may want to check your gateway log (or some LEDs) to make sure it was really sent.
In the "raw" output the immediately flag cannot be used. More complex functions need a more complex interface.

batulzii, 2017-03-28:
All right. I will test it and let you know. Thanks

batulzii, 2017-03-29:
Tested. Uplink is ok in the web admin, but ws.html do not show. Other devices ok in both end.
Differneces class. Class A can be shown on either end. Class C can be seen in web admin's Received Frames. and no in the ws.html. Downlink still could not tested yet. Need fix Uplink first.

gotthardp, 2017-03-29:
Ok. You believe this is a bug on your side or mine? Did you set the application correctly? Did you check the logs?

batulzii, 2017-03-29:
all devices (class A/C) tested with loriot.io in either direction. Any issue with class C in the server handler?

batulzii, 2017-03-29:
any advice on  not getting payload on ws.html, when it is available on Received Frames.

batulzii, 2017-03-29:
error logs are empty

gotthardp, 2017-03-29:
And the console log with level set to 'debug' (see Troubleshooting.md)

batulzii, 2017-03-29:
3 times sended json from ws.html 
out ot console is:
14:46:48.935 [debug] <<1,165,0,169>> <- [dev_status_req]
14:47:12.356 [debug] <<1,165,0,169>> <- [dev_status_req]
14:47:26.460 [debug] <<1,165,0,169>> <- [dev_status_req]

batulzii, 2017-03-29:
upload message from device do not output message to console

batulzii, 2017-03-29:
14:46:48.935 [debug] <<1,165,0,169>> <- [dev_status_req]
14:47:12.356 [debug] <<1,165,0,169>> <- [dev_status_req]
14:47:26.460 [debug] <<1,165,0,169>> <- [dev_status_req]
15:45:57.525 [debug] Expired 2 rxframes from <<1,77,235,241>>

batulzii, 2017-03-29:
Packet Forrwarder:

INFO: [down] PULL_RESP received :)
INFO: [down] a packet will be sent in "immediate" mode
59.61.99.0.0.0.0.9.0.9c.d.18.0.8.0.0.60.a9.0.a5.1.81.a.0.6.bc.ea.fe.66.end
INFO: [down] PULL_ACK received in 7 ms

gotthardp, 2017-03-29:
If you see this, you should get the frame received by the device. Still not receiving the uplinks?

batulzii, 2017-03-30:
device code under debugger, but never detect RX. This code tested with loriot.io.  Any advice
Ohter issue is ws.html can output payloads from class A, but not from class C.  They should be the same in this mode, right?

batulzii, 2017-03-30:
990012FFFFFF1FFF80FF81FFFFFFFFFF01
 this is payload should display on ws.html. this issue not from class, sense i just change class of device to class A. but ws.html still not output payload while web admin displays.

batulzii, 2017-03-30:
this device sends a lot of payload and ws.html catch once.

batulzii, 2017-03-31:
I debugging device and got payload as bellow:
60A900A50181690006BB6A3278 ---- all zeros
60 - DOWNLINK
A900A501 - Device address
81 - FCtrl
69 - FCnt it is the same increesing

Any idea?

gotthardp, 2017-03-31:
This looks like a valid downlink payload. Is it something wrong with it?

batulzii, 2017-03-31:
{"data":"780178", "time":"immediately"}  is what i sending, but still could not get Rxdata in main loop.
i catched above payload on LoRaMac.c on breakpoint.

batulzii, 2017-03-31:
size Mac payload is only 0x0D so there is no data. i guess

gotthardp, 2017-03-31:
For class C you need to specify port explicitly. Please try sending `{"data":"780178", "port":2, "time":"immediately"}` or so

batulzii, 2017-03-31:
Thanks, forget about port at all. sorry for your time

gotthardp, 2017-03-31:
I am sorry. There were examples showing `immediately` without a port number. I added some warning and a little explanation so this doesn't happen in the future. And I also managed to test it works, so I declare Class C as completed. ;-)

closed 2017-03-31 by gotthardp