gotthardp/lorawan-server #24: erlang.cookie

matchxiot, 2017-02-08:
Hi, we would like to run this on OpenWrt,

```
root@OpenWrt:~# ./loraserver/bin/lorawan-server
{error_logger,{{2017,2,8},{12,3,49}},"Error when reading /root/.erlang.cookie: einval",[]}
```

And there shouldn't be any problem with .erlang.cookie, so we totally stuck here.

Is there anyone can help? Thanks!

matchxiot, 2017-02-08:
```
root@OpenWrt:~# cat loraserver/erl_crash.dump 
=erl_crash_dump:0.3
Wed Feb  8 13:28:05 2017
Slogan: Kernel pid terminated (application_controller) ({application_start_failure,kernel,{{shutdown,{failed_to_start_child,net_sup,{shutdown,{failed_to_start_child,auth,{"Error when reading /root/.erlang.c
System version: Erlang/OTP 17 [erts-6.4] [source] [async-threads:10] [kernel-poll:false]
Compiled: Wed Feb  8 12:22:57 2017
Taints: 
Atoms: 5085
=memory
total: 5378272
processes: 2283600
processes_used: 2278540
system: 3094672
atom: 98917
atom_used: 86271
binary: 36168
code: 1539948
ets: 26932
=hash_table:atom_tab
size: 3203
used: 2537
objs: 5085
depth: 8
=index_table:atom_tab
size: 5120
limit: 1048576
entries: 5085
=hash_table:module_code
size: 47
used: 31
objs: 50
depth: 3
=index_table:module_code
size: 1024
limit: 65536
entries: 50
=hash_table:export_list
size: 3203
used: 1438
objs: 1879
depth: 5
=index_table:export_list
size: 2048
limit: 524288
entries: 1879
=hash_table:export_list
size: 3203
used: 1434
objs: 1869
depth: 5
=hash_table:process_reg
size: 11
used: 4
objs: 4
depth: 1
=hash_table:fun_table
size: 397
used: 192
objs: 273
depth: 4
=hash_table:node_table
size: 11
used: 1
objs: 1
depth: 1
=hash_table:dist_table
size: 11
used: 1
objs: 1
[... 263 lines cut ...]
mbcs carriers: 1 1 1
mbcs mseg carriers: 0
mbcs sys_alloc carriers: 1
mbcs carriers size: 131104 131104 131104
mbcs mseg carriers size: 0
mbcs sys_alloc carriers size: 131104
sbcs blocks: 0 0 0
sbcs blocks size: 0 0 0
sbcs carriers: 0 0 0
sbcs mseg carriers: 0
sbcs sys_alloc carriers: 0
sbcs carriers size: 0 0 0
sbcs mseg carriers size: 0
sbcs sys_alloc carriers size: 0
ets_alloc calls: 75
ets_free calls: 71
ets_realloc calls: 2
mseg_alloc calls: 0
mseg_dealloc calls: 0
mseg_realloc calls: 0
sys_alloc calls: 1
sys_free calls: 0
sys_realloc calls: 0
=allocator:fix_alloc
```

gotthardp, 2017-02-08:
Yes, I can try to help.

First of all, please note that you need at least Erlang R18. I guess you are using R17. Could you please migrate to R18 or R19 and let me know if the problem (magically) disappears (or not)?

Could you also please make sure there is no `/root/.erlang.cookie` file before you first start the server?

matchxiot, 2017-02-08:
HI thanks, I have no idea how to migrate it to R18, openwrt seems to only support R17.

I have removed ```/root/.erlang.cookie``` several times, and it didn't work.

gotthardp, 2017-02-08:
When you type `erl` on command line. What happens? Does the Erlang start or not?

matchxiot, 2017-02-08:
It starts

```
Erlang/OTP 17 [erts-6.4] [source] [async-threads:10] [kernel-poll:false]

Eshell V6.4  (abort with ^G)
1> 
```

gotthardp, 2017-02-08:
But Erlang/OTP 17 is missing some of the crypto-functions I need. The server will not really work under R17. Do you build openwrt yourself?

matchxiot, 2017-02-08:
I think so, please advise me what values to change in makefile for erlang R18, thanks :)

```
include $(TOPDIR)/rules.mk

PKG_NAME:=erlang
PKG_VERSION:=*17.5*
PKG_RELEASE:=3

PKG_SOURCE:=otp_src_$(PKG_VERSION).tar.gz
PKG_SOURCE_URL:= http://www.erlang.org/download/ \
	http://erlang.mirror.su.se/
PKG_MD5SUM:=*346dd0136bf1cc28cebc140e505206bb*

PKG_LICENSE:=ErlPL-1.1
PKG_LICENSE_FILES:=EPLICENCE
PKG_MAINTAINER:=Nicolas Thill <nico@openwrt.org>

PKG_BUILD_DEPENDS:=erlang/host openssl
PKG_USE_MIPS16:=0

HOST_BUILD_DIR:=$(BUILD_DIR_HOST)/otp_src_$(PKG_VERSION)
PKG_BUILD_DIR:=$(BUILD_DIR)/otp_src_$(PKG_VERSION)

include $(INCLUDE_DIR)/host-build.mk
include $(INCLUDE_DIR)/package.mk

define Package/erlang/Default
  SUBMENU:=Erlang
  SECTION:=lang
  CATEGORY:=Languages
  TITLE:=Erlang/OTP programming language
  URL:=http://www.erlang.org/
endef

define Package/erlang/Default/description
 Erlang/OTP is a general-purpose programming language and runtime
 environment. Erlang has built-in support for concurrency, distribution
 and fault tolerance.
endef

define Package/erlang
$(call Package/erlang/Default)
  DEPENDS+= +libncurses +librt +zlib
  PROVIDES:= *erlang-erts=6.4 erlang-kernel=3.2 erlang-sasl=2.4.1 erlang-stdlib=2.4*
endef

define Package/erlang/description
$(call Package/erlang/Default/description)
```

gotthardp, 2017-02-08:
This is going to be fun, please be patient. :-) I updated a Makefile and put it to https://github.com/gotthardp/packages/blob/master/lang/erlang/Makefile. If we succeed to compile it, I will create a push request against the openwrt repository.

matchxiot, 2017-02-08:
```
root@OpenWrt:~# ./loraserver/bin/lorawan-server
2017-02-08 15:41:19 Error when reading /root/.erlang.cookie: einval
2017-02-08 15:41:19 crash_report        
    initial_call: {auth,init,['Argument__1']}
    pid: <0.45.0>
    registered_name: []
    error_info: {exit,{"Error when reading /root/.erlang.cookie: einval",[{auth,init_cookie,0,[{file,"auth.erl"},{line,286}]},{auth,init,1,[{file,"auth.erl"},{line,140}]},{gen_server,init_it,6,[{file,"gen_server.erl"},{line,328}]},{proc_lib,init_p_do_apply,3,[{file,"proc_lib.erl"},{line,247}]}]},[{gen_server,init_it,6,[{file,"gen_server.erl"},{line,352}]},{proc_lib,init_p_do_apply,3,[{file,"proc_lib.erl"},{line,247}]}]}
    ancestors: [net_sup,kernel_sup,<0.34.0>]
    messages: []
    links: [<0.43.0>,#Port<0.658>]
    dictionary: []
    trap_exit: true
    status: running
    heap_size: 610
    stack_size: 27
    reductions: 731
2017-02-08 15:41:19 supervisor_report   
    supervisor: {local,net_sup}
    errorContext: start_error
    reason: {"Error when reading /root/.erlang.cookie: einval",[{auth,init_cookie,0,[{file,"auth.erl"},{line,286}]},{auth,init,1,[{file,"auth.erl"},{line,140}]},{gen_server,init_it,6,[{file,"gen_server.erl"},{line,328}]},{proc_lib,init_p_do_apply,3,[{file,"proc_lib.erl"},{line,247}]}]}
    offender: [{pid,undefined},{id,auth},{mfargs,{auth,start_link,[]}},{restart_type,permanent},{shutdown,2000},{child_type,worker}]
2017-02-08 15:41:19 supervisor_report   
    supervisor: {local,kernel_sup}
    errorContext: start_error
    reason: {shutdown,{failed_to_start_child,auth,{"Error when reading /root/.erlang.cookie: einval",[{auth,init_cookie,0,[{file,"auth.erl"},{line,286}]},{auth,init,1,[{file,"auth.erl"},{line,140}]},{gen_server,init_it,6,[{file,"gen_server.erl"},{line,328}]},{proc_lib,init_p_do_apply,3,[{file,"proc_lib.erl"},{line,247}]}]}}}
    offender: [{pid,undefined},{id,net_sup},{mfargs,{erl_distribution,start_link,[]}},{restart_type,permanent},{shutdown,infinity},{child_type,supervisor}]
2017-02-08 15:41:19 crash_report        
    initial_call: {application_master,init,['Argument__1','Argument__2','Argument__3','Argument__4']}
    pid: <0.33.0>
    registered_name: []
    error_info: {exit,{{shutdown,{failed_to_start_child,net_sup,{shutdown,{failed_to_start_child,auth,{"Error when reading /root/.erlang.cookie: einval",[{auth,init_cookie,0,[{file,"auth.erl"},{line,286}]},{auth,init,1,[{file,"auth.erl"},{line,140}]},{gen_server,init_it,6,[{file,"gen_server.erl"},{line,328}]},{proc_lib,init_p_do_apply,3,[{file,"proc_lib.erl"},{line,247}]}]}}}}},{kernel,start,[normal,[]]}},[{application_master,init,4,[{file,"application_master.erl"},{line,134}]},{proc_lib,init_p_do_apply,3,[{file,"proc_lib.erl"},{line,247}]}]}
    ancestors: [<0.32.0>]
    messages: [{'EXIT',<0.34.0>,normal}]
    links: [<0.32.0>,<0.31.0>]
    dictionary: []
    trap_exit: true
    status: running
    heap_size: 987
    stack_size: 27
    reductions: 173
2017-02-08 15:41:19 std_info            
    application: kernel
    exited: {{shutdown,{failed_to_start_child,net_sup,{shutdown,{failed_to_start_child,auth,{"Error when reading /root/.erlang.cookie: einval",[{auth,init_cookie,0,[{file,"auth.erl"},{line,286}]},{auth,init,1,[{file,"auth.erl"},{line,140}]},{gen_server,init_it,6,[{file,"gen_server.erl"},{line,328}]},{proc_lib,init_p_do_apply,3,[{file,"proc_lib.erl"},{line,247}]}]}}}}},{kernel,start,[normal,[]]}}
    type: permanent
{"Kernel pid terminated",application_controller,"{application_start_failure,kernel,{{shutdown,{failed_to_start_child,net_sup,{shutdown,{failed_to_start_child,auth,{\"Error when reading /root/.erlang.cookie: einval\",[{auth,init_cookie,0,[{file,\"auth.erl\"},{line,286}]},{auth,init,1,[{file,\"auth.erl\"},{line,140}]},{gen_server,init_it,6,[{file,\"gen_server.erl\"},{line,328}]},{proc_lib,init_p_do_apply,3,[{file,\"proc_lib.erl\"},{line,247}]}]}}}}},{kernel,start,[normal,[]]}}}"}
Kernel pid terminated (application_controller) ({application_start_failure,kernel,{{shutdown,{failed_to_start_child,net_sup,{shutdown,{failed_to_start_child,auth,{"Error when reading /root/.erlang.coo
```

yes the compile is successfull, and the version goes to R19.2, but it seems that the cookies problem persists. :(

```
root@OpenWrt:~# erl
Erlang/OTP 19 [erts-8.2] [source] [async-threads:10] [kernel-poll:false]

Eshell V8.2  (abort with ^G)
1> 

```

gotthardp, 2017-02-08:
As a workaround, using the `-nocookie` will do the job.

gotthardp, 2017-02-13:
This is related to https://github.com/openwrt/packages/issues/1370

In conclusion, we just need to set the SIZE_OFF_T.
People in that #1370 stated it is solved, but it is not. after I add this CFLAGS, we don't need to set cookie anymore.

closed 2017-02-13 by gotthardp