gotthardp/lorawan-server #25: one wesocket connection for several devices

ShahabR, 2017-02-12:
Hi,
How can I make one websocket connection and get data of several devices?Now, I should make one connection per device, am I right?
Thanks a lot.

gotthardp, 2017-02-12:
Hi, yes. At this moment you need to make multiple websockets, one per device.

ShahabR, 2017-02-13:
Thank you gotthardp for your reply.In loriot platform , the received data has a port number, how can I get the port number in your lorawan server? thank you again.

gotthardp, 2017-02-13:
Right now this is not possible. It will be in the release 0.4.0.

ShahabR, 2017-02-13:
Thanks a lot.

gotthardp, 2017-02-18:
Hi again. I just enhanced the WS interface to (hopefully) meet your needs. See the updated documentation and the commit 8a8607b4507dc875cd83f9caacd8e83d9436e33c.
Please let me know if it works for you.

ShahabR, 2017-02-19:
Hi,
Thank you very much for your enhancements, if I don't make mistake, I should build the source code of the project with your new modified files, am I right? I tried building the project on the Windows 7 OS, but when I try to make it I face an error:"failed to open file:<MyDirectoy>/rebar3". I've copied the rebar3 in one of $Path directories. any idea?Thanks again

gotthardp, 2017-02-19:
I'm checking how to automate the builds so you don't have to do this. Give me an hour or two, please :)

gotthardp, 2017-02-19:
I made a pre-release (v0.3.0-pre.0) and built the binaries for you. Note that the new version can be run as a Windows service too (see lorawan-service.bat).

ShahabR, 2017-02-20:
Dear gotthardp,
Thanks a lot for your new release, I tried with  /links/*456*/... and  /groups/*ABC*/... with both "jason" and "raw" format and they worked correctly.I also used the server as  a windows service and it worked again, great!Only two points:First, the websocket connection closes after about one minute, in my code and also in the "ws.html", and second I tested it with one device, I will try with more devices and will tell you, if any problem, thanks for your time.
[image]

gotthardp, 2017-02-20:
Thanks. I confirm the websocket disconnects after 1 minute. I will try to fix that, it's not on purpose.

gotthardp, 2017-02-20:
The keep-alive is a browser issue-- you can configure your browser to send ping frames. For example Firefox uses `network.websocket.timeout.ping.request`. I described this in [WebSockets.md](https://github.com/gotthardp/lorawan-server/blob/master/doc/WebSockets.md).

ShahabR, 2017-02-21:
Hi,
Thanks a  lot , I followed the point you mentioned and it solved the issue.It seems that  it's not only the browser issue, I had the same problem in my C# code, I solved it by sending ping command by my websocket instance.
And one question, I tested it with the version 0.2.0 of the server, but it was working well without sending keep alive frames, both in C# code and in the browser, any idea?

gotthardp, 2017-02-21:
Oh, yes. The webserver I am using changed its default behavior. I can make the value configurable in the server config to avoid the need to send pings.

gotthardp, 2017-02-21:
I just did like I said. The default timeout is now 1 hour (instead of 1 minute) and you can increase it to `infinity` in the server config (as explained in documentation). It may have security implications, so I don't want to have `infinity` by default.

ShahabR, 2017-02-22:
Thank you very much  for your time and  all of the enhancements, is it implemented in the latest pre release? I couldn't see the wesocket configuration in the last prerelease.

gotthardp, 2017-02-24:
It was not included in any of the 0.3.0-pre releases. It is now included in the final v0.3.0.

closed 2017-02-24 by gotthardp

ShahabR, 2017-02-25:
Hi ,
Thanks for your final V0.3.0, I tested this version ( windows service and group jason format) it worked correctly, but still I can not find  the websocket timeout configuration . The default timeout is 6 minutes I think.

gotthardp, 2017-02-25:
Oh. I forgot one zero. Sorry about that. Simply add {websocket_timeout, 3600000} to your config.