gotthardp/lorawan-server #26: Support for 919MHz to 923MHz

shmrymbd, 2017-02-13:
Can anyone guide me howto start for support for above freq. Thanks

gotthardp, 2017-02-13:
Hello. I am about to release a new version, which will make introduction of new regions much easier.
However, there are many region specific parameters, not just the frequency. Is the band you want to support listed somewhere in the LoRaWAN standard, or is this some unofficial band?

gotthardp, 2017-02-21:
Australia 915-928MHz is supported in the latest version 0.3.0, but not tested. Testers are welcome. Does this meet your needs?

shmrymbd, 2017-03-24:
hi. our freq in Msia.. 919-923MHz. maybe some minor tweak

gotthardp, 2017-03-24:
Could you be more specific, please? Does your frequency belong to the AS923ISM Band, as described in Section 2.7 of the Regional Parameters, or is it something else?

gotthardp, 2017-03-31:
I added support also for South Korean 920-923. It *should* comply with Malaysian 919-923 as well.
Closing this issue for now as I cannot do anything else until the LoRaWAN regional parameters are defined for 919-923.

closed 2017-03-31 by gotthardp