gotthardp/lorawan-server #31: lorawan-server.bat doestn't work

chunis, 2017-03-09:
Just found out the batch script does not work on my win7. And found the causes:
a. my erl.exe is located under folder "Program Files (x86)/" instead of "Program Files/".
b. my erl.exe has a suffix ".exe" instead of just call "erl". I guess if I make Win7 hides the suffix, maybe it will just be "erl", but not tested yet.

So I just replaced the first section of code with "set ERLDIR="C:\Program Files (x86)\erl8.1\bin\erl.EXE", and it works. Besides, my erl.exe is included in the %PATH%, so I think maybe we don't need to try to find the full path, but use "erl" or "erl.exe" is enough.

gotthardp, 2017-03-09:
Thanks for reporting this. Some users have different environment and need %PATH%, some don't. It will take some time to cover all options. Reports like the one you made help a lot. Thanks again. I will fix this in a next release.

gotthardp, 2017-03-11:
I made some changes to the BAT file. If you got a chance, please check whether it works for you now. Thanks.

closed 2017-03-19 by gotthardp

chunis, 2017-03-22:
Sorry for my checking late. Just want to confirm that this bug is fixed. Thanks.

gotthardp, 2017-03-22:
Thank you for the confirmation!