gotthardp/lorawan-server #32: error on euint test.

batulzii, 2017-03-11:
Server installed and performing ok. But error coming in eunit test as bellow:
Please, advice on that issue.

23:18:51.066 [info] Application lager started on node nonode@nohost
23:18:51.068 [info] Application cowlib started on node nonode@nohost
23:18:51.072 [info] Application ranch started on node nonode@nohost
23:18:51.074 [info] Application cowboy started on node nonode@nohost
23:18:51.329 [info] Application mnesia started on node nonode@nohost
23:18:51.334 [info] Application jsx started on node nonode@nohost
23:18:51.335 [info] Application iso8601 started on node nonode@nohost
23:18:51.338 [info] Application lorawan_server_api started on node nonode@nohost
23:18:51.409 [debug] Lager installed handler lager_backend_throttle into lager_event

Pending:
  module 'loramote_tests'
    %% Unknown error: {abort,
                   {setup_failed,
                       {error,
                           {badmatch,
                               {ok,{{"HTTP/1.1",401,"Unauthorized"},
                                    [{"date","Sat, 11 Mar 2017 15:18:51 GMT"},
                                     {"server","Cowboy"},
                                     {"www-authenticate",
                                      "Basic realm=\"lorawan-server\""},
                                     {"content-length","0"}],
                                    []}}},
                           [{test_admin,post_json,2,
                                [{file,
                                     "/home/ulzii/lorawan-server/_build/test/lib/lorawan_server/test/test_admin.erl"},
                                 {line,21}]},
                            {loramote_tests,'-loramote_test_/0-fun-0-',0,
                                [{file,
                                     "/home/ulzii/lorawan-server/_build/test/lib/lorawan_server/test/loramote_tests.erl"},
                                 {line,15}]}]}}}

Top 10 slowest tests (0.000 seconds, 0.0% of total time):
  lorawan_admin:bits_test_/0:147
    0.000 seconds
  lorawan_mac_commands:bits_test_/0:211
    0.000 seconds
  lorawan_mac_commands:bits_test_/0:219
    0.000 seconds
  lorawan_mac_region:region_test_/0:200
    0.000 seconds
  lorawan_mac_region:region_test_/0:202
    0.000 seconds
  lorawan_mac_region:region_test_/0:198
    0.000 seconds
  lorawan_admin:bits_test_/0:151
    0.000 seconds
  lorawan_mac_commands:bits_test_/0:215
    0.000 seconds
  lorawan_admin:bits_test_/0:149
    0.000 seconds
  lorawan_mac_region:region_test_/0:201
    0.000 seconds

Finished in 0.718 seconds
24 tests, 0 failures, 1 cancelled
===> Error running tests

gotthardp, 2017-03-11:
There is a known issue (which I struggle fixing): the eunit cannot be run twice, without removing the mnesia directory.

In the directory where you run `make test` may be a directory named `Mnesia.nonode@nohost` or something similar. Could you try deleting this and trying again?

batulzii, 2017-03-11:
Thanks. Yes, it is indeed ok after removing this directory

gotthardp, 2017-03-11:
Oh, I was enlighted and even fixed this issue. Update to the latest code, delete the directory again and you can run eunit multiple times, without deleting anything.

closed 2017-03-12 by gotthardp