gotthardp/lorawan-server #33: Gateway control

batulzii, 2017-03-12:
It will nice, if there are PING and RESET button in admin.
So can do remote reset and check Ping to gateway.
Thanks

gotthardp, 2017-03-12:
You mean a packet_forwarder based gateway? There is no ping nor reset command in the protocol. How should the reset button work?
With the current packet_forwarder protocol the server could only send some empty frames to the gateway to make sure the network connectivity is OK.

batulzii, 2017-03-12:
YES. Thank you

gotthardp, 2017-03-12:
Hi again. The packet_forwarder gateways are already constantly ping-ing the server (by default once per 10 seconds), so a server initiated ping is probably not necessary.
I created a new gateway field (Last RX), which indicates when the last ping was received and then added a little icon (Alive) to the web admin indicating if the Last RX is too old (older than a minute). This is effectively the gateway connectivity check.

batulzii, 2017-03-12:
Hi, I got it. That is useful. Thank you

closed 2017-03-12 by gotthardp