gotthardp/lorawan-server #34: steps for new connection

batulzii, 2017-03-12:
I'd like to move gateway and devices to your server.Server installed at local ubuntu PC.
I used all (EUI, key) IDs from loriot.io and configured Gateway to 192.168.0.100 (IP of my ubuntu PC).
Added gateway and device. send packet from device.  Not received data yet. Please,advice short checklist.
Thanks

gotthardp, 2017-03-12:
Did you update your gateway configuration to point to the server?
Do you see something in the server log?

batulzii, 2017-03-12:
here is my config file:
{
/* Put there parameters that are different for each gateway (eg. pointing one ga
/* Settings defined in global_conf will be overwritten by those in local_conf

    "gateway_conf": {
        "gateway_ID": "0023CDFFFF000E7E",
        "serv_port_up": 1680,
        "serv_port_down": 1680,
        "server_address" : "192.168.0.100",
        "forward_crc_valid": true,
        "forward_crc_error": false,
        "forward_crc_disabled": true

    }
}

the same in   global config file.
i see in log as bellow:
2017-03-12 19:19:31.930 [error] <0.1073.0>@lorawan_iface_forwarder:process_frame:149 ERROR: {unknown_devaddr,<<1,165,0,169>>}
2017-03-12 19:22:16.614 [error] <0.1076.0>@lorawan_iface_forwarder:process_frame:149 ERROR: {unknown_devaddr,<<1,165,0,169>>}

gotthardp, 2017-03-12:
`unknown_devaddr` means there is an active device, which is not configured in the *Links* section. If it's an OTAA device, restart the device. If it's an ABP device, add it to the *Links* list.

batulzii, 2017-03-12:
Hi Petr,
here is my log.
2017-03-12 20:25:33.213 [error] emulator Error in process <0.1388.0> on node lorawan@GreenMeter with exit value:
{function_clause,[{lists,sublist_2,[undefined,19],[{file,"lists.erl"},{line,355}]},{lists,sublist_2,2,[{file,"lists.erl"},{line,356}]},{lorawan_mac_commands,send_adr,2,[{file,"/home/ulzii/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac_commands.erl"},{line,93}]},{lorawan_mac_commands,handle,4,[{file,"/home/ulzii/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac_commands.erl"},{line,23}]},{lorawan_mac,handle_uplink,5,[{file,"/home/ulzii/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac.erl"},{line,278}]},{lorawan_iface_forwarder,process_frame,3,[{file,"/home/ulzii/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_iface_forwarder.erl"},{line,144}]}]}
2017-03-12 20:25:33.213 [error] <0.275.0>@lorawan_iface_forwarder:handle_info:105 Hanler terminated: {function_clause,[{lists,sublist_2,[undefined,19],[{file,[108,105,115,116,115,46,101,114,108]},{line,355}]},{lists,sublist_2,2,[{file,[108,105,115,116,115,46,101,114,108]},{line,356}]},{lorawan_mac_commands,send_adr,2,[{file,[47,104,111,109,101,47,117,108,122,105,105,47,108,111,114,97,119,97,110,45,115,101,114,118,101,114,47,95,98,117,105,108,100,47,100,101,102,97,117,108,116,47,108,105,98,47,108,111,114,97,119,97,110,95,115,101,114,118,101,114,47,115,114,99,47,108,111,114,97,119,97,110,95,109,97,99,95,99,111,109,109,97,110,100,115,46,101,114,108]},{line,93}]},{lorawan_mac_commands,handle,4,[{file,[47,104,111,109,101,47,117,108,122,105,105,47,108,111,114,97,119,97,110,45,115,101,114,118,101,114,47,95,98,117,105,108,100,47,100,101,102,97,117,108,116,47,108,105,98,47,108,111,114,97,119,97,110,95,115,101,114,118,101,114,47,115,114,99,47,108,111,114,97,119,97,110,95,109,97,99,95,99,111,109,109,97,110,100,115,46,101,114,108]},{line,23}]},{lorawan_mac,handle_uplink,5,[{file,[47,104,111,109,101,47,117,108,122,105,105,47,108,111,114,97,119,97,110,45,115,101,114,118,101,114,47,95,98,117,105,108,100,47,100,101,102,97,117,108,116,47,108,105,98,47,108,111,114,97,119,97,110,95,115,101,114,118,101,114,47,115,114,99,47,108,111,114,97,119,97,110,95,109,97,99,46,101,114,108]},{line,278}]},{lorawan_iface_forwarder,process_frame,3,[{file,[47,104,111,109,101,47,117,108,122,105,105,47,108,111,114,97,119,97,110,45,115,101,114,118,101,114,47,95,98,117,105,108,100,47,100,101,102,97,117,108,116,47,108,105,98,47,108,111,114,97,119,97,110,95,115,101,114,118,101,114,47,115,114,99,47,108,111,114,97,119,97,110,95,105,102,97,99,101,95,102,111,114,119,97,114,100,101,114,46,101,114,108]},{line,144}]}]}

Please, advice
Thanks,

gotthardp, 2017-03-12:
This is a bug, which I fixed yesterday. I'm very sorry for this. It should disappear when you `git pull` the latest code.

batulzii, 2017-03-12:
Hi Petr,
Got compile error..
===> Verifying dependencies...
===> Compiling lorawan_server
===> Compiling src/lorawan_mac.erl failed
src/lorawan_mac.erl:134: field last_qs undefined in record link
src/lorawan_mac.erl:200: field last_qs undefined in record link

gotthardp, 2017-03-12:
Do `make upgrade` before running `make`, please.

batulzii, 2017-03-12:
Hi Petr, 
Fantastic!
Thank you for your time!

closed 2017-03-12 by batulzii

gotthardp, 2017-03-12:
Cool! I am sorry for the troubles. This is a development version and sometimes I make mistakes.
I am writing some basic troubleshooting instructions, so other don't fall in the same trap.