gotthardp/lorawan-server #35: Range Test, Last Seen Online and other questions

Artysteel, 2017-03-12:
**Good day,**

Thank you for such good software.
But there are several questions, if you could answer, would definitely help me out.
Thanks in advance.

**Before I ask here is what I am using:**
1. RPi 2 Model B
2. ic880a (IMST) connected via SPI (reset is made by using script)
3. Raspberry has installed: lorawan_gateway and packet_forwarder (found here on GIT)
4. iu880B USB (IMST). Used as a node, to test, to send to gateway.
5. Your server. Installed on Windows right now on a laptop. 

**Questions:**
1. Which UTILS do I need to launch for this chain to work? (Maybe I am mistaking, but I see everything working when these run: packet logger, packet forwarder, your server) Is this correct?

2. Is it possible to get a status like: Last Seen Online? Example: I unplug my iu880B and see on webpage, that Last Seen Online: 5mins ago / 1 second ago etc.

3. Could you suggest me, please, how to perform a range test with this software/hardware?

4. How can I use collected packet data by your server in my own Web Interface or Mobile apps? Query to database?

5. How does exactly Gate and Node work between each other? I mean - When does gateway knows to receive data from node or not?

Thanks again.

gotthardp, 2017-03-13:
Hello. To answer your questions:

1) It is enough to have the packet_forwarder and the lorawan-server. The logger is not needed.
2) Which version are you using? More recent versions of the user interface have a "Last RX" entry, which means "last seen online". It displays the date instead of "1 sec ago".
3) For a range test it is also good to know the RF Quality parameters. I'd recommend to use what is displayed in the "RX Quality" field.
4) The server stores the last 50 frames for each device only. You can retrieve it via REST API http://server:8080/rxframes
5) This is described in the LoRaWAN standard. The gateway is always listening to frames from devices.

Artysteel, 2017-03-13:
Thanks for answers.

2. Yes I do have Last RX and etc. Will look into this.

P.S

* Is it possible to use 2 concentrators on 1 Raspberry? Will your software this this?
* Is it possible to lower channel to 7.8kHz and 585bps. I see that there are only standarts in concentrator/node and in your software.

gotthardp, 2017-03-13:
* Why would you need to connect two ic880a to one raspberry? It may be possible, but you better ask the packet_forwarder experts. My server doesn't care on what machine the gateway is running.
* Non-standard settings are not possible with the server. You can define your own custom regional settings, which use these values, but I don't recommend doing that. It would not be LoRaWAN anymore.

Artysteel, 2017-03-14:
- Main idea was to make the signal go even farther, if we shorten the radius of the signal. To make it purposeful. This is great for reaching even longer distances. For example we can set 3 concentrators on 1 building all 3 working on 7.8kHz (lowest available) and 585bps (btw these numbers are calculated in LoraWan calculator, so everything should be ok). SO visually we change the way it works from a circle radius signals to an arrow (by using special antennas).

Ahha, so if there will be enough PINs on raspberry, 2 concentrators can be connected, and probably 2 packet forwarders needed, running for each gate

gotthardp, 2017-03-14:
With directional antennas it make sense to have multiple concentrators connected to one raspberry. If you have enough SPI's you should be able to connect multiple ic880a to one PC and run multiple packet_forwarders, one for each ic880a.

The **LoRa** calculator is for the physical layer only. The **LoRaWAN** standard allows few combinations only. However, as long as you don't violate the ISM band legal conditions in your country your are free to develop any **LoRa** based proprietary communication. Of course you can extend my server to support it.

closed 2017-03-19 by gotthardp