gotthardp/lorawan-server #36: Refreshing Received Frames

batulzii, 2017-03-13:
Hi Petr,
There are 2 pages, that contains received frames. Received frames displaying correctly, but refresh done manually. Cool, if data refreshed automatically.
Kind Regards!

gotthardp, 2017-03-13:
It would be cool to auto-refresh the lists, but my current knowledge of Angular and ng-admin is not sufficient to implement it.

kazi93, 2017-04-30:
@gotthardp I would like to work with it or at least test how it works! :) . I will write you in private for instruction. :)

gotthardp, 2017-04-30:
I asked for help at http://stackoverflow.com/questions/43171429/automatically-refresh-fields
Feel free to dig into this problem and propose a solution.

gotthardp, 2017-04-30:
You may want to start with the Dashboard first. I think the auto-refresh would be most useful there.

kazi93, 2017-06-08:
@gotthardp Could you please share the link or the source code for refreshing "Received Frames"? Many thanks.

gotthardp, 2017-06-08:
The source is in https://github.com/gotthardp/lorawan-server/blob/master/priv/admin/admin.js, but it does not refresh and I have no clue how to do it in a clean way.

There is a dirty way to reload the entire page every few seconds, but this is **not** the clean way I have in mind.

gotthardp, 2018-06-26:
With the dashboard timeline I don't think this is needed anymore.

closed 2018-06-26 by gotthardp