gotthardp/lorawan-server #37: Web socket client - no received frame

batulzii, 2017-03-13:
Hi Petr,
I've checked web socket client and its connection ok.
But no received frame yet.
Send data it displays:
2017-3-13 20:06:12 SEND: 0909
Please, advice...
Best Regards!

gotthardp, 2017-03-13:
Is there something suspicious in the log file? Did you set the application to "websocket"? Did you correctly set the DevEUI or DevAddr?

batulzii, 2017-03-13:
Connect and Disconnect are ok.(there are no logs and browser console also see no error).
And outpu is:
2017-3-13 21:28:25 CONNECTED
2017-3-13 21:28:25 Connecting to: ws://localhost:8080/ws/devices/BE7A000000000065/raw
2017-3-13 21:28:24 DISCONNECTED
means client connected with correct device.

batulzii, 2017-03-13:
i have to add port number to line 40 of ws.html, because both server and client(under lampp) are in localhost. change is bellow:
       wsHost = "ws://" + window.location.host + ":8080"+ "/ws/" + $("#type").val() + "/" + $("#server").val() + "/" + $("#format").val();

gotthardp, 2017-03-13:
There are not many error checks. You are probably CONNECTED to a device that is not sending anything.

batulzii, 2017-03-14:
Hi Petr,
Conected device sending payloads. It can be seen in Received Frames in Web client. But there no frames in the samples ws.html. and No error log in the server and No error on the console of browser.
How to debug websocket part of server side? Connected but not sending frames to outside..

gotthardp, 2017-03-14:
Are you sure the *Devices* Iist includes a DevEUI `BE7A000000000065` and its Link points to the DevAddr of the device that is sending the frames? Did you try using this DevAddr instead and connect to `ws://138.197.208.169:8080/ws/links/01010101/raw` instead?

batulzii, 2017-03-14:
Yes, DevEUI according to Link points DevAddr - for the same device that sending. Web client displays payloads. Yes, I checked with Dev Addr without success. How to debug or log Server sending payload to opened socket?

batulzii, 2017-03-14:
I just copy paste all IDs (Keys) from Loriot device page for one device. And tested with your server (web admin) - ok.  Lorawan server and Lampp running on the same PC. ws.html is under Lampp.

gotthardp, 2017-03-14:
Could you please uncomment (remove the `%` character) from the `% {lager_console_backend, debug}` line in your `lorawan-server/releases/<VERSION>/sys.config`? This enables more logs and you should see to what devaddr you connect. The individual frames are not logged though.

batulzii, 2017-03-15:
logs out -> 
13:17:42.186 [debug] WebSocket to link <<1,77,235,241>>
13:20:04.598 [debug] DataRate <<1,77,235,241>>: average snr 8 -30.0 = 38, adr 0 -> step 12
13:20:04.598 [debug] PUSH_DATA <<1,77,235,241>> 8441 2368 67
13:20:14.891 [debug] DataRate <<1,77,235,241>>: average snr 8 -30.0 = 38, adr 0

see websocket to correct link, and PUSH_DATA to the same device..
But can not see payload on the ws.html.

batulzii, 2017-03-15:
ws.html output payload now. nothing touched, but first try was only 2 bytes of data. then i used semtech mote and used as websocket. Then it outputs normaly.

gotthardp, 2017-03-15:
When you see `PUSH_DATA` it suggests you didn't set *Application* of your *Link* to `websocket`.

batulzii, 2017-03-15:
I got it. Thanks

closed 2017-03-15 by batulzii