gotthardp/lorawan-server #38: Devices grouped by Application / AppID

batulzii, 2017-03-15:
Please, add AppID and group devices under Applications. So several applications to be connectd to the server. Currently, Websocket ws.html connecting to server by group/device., it will be connect by AppID and send all devices payloads with correct AppID.

gotthardp, 2017-03-15:
Oh yeah, this is not possible right now. I will invent something so you can connect multiple apps as web sockets.

gotthardp, 2017-03-18:
Done. The server has now both AppID and AppArgs. AppID is used as a WebSocket group or (the future MQTT key), whereas Args are passed to the application. Please let me know if it does not match your needs.

batulzii, 2017-03-18:
Sounds Cool, test it and let you know.
Thanks a lot!

batulzii, 2017-03-19:
Hi, Can you reflected changes in wx.html and web admin(**received frames** could be grouped with AppID).

gotthardp, 2017-03-19:
Can you be more precise, please? wx.html is data agnostic. What comes is displayed.
And what do you mean by "grouping"? You want to add AppID to the received frames list?

batulzii, 2017-03-19:
for example, ws.html websocket sample application can be display payloads by Device EUD/Addr for specific device or all device's payloads by AppID. It will be helpful simply to show how AppID works.
~ just my thought. Then someone will develop websocket application with filter of AppID.

gotthardp, 2017-03-19:
I understand. Try now. I changed the ws.html to be more consistent and added ability to sort and even filter rxframes based on various values, including the AppID.

batulzii, 2017-03-22:
Any movement?  How do you find, if ws.html and server connect under AppID like bellow:
ws://127.0.0.1:8080/ws/appID/11223344/json connects to the application with AppID 11223344 using the JSON format.It displays all devices payloads within this AppID and can be select certain device from list or input.

gotthardp, 2017-03-22:
The URI `ws://127.0.0.1:8080/ws/group/11223344/json` shall work. Doesn't it?

batulzii, 2017-03-22:
what purpose is group, multicast or application. Other question about what is link ?  because we have register devices and that will be enough for connect without link usage? I've been thinking about link is application group when i familiar server first time.

gotthardp, 2017-03-22:
link is what is created once a device joins the network. application is the algorithm receiving the data, multicast group is a special devaddr (as explained in the standard). the WS group is however something different. the terminology is not well settled down obviously

gotthardp, 2017-04-17:
I renamed AppID (again) for (hopefully) better consistency. Before 0.4.0 out will write more documentation.

closed 2017-04-17 by gotthardp