gotthardp/lorawan-server #39: Development & Troubleshouting guide

batulzii, 2017-03-18:
These two guides makes useful for both contributors and users. Please, include detailed server structure.
Seen some other open source server written in Go, but Erlang is robust and was born for this kind of application. So Petr's server is visionary movement. Thank you!

gotthardp, 2017-03-18:
Yeah, I should probably describe:
 * design requirements (that drove some of the decisions made)
 * high-level process architecture and module relationships (who calls whom, who sends messages to whom and why)

Is there something else that you'd appreciate?

batulzii, 2017-03-19:
super. as you've outlined guide content - it will be valuable document.

lhrossouw, 2017-03-22:
Hi,

I used the Things network previously, but I want to setup my own private network.
I got the lorawan-server up and running without any issues, super easy thanks.
But when my node want's to join the network using OTAA I get a join fail on my node side.
I basicaly used the same global_conf.json file as what I used for TTN. 
On my gateway I see
# WARNING: [down] lgw_send failed
# TX errors: 1
Which I didn't see when I used TTN.
Could you maybe give me some tips and advice to further troubleshoot my issue?

see attached log from the lorawan-server, node and gateway.

[lorawan.txt](https://github.com/gotthardp/lorawan-server/files/861786/lorawan.txt)
[global_conf.txt](https://github.com/gotthardp/lorawan-server/files/861788/global_conf.txt)

Thanks,
Lucas

gotthardp, 2017-03-22:
I will try my best assisting you. What gateway are you using?

lhrossouw, 2017-03-22:
I use a custom router with a Multitech MTAC 8 channel LORA card
running:
*** Poly Packet Forwarder for Lora Gateway ***
Version: 2.1.0
*** Lora concentrator HAL library version info ***
Version: 3.1.0; Options: ftdi;

gotthardp, 2017-03-22:
I did setup a new issue for this, lets move there. https://github.com/gotthardp/lorawan-server/issues/41

gotthardp, 2017-04-29:
@batulzii I just added the figure you asked for into https://github.com/gotthardp/lorawan-server/blob/master/doc/Development.md

Please let me know if it meets your expectations.

batulzii, 2017-04-30:
Yes, Thank you ! Petr. It is helpful indeed.

closed 2017-04-30 by gotthardp