gotthardp/lorawan-server #41: gw_send failed

gotthardp, 2017-03-22:
Continuation of the discussion from https://github.com/gotthardp/lorawan-server/issues/39

gotthardp, 2017-03-22:
@lhrossouw, we had similar problems with Multitech gateways in the past, but never really found the root cause. Are you using the lorawan server 0.3.0 or later? Would you be willing to spend some time debugging the gateway behaviour, please?

lhrossouw, 2017-03-22:
Hi Petr,

Yes using the latest lorawan-server downloaded Monday.
I will definitely spend some time on debugging it, but will only be able to
do it tomorrow.

Lets see if we can find the root cause.

gotthardp, 2017-03-22:
Your help would be much appreciated. I never had a gateway with such a problem in my hands. As far as I know:
 * the error is generated by the HAL layer (https://github.com/Lora-net/lora_gateway/blob/v3.1.0/libloragw/src/loragw_hal.c), but it does not provide any detailed reason. Would you be able to rebuild it with `DEBUG_HAL == 1`? This will print the reason what is wrong.
 * there may be a difference in what the various servers send to the gateway (although both are standard compliant). Would you be able to use wireshark.org to record what TTN is sending and see how this is different to what my server sends?

lhrossouw, 2017-03-22:
>    lora_gateway/blob/v3.1.0/libloragw/src/loragw_hal.c

Yes I will be able to do both. Will send you the logs etc ASAP.

lhrossouw, 2017-03-23:
Hi Petr,

##### 2017-03-23 16:00:23 UTC #####
### [UPSTREAM] ###
# RF packets received by concentrator: 0
# CRC_OK: 0.00%, CRC_FAIL: 0.00%, NO_CRC: 0.00%
# RF packets forwarded: 0 (0 bytes)
# PUSH_DATA datagrams sent: 1 (197 bytes)
# PUSH_DATA acknowledged: 100.00%
### [DOWNSTREAM] ###
# PULL_DATA sent: 1 (100.00% acknowledged)
# PULL_RESP(onse) datagrams received: 0 (0 bytes)
# RF packets sent to concentrator: 0 (0 bytes)
# TX errors: 0
### [GPS] ###
# GPS sync is disabled
##### END #####
INFO: [up] PUSH_ACK for server 192.168.222.19 received in 8 ms
lgw_receive:1197: FIFO content: 1 10 0 5 17
lgw_receive:1212: [0 17]
Note: LoRa packet
INFO: [up] PUSH_ACK for server 192.168.222.19 received in 2 ms
INFO: [down] for server 192.168.222.19 serv_addr[ic]PULL_RESP received :)
INFO: [down] a packet will be sent on timestamp value 84373179
**ERROR: INVALID RF_CHAIN TO SEND PACKETS**
WARNING: [down] lgw_send failed
INFO: [down] for server 192.168.222.19 PULL_ACK received in 93 ms

I will send the cap files later

lhrossouw, 2017-03-23:
here is the files tcpdump files, just rename it to .cap

[lorawanserver.txt](https://github.com/gotthardp/lorawan-server/files/865453/lorawanserver.txt)
[ttn.txt](https://github.com/gotthardp/lorawan-server/files/865454/ttn.txt)

gotthardp, 2017-03-23:
`INVALID RF_CHAIN` means you may have an error in your Gateway setting of the server. Could you please check that the RF chain you set as "TX channel" can actually transmit? Gateways have multiple channels numbered 0,1... but only some of them (one) can transmit.

gotthardp, 2017-03-23:
And thanks for the logs. Yes, the analysis of logs proves that. In the server config -- Gateways you did set *TX channel* to "6", but you need to set this to "0".

lhrossouw, 2017-03-24:
Hi Petr,
I'm embarrassed... sorry for wasting your time.
maybe I should have RTFM a bit.

Gateways

For each LoRaWAN gateway you can set:

    MAC address of the gateway
    TX Channel used for downlinks; usually 0

Only I will know why I set it to 6, and I don't even know. :-)

gotthardp, 2017-03-24:
Don't worry. I am glad we solved the issue.

closed 2017-03-24 by gotthardp