gotthardp/lorawan-server #43: Missining AppID setting in configuration panel

Marcze, 2017-03-31:
Hi

I cannont find AppID field in Device configuration, see my screenshot below

[image]

In your documentation you show an example where AppID is present

[image]

Now I'm not sure how to use websocket:
/groups/ABC/... Connects to all websocket devices whose AppID is set to ABC 
- what to put instead of ABC if AppID is not visible in configuration.

My goal is to receive data from several sensors under single websocket connection. I guess I need to assign the same AppID to all sensors.

gotthardp, 2017-03-31:
Hello. Yes, to receive data from multiple devices via single websocket, you need to setup a group and put its name in "AppID". In your version the "AppID" is named "Arguments". Just pick a names, set app to websocket, set "Arguments" to the name you picked and then use it as *ABC*.

gotthardp, 2017-04-10:
Is everything fine? May I close this issue?

Marcze, 2017-04-10:
Hi Petr,

Yes, I recompiled my version.
Anyway, I'm not very sure what to do with the database after update. Now
I'm just deleting everything but need to place all the settings again that
is troublesome.

Thank you for your great work and support, if you visit Krakow/Poland by a
chance pls let me know, we can have beer together

Kind regards,
Marek

gotthardp, 2017-04-10:
Hi Marek. The database is able to upgrade its structure automatically, without data loss. Also, there are scripts doing `dbexport` and `dbimport` if you need to backup you data.

closed 2017-04-13 by gotthardp