gotthardp/lorawan-server #49: web sockets for Links inop

dlarue, 2017-04-04:
I have 2 clients, the demo(localhost:8080/admin/ws) and a Chrome app but nether one receives anything after connecting and I trigger the device/link to send a message.

The admin/ws demo shows:
4/3/2017 9:27:13 PM CONNECTED
4/3/2017 9:27:13 PM Connecting to: ws://localhost:8080/ws/links/26021B1B/json

Received Frames listing shows:
2017-04-03 21:29:25	18FE34FFFFCE8D6B	26021B1B	websocket	myWebSocket	10	0	1	0303032A4A9B41084CCA42

If I enter an incorrect link device Address the connection doesn't occur so it seems web sockets in the server process is there.

gotthardp, 2017-04-04:
Hi. Did you set Application to WebSocket?

dlarue, 2017-04-04:
yes

dlarue, 2017-04-04:
I created the gateway

created a node, specifying:
DevAddr *   26021B1B
Region * US 902-928MHz (Private)
Application *websocket
NwkSKey * xxxxxxx
AppSKey * xxxxxxx

And see the data showing up in the "Received Frames" listing which looks like this:
 Received	MAC	DevAddr	Application	AppID	SNR	FCnt	Port	Data
2017-04-03 23:10:00	18FE34FFFFCE8D6B	26021B1B	websocket		8	0	1	03030370B397411C69CA42

Using the demo ws page( http://localhost:8080/admin/ws.html )
4/3/2017 11:09:50 PM CONNECTED
4/3/2017 11:09:50 PM Connecting to: ws://localhost:8080/ws/links/26021B1B/json

But I don't see any data/json output on the web socket demo page when I trigger the node to send data.

gotthardp, 2017-04-04:
This is strange. Is there something in the log?

gotthardp, 2017-04-04:
Which version are you using?

dlarue, 2017-04-04:
I didn't see anything in the logs and I'd used the error.log to initially get my node ID correct.

as for versions, I see both 0.2.0 and 0.3.0 in the releases directory.  I don't see what's being run in the admin pages so not sure exactly but I did delete the container and rebuild it to make sure.  Dockerfile follows:

FROM solarisbank/erlang:19.0
MAINTAINER Petr Gotthard <petr.gotthard@centrum.cz>

RUN apk --no-cache add nodejs curl git
RUN curl -o /usr/bin/rebar3 "https://s3.amazonaws.com/rebar3/rebar3" && chmod +x /usr/bin/rebar3
RUN git clone https://github.com/gotthardp/lorawan-server.git && cd lorawan-server && rebar3 release
#RUN git clone https://github.com/dlarue/lorawan-server.git && cd lorawan-server && rebar3 release

RUN apk --no-cache add bash

# data from port_forwarders
EXPOSE 1680/udp
# admin interface
EXPOSE 8080/tcp

WORKDIR /lorawan-server/_build/default/rel/lorawan-server

RUN ifconfig
CMD bin/lorawan-server

dlarue, 2017-04-04:
I just built and installed from scratch on Ubuntu 16.04.2 LTS and still no websocket output.

FWIW, here's my script for installing: (note: I already have some dev tools installed but had no Erlang)
#!/bin/bash
sudo curl -o /usr/bin/rebar3 "https://s3.amazonaws.com/rebar3/rebar3" && sudo chmod +x /usr/bin/rebar3

sudo apt-get install erlang-base erlang-asn1 erlang-public-key erlang-ssl erlang-inets erlang-eunit erlang-dev

git clone https://github.com/gotthardp/lorawan-server.git && cd lorawan-server && rebar3 release
cd lorawan-server/
rebar3 release

cd _build/default/rel/
./bin/lorawan-server

dlarue, 2017-04-05:
When I trigger the node message, I do get this on the console.log and nothing in the other logs. I still see the data for the node/link show up in the "Received Frames" page of the Admin page.

2017-04-04 19:43:25.632 [info] <0.296.0>@lorawan_gw_router:handle_cast:48 Gateway <<24,254,52,255,255,206,141,107>> at {global,lorawan_gw_forwarder} {{192,168,2,110},1680,1}
2017-04-04 19:43:28.757 [warning] <0.6.0> lager_error_logger_h dropped 40 messages in the last second that exceeded the limit of 50 messages/sec

gotthardp, 2017-04-05:
Would you be able to edit the config file and set loging error level to `debug`?

dlarue, 2017-04-05:
unfortunately all I see is the error that it dropped a bunch of messages the moment I make a WebSocket connection. But subsequent WebSocket connections do not throw errors(note, the unknown addr is from another unregistered node in the area of the gateway):

07:08:57.653 [debug] Lager installed handler lager_backend_throttle into lager_event
07:09:04.988 [debug] WebSocket to link <<38,2,27,27>>
07:09:04.991 [warning] lager_error_logger_h dropped 90 messages in the last second that exceeded the limit of 50 messages/sec
07:09:04.991 [debug] Supervisor kernel_safe_sup started pg2:start_link() at pid <0.301.0>
07:09:26.684 [info] Gateway <<24,254,52,255,255,206,141,107>> at {global,lorawan_gw_forwarder} {{192,168,2,110},1680,1}
07:10:13.478 [warning] Unknown DevAddr 26092653
07:10:13.478 [error] ERROR: {unknown_devaddr,<<38,9,38,83>>}
07:10:52.237 [debug] WebSocket to link <<38,2,27,27>>
07:10:54.860 [debug] WebSocket to link <<38,2,27,27>>
07:10:56.201 [debug] WebSocket to link <<38,2,27,27>>

gotthardp, 2017-04-05:
would you mind trying to set AppID to something (e.g. "test") and the connecting to that group via ws.html?

dlarue, 2017-04-05:
ok, will try that.

dlarue, 2017-04-05:
FYI, with debug enabled in sys.config I was surprised to not see the node message in the console.log even though it shows up on the Received Frames page.  Is there a way to see the node/link messages passing into the server?

gotthardp, 2017-04-05:
All messages coming in are listed on the "Received Frames" page, They are not in the log.file if everything goes well.

dlarue, 2017-04-05:
ok, I was hoping the system worked in such a way as to the messages shown in "Received Frames" page had the meaning that they were valid and checked. Is there any chance an Erlang subsystem wasn't install and is causing the websocket handler to error out on load? Or a way to slow down the errors so the error handler doesn't hide/drop all the errors?

gotthardp, 2017-04-05:
Oh yes. What is in the "received frames" is checked and valid. Could there be some proxy in the way?

dlarue, 2017-04-05:
I'm running it all on the local machine(localhost:8080).  Does websocket require another port?  I'll turn off the firewall just to see but I don't think anything else is in the way.

dlarue, 2017-04-05:
I decided to just reboot the system since I was running some docker stuff with socat redirections(but had killed them) and other things. Now the links page will not load and I don't see messages coming through the server. ugh.

dlarue, 2017-04-05:
FWIW, I deleted by build and rebuilt to look at the build output. I noticed a couple of warning about coffee-script and jsonlint and thought they might apply to the websocket output. So I installed jsonlint(apt-get) and the ran 'sudo npm -g install jsonlint coffee-script'. then I removed an rebuilt lorawan-server but still saw the warnings:

npm WARN prefer global coffee-script@1.10.0 should be installed with -g
npm WARN prefer global jsonlint@1.6.2 should be installed with -g

gotthardp, 2017-04-06:
I added some more error handling. Could you download the latest 'master' and try again. It will at least rule out what is not the cause of this.

dlarue, 2017-04-06:
what changed because I don't see any more debug info and I enabled the console debug line in sys.config.  I still just get this as soon as I connect a ws client.

09:48:26.598 [debug] WebSocket to link <<38,2,27,27>>
09:48:26.600 [warning] lager_error_logger_h dropped 90 messages in the last second that exceeded the limit of 50 messages/sec
09:48:26.601 [debug] Supervisor kernel_safe_sup started pg2:start_link() at pid <0.326.0>

dlarue, 2017-04-06:
Is there anything I'm missing? I created a gateway and created a node since I'm not using OTAA. I've entered data in the require fields(marked *) and I've also tried using the groupID/test and also added a gateway filter(it provided the ID and it matches).

I get messages in the Recieved Frames page.

dlarue, 2017-04-06:
[image]

dlarue, 2017-04-06:
It really looks like the error occurs when the first ws connection is made and it triggers dozens of errors but they are so fast that the High Water Mark(hwm) blocks them from even being shown.  This seems to be where we need to look and somehow disable the hwm from blocking the errors.

This is what happens every time I restart the server and make the first web socket client connection:

10:17:38.918 [debug] WebSocket to link <<38,2,27,27>>
10:17:38.920 [warning] lager_error_logger_h dropped 41 messages in the last second that exceeded the limit of 50 messages/sec
10:17:38.920 [debug] Supervisor kernel_safe_sup started pg2:start_link() at pid <0.311.0>

gotthardp, 2017-04-06:
The hwm can be disabled by inserting `{error_logger_hwm, undefined}` into the `lager` section of config.

dlarue, 2017-04-06:
that just eliminated the mention of repeating errors and does not show what those errors are.  So now I see nothing happen when a ws client connects for the first time...

gotthardp, 2017-04-10:
This was because the device kept sending FCnt=0, which was treated as retransmissions even when the FCnt Check was disabled.

Fixed by commit 495461906bd63f582cd4b54b15ce6d66ab9c3128 and previous. The retransmissions are now ignored when FCnt Check is Disabled.

closed 2017-04-10 by gotthardp