gotthardp/lorawan-server #52: OTAA

Kralg, 2017-04-06:
I am learning about handlers and I would like to ask how is it possible to implement OTAA joining. I have tried ABP, in this case I created a link and I can see the frames from my device in /admin. But if I try to join the network (and create corresponding device) it seems that gateway do not respond at all to the join requests (Application:websocket, Can join:true I can see in the dashboard that device got Link number and that Last join time is right now and updates, but in WiMod LoraWan endnode studio (for wimode lorawand end device configuration) I can see Error: device not activated and do not receive the address, on SDR I do not see no response on joining RF packets). 

Is this a bug or is there any way to automatically run OTAA, or do I have to implement my handler for this? Maybe somebody has already done this before?

gotthardp, 2017-04-06:
Hello. This shall work. This is probably because your device does not hear the response from the server. Which ISM band are you using? Does your gateway log indicate there was a downlink received from the server and correctly transmitted to the device?

Kralg, 2017-04-10:
Hi, in the gateway log (packet forwarding) I can see that once join request is sent I am getting on of two messages either this one (in this case in lorawan-server log I do not see no Join Request messages):
##### 2017-04-07 15:25:19 GMT #####
### [UPSTREAM] ###
# RF packets received by concentrator: 2
# CRC_OK: 100.00%, CRC_FAIL: 0.00%, NO_CRC: 0.00%
# RF packets forwarded: 2 (46 bytes)
# PUSH_DATA datagrams sent: 2 (408 bytes)
# PUSH_DATA acknowledged: 0.00%
### [DOWNSTREAM] ###
# PULL_DATA sent: 3 (0.00% acknowledged)
# PULL_RESP(onse) datagrams received: 0 (0 bytes)
# RF packets sent to concentrator: 0 (0 bytes)
# TX errors: 0
# BEACON queued: 0
# BEACON sent so far: 0
# BEACON rejected: 0
### [JIT] ###
lgw_spi_rb:371: BURST READ: to trans 4 # chunk 4 # transferred 4 
Note: SPI burst read success
# SX1301 time (PPS): 2431202
src/jitqueue.c:448:jit_print_queue(): INFO: [jit] queue is empty
### [GPS] ###
# GPS sync is disabled
##### END #####
lgw_spi_rb:371: BURST READ: to trans 5 # chunk 5 # transferred 5 
Note: SPI burst read success
JSON up: {"stat":{"time":"2017-04-07 15:25:19 GMT","rxnb":2,"rxok":2,"rxfw":2,"ackr":0.0,"dwnb":0,"txnb":0}}

Or message (In this case I can see that [INFO ]Join Request was received in lorawan-server log):

lgw_receive:1155: FIFO content: 1 10 0 5 17
lgw_spi_rb:371: BURST READ: to trans 39 # chunk 39 # transferred 39 
Note: SPI burst read success
lgw_receive:1174: [1 17]
Note: LoRa packet
Note: SPI write success
lgw_spi_rb:371: BURST READ: to trans 5 # chunk 5 # transferred 5 
Note: SPI burst read success

INFO: Received pkt from mote: EEFF1122 (fcnt=48076)

JSON up: {"rxpk":[{"tmst":5642947,"chan":1,"rfch":1,"freq":868.300000,"stat":1,"modu":"LORA","datr":"SF7BW125","codr":"4/5","lsnr":9.5,"rssi":-39,"size":23,"data":"ACIR/+7dzLuqkDMD8I/Vs3Aq2HwBVII="}]}

The end device is set to EU 868MHz SF7 / 125000Hz / 5470bps, ADR, C class support, Automatic power saving.
[lorawan_gateway_local_configurations.zip](https://github.com/gotthardp/lorawan-server/files/909578/lorawan_gateway_local_configurations.zip)

I am attaching my global_configuration files that I am using. I have tried both ttn-gateway install and packet-forwarding packets.

gotthardp, 2017-04-10:
If you are seeing this, then the gateway didn't receive anything. Would you be able to use wireshark.org and check the server-to-gateway communication?

The gateway is strangely changing ports where it is listening for downlinks (why? could it be the power saving?). Perhaps the server is sending the downlink to a port were nobody is listening.

Kralg, 2017-04-11:
I have not used wireshark.org before. If you have any good example on how to set it up to capture local server it would be very helpful. 
I also noticed that there is one more erl error. 
07:37:46.749 [info] Gateway <<184,39,235,255,254,90,18,52>> at {127,0,0,1}:41356
07:38:30.803 [info] JOIN REQUEST <<112,179,213,126,240,0,65,151>> <<112,179,213,143,240,3,51,144>> -> <<84,248,2,200>>
07:38:37.068 [info] JOIN REQUEST <<112,179,213,126,240,0,65,151>> <<112,179,213,143,240,3,51,144>> -> <<84,248,2,200>>
07:38:43.377 [info] JOIN REQUEST <<112,179,213,126,240,0,65,151>> <<112,179,213,143,240,3,51,144>> -> <<84,248,2,200>>
07:38:43.377 [error] Error in process <0.373.0> on node lorawan@raspberry with exit value:
{{badmatch,<<145,43,100,54,27,16,183,60,187,179,197,174,28,149,230,69,231,199>>},[{lorawan_mac,process_frame1,5,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac.erl"},{line,75}]},{lorawan_handler,rxpk,4,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_handler.erl"},{line,16}]}]}
07:39:09.177 [info] JOIN REQUEST <<112,179,213,126,240,0,65,151>> <<112,179,213,143,240,3,51,144>> -> <<84,248,2,200>>
07:39:31.606 [error] Error in process <0.375.0> on node lorawan@raspberry with exit value:
{{badmatch,<<0,107,134,211,37,159,15,167,75,131,238,150>>},[{lorawan_mac,process_frame1,5,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac.erl"},{line,75}]},{lorawan_handler,rxpk,4,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_handler.erl"},{line,16}]}]}
07:39:35.082 [info] JOIN REQUEST <<112,179,213,126,240,0,65,151>> <<112,179,213,143,240,3,51,144>> -> <<84,248,2,200>>
07:39:36.809 [warning] Unknown DevAddr 59005F71
07:39:36.809 [error] ERROR: {unknown_devaddr,<<89,0,95,113>>}
07:40:08.781 [warning] Unknown DevEUI 70B3D54330033F90
07:40:08.781 [error] ERROR: {unknown_deveui,<<112,179,213,67,48,3,63,144>>}
07:40:08.781 [info] JOIN REQUEST <<112,179,213,126,240,0,65,151>> <<112,179,213,143,240,3,51,144>> -> <<84,248,2,200>>
07:41:02.134 [error] ERROR: bad_mic
07:41:02.136 [info] JOIN REQUEST <<112,179,213,126,240,0,65,151>> <<112,179,213,143,240,3,51,144>> -> <<84,248,2,200>>

gotthardp, 2017-04-11:
If you connect to the server on 127.0.0.1, could you try `sudo tcpdump -i lo -w file.pcap` and send me the `file.pcap`? If there is no tcpdump you should `sudo apt-get install tcpdump`.

Kralg, 2017-04-11:
Well, I think it worked. I opened the file with wireshark and there are messages of Destination unreachable, probably this message corresponds to Source port: 1680 Destination port: 35088.
[file.pcap.txt](https://github.com/gotthardp/lorawan-server/files/912882/file.pcap.txt)

gotthardp, 2017-04-11:
Thank you for that. The log shows ICMP "port unreachable" everytime the server sends something back to the gateway. This looks like a networking issue beyond the lorawan-server.

Kralg, 2017-04-11:
Thanks for the information. I was looking around on the web for similar issues. I found some, but not a clear solution for this. Would you have any insights on how this could be solved?

gotthardp, 2017-04-11:
I guess you run TTN against different packet_forwarder than this server. Could you try running the server against the same packet_fwd and gateway which works with TTN? You will have to change the server IP address and port, but don't change anything else.

Kralg, 2017-04-12:
Yes, I am using ttn-zh/ic880a-gateway  I have tried to test it wiht ttn and it seems to be working with OTAA. I changed to my server lorawan@raspberry and it doesn't seem to be working. However I don't see ICMP "port unreachable". I am attaching the configurations and log files.
[file.pcap.txt](https://github.com/gotthardp/lorawan-server/files/916237/file.pcap.txt)
[global_conf.json.txt](https://github.com/gotthardp/lorawan-server/files/916235/global_conf.json.txt)
[local_conf.json.txt](https://github.com/gotthardp/lorawan-server/files/916234/local_conf.json.txt)

[sys.config.txt](https://github.com/gotthardp/lorawan-server/files/916236/sys.config.txt)

gotthardp, 2017-04-12:
This is the same file.pcap as before. Could you please collect a large PCAP file over a longer period of time, capturing more than one frame?

altishchenko, 2017-04-12:
It bothers me Petr, that the only UDP communication we are seeing is: pkt received, reply sent, icmp unreachable received. Otherwise, there is some other TCP session in there. As if the device responsible to get the answers is NOT listening on the port it sent the message from. Do they treat serv_down_port as in FTP?! Meaning you must _reply_ _to_ the down port number (not from)? (Ex, you are getting a message from 33598 to 1680, and on that you must reply to 1681 (whatever) of the sending station?!). Otherwise, you’ll of course get the dest unreach over icmp… ?!

Alexander Tishchenko
mob: +7 (916) 704-61-57
atishch@gmail.com

gotthardp, 2017-04-12:
It works for all other gateways the way it is done now. The spec says the PULL_DATA and PULL_ACK is there to maintain NAT open, hence I assume the PULL_RESP is sent the same way as PULL_ACK and both are using the origination port.

altishchenko, 2017-04-12:
Ok, I see. I am not that much used to traditional gateways, but it does look suspicious..

Alexander Tishchenko
mob: +7 (916) 704-61-57
atishch@gmail.com

Kralg, 2017-04-13:
I just updated the previous file. ICMP messages are there too.

gotthardp, 2017-04-13:
this feels like a gateway or a protocol issue. could you find out on which ports the gateway listens while running? (try `netstat -ulp`)

Kralg, 2017-04-13:
This is the netstat result:
[netstat.txt](https://github.com/gotthardp/lorawan-server/files/919296/netstat.txt)

altishchenko, 2017-04-13:
I am sorry to barge in, but given that all of this runs on the same box, with localhost being the middle man, I really can’t see anybody waiting for a response in the log provided. It does look like a gateway is sending forward a message to the server, but forgets about the UDP port it sent it from straight after the transmission, that is why you are getting all those ICMPs - nobody is listening on that port. From your printout I can see only one port that might be interested in the server’s replies - 45437/UDP, but there is no momentary tcpdump, to see that the reply was destined to that particular port.

Alexander Tishchenko
mob: +7 (916) 704-61-57
atishch@gmail.com

dlarue, 2017-04-14:
Not sure if it relates, I have a MultiTech MTAC gateway card running packet_forwarder to this lorawan-server and a ProMini Node running arduino_lmic and it works in both single channel and full channel modes. Which validates the node/Arduino code.  BTW, it always joins on US freq 65 which is the default dn2Chnl.  But when I swap in a single channel Lua based gateway it never joins if I use lorawan-server but will join and work fine going to TTN.

gotthardp, 2017-04-14:
@dlarue would you be able to provide me with gateway logs that could help me identifying the issue? Everything has to work just as fine as with TTN. Could it be the hardcoded port issue?

Kralg, 2017-04-14:
Would you be interested to test the server if I will send you my card image?

gotthardp, 2017-04-14:
I am becoming really desperate here. Would you be able to connect it back to TTN, make the .pcap file showing the correct flow and send me that?

altishchenko, 2017-04-14:
Copy that. Real session may reveal interesting details.
 
Alexander Tishchenko
mob: +7 (916) 704-61-57
atishch@gmail.com

Kralg, 2017-04-15:
Hi, I am attaching two files: listening on eth0 with TTN and lo with lorawan-server. I have installed everything once again. And tried to install only erlang, ic88a gateway, erlang http://elinux.org/Erlang and lorwan-server.

[file.pcap.txt](https://github.com/gotthardp/lorawan-server/files/923500/file.pcap.txt)
[file_ttn.pcap.txt](https://github.com/gotthardp/lorawan-server/files/923501/file_ttn.pcap.txt)

gotthardp, 2017-04-15:
One more thing: Could you please start the gateway, wait 10 seconds, then do `netstat -uapn` and post here the output?

Kralg, 2017-04-15:
I am attaching the file with three cases.
[netstat.txt](https://github.com/gotthardp/lorawan-server/files/923521/netstat.txt)

gotthardp, 2017-04-15:
I may have found the issue: the gateway sends the request from 127.0.0.1 to 127.0.1.1 (note the difference!) However the response is sent from 127.0.0.1 to 127.0.0.1 (note there is no difference).
Erlang is apparently confused by the Debian's address https://www.debian.org/doc/manuals/debian-reference/ch05.en.html#_the_hostname_resolution

Kralg, 2017-04-15:
That is a really good point! I have read the paragraph, but don't really know what should be done. Do you see any solution for this?

gotthardp, 2017-04-15:
Could you try configuring the gateway to send the frames to "localhost" or "127.0.0.1" instead of "raspberry"?

Kralg, 2017-04-15:
Here, we go. 127.0.0.1 it works. :)

[file_IP.pcap.txt](https://github.com/gotthardp/lorawan-server/files/923557/file_IP.pcap.txt)
[file_localhost.pcap.txt](https://github.com/gotthardp/lorawan-server/files/923558/file_localhost.pcap.txt)

gotthardp, 2017-04-15:
Cool. I will write some documentation to warn others.

closed 2017-04-15 by gotthardp