gotthardp/lorawan-server #53: Hanler Terminated

dlarue, 2017-04-06:
While watching the console regarding the websocket issue, I saw this come though the console and it looks like something triggered a failure of a handler:

12:56:33.857 [error] Error in process <0.559.0> on node lorawan@kubuntu with exit value:
{{badmatch,<<94,113,117,104,221,162,128,13,175,45,115,180,233,39,105,229,175>>},[{lorawan_mac,process_frame1,5,[{file,"/home/dlarue/mnt/sda5/home/dlarue/Projects/FabLab/LoRaWAN/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac.erl"},{line,88}]},{lorawan_gw_router,process_frame,3,[{file,"/home/dlarue/mnt/sda5/home/dlarue/Projects/FabLab/LoRaWAN/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_gw_router.erl"},{line,136}]}]}
12:56:33.857 [error] Hanler terminated: {{badmatch,<<94,113,117,104,221,162,128,13,175,45,115,180,233,39,105,229,175>>},[{lorawan_mac,process_frame1,5,[{file,[47,104,111,109,101,47,100,108,97,114,117,101,47,109,110,116,47,115,100,97,53,47,104,111,109,101,47,100,108,97,114,117,101,47,80,114,111,106,101,99,116,115,47,70,97,98,76,97,98,47,76,111,82,97,87,65,78,47,108,111,114,97,119,97,110,45,115,101,114,118,101,114,47,95,98,117,105,108,100,47,100,101,102,97,117,108,116,47,108,105,98,47,108,111,114,97,119,97,110,95,115,101,114,118,101,114,47,115,114,99,47,108,111,114,97,119,97,110,95,109,97,99,46,101,114,108]},{line,88}]},{lorawan_gw_router,process_frame,3,[{file,[47,104,111,109,101,47,100,108,97,114,117,101,47,109,110,116,47,115,100,97,53,47,104,111,109,101,47,100,108,97,114,117,101,47,80,114,111,106,101,99,116,115,47,70,97,98,76,97,98,47,76,111,82,97,87,65,78,47,108,111,114,97,119,97,110,45,115,101,114,118,101,114,47,95,98,117,105,108,100,47,100,101,102,97,117,108,116,47,108,105,98,47,108,111,114,97,119,97,110,95,115,101,114,118,101,114,47,115,114,99,47,108,111,114,97,119,97,110,95,103,119,95,114,111,117,116,101,114,46,101,114,108]},{line,136}]}]}

gotthardp, 2017-04-06:
It was an errorneous frame that caused this. Perhaps the message shouldn't be so scary, but in this case everything works as expected.

dlarue, 2017-04-06:
Ok, when I saw that it said it terminated a process I became concerned. Closing it. Thanks.

closed 2017-04-06 by dlarue