gotthardp/lorawan-server #54: websocket_timeout still occurs with message transmissions

dlarue, 2017-04-11:
Using Node setup( ABP ) and websocket application the lorawan-server still closes the websocket connection after the amount of time specified in the sys.config file. In my case, 1 hour. The admin/ws.html client gets disconnected while the Google Chrome app WebSocket Client just stops getting JSON messages.

sys.config entry:
    % websocket expiration if client sends no data
    {websocket_timeout, 3600000} % ms

gotthardp, 2017-04-11:
This works as described here: https://github.com/gotthardp/lorawan-server/blob/master/doc/WebSockets.md#keep-alive
If you want to be never disconnected, use `infinity`.

dlarue, 2017-04-11:
I'm saying it does not work as described: "if the client sends no data for 1 hour"

Data is being sent every 15 minutes but it still closes the socket 1 hour of the websocket is opened. ie 4 messages get through and then it stops.

batulzii, 2017-04-11:
yes. it is. i have the same issue.

gotthardp, 2017-04-11:
But in your case it's the *server* sending the data, not the *client*. The client (e.g. the browser) does not send anything. Am I right?

dlarue, 2017-04-11:
correct, the LoraWAN node is sending data to the server and the websocket client is displaying it. Node sends a message every 15 minutes.  So the timeout is triggered if the websocket client doesn't sent any data back to the server?

gotthardp, 2017-04-11:
If you don't like keep alives you can always use `infinity`, i.e. set `websocket_timeout, infinity}`. The documentation is poor I know, but it's getting better. ;-)

dlarue, 2017-04-11:
I've never used websockets( just tcp/ip sockets ) so I'm not sure how reliable they are and maybe keepalive makes sense and is needed.  But in more controlled and hands-on usage like during long range weather balloon tracking, infinity setting would be fine.

As for the documentation, I've been getting only about 4 hours of sleep the last few night so most of the problem is probably in the seat I'm sitting on. ;)

gotthardp, 2017-04-11:
I will update the documentation to make this more explicit, but since I also think it is useful I don't want to disable the expiration by default.

closed 2017-04-11 by gotthardp