gotthardp/lorawan-server #55: websocket - timestamp

Marcze, 2017-04-12:
It would be beneficial to add timestamp into json structure trasmitted by websockets to clients.
The best if the field is copied from packet_forwarder structure (precision time when rx radio frame is received by a gateway).

gotthardp, 2017-04-12:
Agreed.

gotthardp, 2017-04-13:
Well, I believe the latest 'master' branch already includes this timestamp (in the "rxq" sub-structure). Cpuld you confirm? Please note it is included only when the GPS signal is available.

Marcze, 2017-04-14:
Unfortunately, I don't have GPS in my gateway, moreover it is located
inside a building, can have troubles to get gps fix.

gotthardp, 2017-04-14:
Without GSP you get only the gateway internal time. For what purpose do you need this?

Marcze, 2017-04-14:
You can add the best time you have, gps or if not available gateway time.
My gateway time is synched with network time. I need to have measurements
sychronised. With Gps time you can try to do localisation, when more
gateways are available. I put jsons to my application database at different
server, now I add time at database level that is less reliable - network
latency.

gotthardp, 2017-04-17:
Done. Look for `datetime`.

closed 2017-04-17 by gotthardp

Marcze, 2017-05-10:
Hi Peter,

Have troubles to compile sources. See photo below. Pls advise.

Marek

[image: Inline image 1]

gotthardp, 2017-05-10:
I don't see the photo, but:
 * do you have Erlang 19+? It no longer compiles with Erlang 18. (I have to respect a decision taken by the cowboy server developer.)
 * did you do `make upgrade` before compiling?

Marcze, 2017-05-12:
Hi, You're right Erlang 19. Best regards, Marek