gotthardp/lorawan-server #58: badarith using links

orotemo, 2017-04-19:
new links work fine! however, after some time (several days that all is off), maybe related to `fcnt` mismatches, my packets hit a code path that leads to:
`[debug] <0.354.0>@lorawan_mac_commands:auto_adr0:180 Power <<38,1,31,64>>: average rssi -143, power <<117,110,100,101,102,105,110,101,100>> -> down by -22`

which leads to `badarith`:

```log
 [error] <0.291.0>@lorawan_gw_router:handle_info:101 Hanler terminated: {badarith,[{lorawan_mac_commands,auto_adr0,2,[{file,[47,104,111,109,101,47,112,105,47,119,111,114,107,47,108,111,114,97,119,97,110,45,115,101,114,118,101,114,47,95,98,117,105,108,100,47,100,101,102,97,117,108,116,47,108,105,98,47,108,111,114,97,119,97,110,95,115,101,114,118,101,114,47,115,114,99,47,108,111,114,97,119,97,110,95,109,97,99,95,99,111,109,109,97,110,100,115,46,101,114,108]},{line,182}]},{lorawan_mac_commands,handle,4,[{file,[47,104,111,109,101,47,112,105,47,119,111,114,107,47,108,111,114,97,119,97,110,45,115,101,114,118,101,114,47,95,98,117,105,108,100,47,100,101,102,97,117,108,116,47,108,105,98,47,108,111,114,97,119,97,110,95,115,101,114,118,101,114,47,115,114,99,47,108,111,114,97,119,97,110,95,109,97,99,95,99,111,109,109,97,110,100,115,46,101,114,108]},{line,22}]},{lorawan_mac,handle_uplink,5,[{file,[47,104,111,109,101,47,112,105,47,119,111,114,107,47,108,111,114,97,119,97,110,45,115,101,114,118,101,114,47,95,98,117,105,108,100,47,100,101,102,97,117,108,116,47,108,105,98,47,108,111,114,97,119,97,110,95,115,101,114,118,101,114,47,115,114,99,47,108,111,114,97,119,97,110,95,109,97,99,46,101,114,108]},{line,317}]},{lorawan_gw_router,process_frame,3,[{file,[47,104,111,109,101,47,112,105,47,119,111,114,107,47,108,111,114,97,119,97,110,45,115,101,114,118,101,114,47,95,98,117,105,108,100,47,100,101,102,97,117,108,116,47,108,105,98,47,108,111,114,97,119,97,110,95,115,101,114,118,101,114,47,115,114,99,47,108,111,114,97,119,97,110,95,103,119,95,114,111,117,116,101,114,46,101,114,108]},{line,141}]}]}
```

this is using master on both `lorawan-server` and `lorawan-server-api`.

WDYT?

B.T.W.: this projects stack is 100% what I imagined to be the best possible. thanks so much for all the hard work you put in here!

gotthardp, 2017-04-19:
This is due to the power being set to `<<117,110,100,101,102,105,110,101,100>>`, which is `<<"undefined">>`. This is probably caused by a wrong conversion to/from the web admin page. I made some updates to avoid that in the future, but you may need to restart your device to fix the database inconsistencies you already have.

gotthardp, 2017-04-19:
BTW, thank you for reporting this and thank you also for your interest in this project. :-)

orotemo, 2017-04-19:
All the thanks goes to you! Beyond doubt.

closed 2017-04-22 by gotthardp