gotthardp/lorawan-server #60: Access to RXQ data and what Gateway send the data

matiasz8, 2017-04-19:
Hi gotthardp, thanks for the documentation,

I'm follow the demo that you give us.

At the moment, I run the lorawan-server in linux.

In the Server Admin I create a new gateway and a new link (in this case I use a mote for microchips).

I start up the internal WebSocket (ws.html) located in the url: "http://localhost:8080/admin/ws.html".

When you press "Connect", built the following url: ws://localhost:8080/ws/groups/Test/json.

Following the dots, we understood that the clients receive the fields: (devaddr, appargs, port, fcnt, data, fields, shall_reply, last_lost, datetime, rxq) //_https://github.com/gotthardp/lorawan-server/blob/master/doc/WebSockets.md_

The questions focus in this point:

In the WebSocket as Json Format, we receive: a timestamp (from the server) and Data, that contains the fields: devaddr, port and data (related to the data sended).

1. How can I add to this answer the rxq data to get the rssi field for example?
    Is there any way to change which field receive?

2. And another question, Can we receive information on which gateway transmit to the WebSocket Client?
	
Regards,
Nahuel Quiroga
nahuelq.78@gmail.com

gotthardp, 2017-04-19:
Hello. Are you using the 0.3.0 release or older? The rxq params are alredy sent by the most recent 0.4.0 (unreleased master at the moment).
Gateway mac can be added.

matiasz8, 2017-04-19:
Thanks for answer, I'm using the 0.3.0 release. But I can use the 0.4.0 for tests, I'll still be at work on this version then.
Is the logic in the json / raw format programmed in Erlang? I mean, is there a way to modify this data ourselves?

gotthardp, 2017-04-19:
What kind of data "modification" would you need?

matiasz8, 2017-04-19:
For example, add fields like rssi or instead of receive the port, receive the rssi.
At the moment, I need the gateway mac and the rssi. But if in another case you need to modify any field you receive, is there any file that can be accessed to change this?

gotthardp, 2017-04-19:
Right now the server sends you everything and you can pick what you want. If I add RSSI this should meet you needs. Or you really need for some reason to hide the other fields?

matiasz8, 2017-04-20:
This is in the 0.4.0 version, really? I'll try with this version then. Trying to start up the lorawan-server but I must have something I'm not doing correctly.
I run the lorawan-server whitin scipts, but show me this: 

$ scripts/lorawan-server
{error_logger,{{2017,4,20},{11,24,56}},"Error in process ~p with exit value:~n~p~n",[<0.7.0>,{{case_clause,{'EXIT',{function_clause,[{application_controller,'-check_conf/0-fun-0-',[[],[]],[{file,"application_controller.erl"},{line,1809}]},{lists,foldl,3,[{file,"lists.erl"},{line,1262}]},{application_controller,check_conf,0,[{file,"application_controller.erl"},{line,1808}]},{application_controller,init,2,[{file,"application_controller.erl"},{line,485}]}]}}},[{application_controller,init,2,[{file,"application_controller.erl"},{line,485}]}]}]}
{"could not start kernel pid",application_controller,"{{case_clause,{'EXIT',{function_clause,[{application_controller,'-check_conf/0-fun-0-',[[],[]],[{file,\"application_controller.erl\"},{line,1809}]},{lists,foldl,3,[{file,\"lists.erl\"},{line,1262}]},{application_controller,check_conf,0,[{file,\"application_controller.erl\"},{line,1808}]},{application_controller,init,2,[{file,\"application_controller.erl\"},{line,485}]}]}}},[{application_controller,init,2,[{file,\"application_controller.erl\"},{line,485}]}]}"}

Crash dump is being written to: erl_crash.dump...could not start kernel pid (application_controller) ({{case_clause,{'EXIT',{function_clause,[{application_controller,'-check_conf/0-fun-0-',[[],[]],[{file,"application_controller.erl"},{line,1809}]}

gotthardp, 2017-04-20:
Oh, you should always start it using the scripts in the release directory. The scripts includes just templates.
```bash
cd lorawan-server/_build/default/rel/lorawan-server
bin/lorawan-server
```

gotthardp, 2017-04-22:
Did it help?

BTW, I wonder why do you need to see the gateway MAC? Could you explain your thinking, please?

matiasz8, 2017-04-24:
Sorry, For example. If you have more than one gateway you know which gateway receive the data. Basically that. 
I try to run the master code, but I do not find the folder "_build ".
I detail you my steps:
Download the lorawan-server-master code. 
Unzip. 
Config the lorawan_server.config
Then I try to find the lorawan-server to run

kazi93, 2017-04-29:
@matiasz8 after config you need to make the source code that will create "_build" inside the lorawan-server directory, you need to run the "$ sudo ./lorawan-server" there.

kazi93, 2017-04-29:
For example: 
mac:~ kazi$ cd lorawan-server/_build/default/rel/lorawan-server/bin/
mac:bin kazi$ ls
lorawan-daemon		lorawan-server		lorawan-server.bat	lorawan-server.service	lorawan-service.bat	lorawanctl		start_clean.boot
mac:bin kazi$ ./lorawan-server

gotthardp, 2017-05-10:
For now I recommend using the pre-compiled `lorawan-server-0.4.0.tar.gz` that I just released. Please let me know whether I may close this issue.

The enhanced JSON is planned for the release 0.5.0 and will be tracked by issue #86.

gotthardp, 2017-06-10:
I just implemented the JSON updates; will be in the future release 0.4.5.

closed 2017-06-10 by gotthardp