gotthardp/lorawan-server #61: Deletion of entries does not work.

hpc32, 2017-04-19:
Do not delete one, more or all of the entries in the Received Frames. See the attached file with an example. The delete function does not work only in this section, the rest is OK. Is it just me or a common problem?

[image]

gotthardp, 2017-04-19:
Nice animation. :) The rxframes are declared as read only, so the Delete button should be hidden (but it isn't). Why do you want to delete the frames?

hpc32, 2017-04-20:
I thought that if there is a button, it should work. ;-) Removal could only be required for those frames that are not needed here, for example, erroneous or received for debugging, etc. But this is not necessary.
Thanks.

gotthardp, 2017-04-22:
Oh yes, if there is a button, it should work. This is definitely a bug. But I would remove the button from there. The rxframes are used to draw the various graphs, so removing the entries could lead to surprising results.

I made the change, so you should no longer be able to see that button.

closed 2017-04-22 by gotthardp