gotthardp/lorawan-server #62: Installation issue on MacOSX Sierra!! Any help!!

kazi93, 2017-04-23:
mac:lorawan-server-0.3.0 kazi$ bin/lorawan-server 

=PROGRESS REPORT==== 23-Apr-2017::02:30:43 ===
          supervisor: {local,sasl_safe_sup}
             started: [{pid,<0.65.0>},
                       {id,alarm_handler},
                       {mfargs,{alarm_handler,start_link,[]}},
                       {restart_type,permanent},
                       {shutdown,2000},
                       {child_type,worker}]

=PROGRESS REPORT==== 23-Apr-2017::02:30:43 ===
          supervisor: {local,sasl_sup}
             started: [{pid,<0.64.0>},
                       {id,sasl_safe_sup},
                       {mfargs,
                           {supervisor,start_link,
                               [{local,sasl_safe_sup},sasl,safe]}},
                       {restart_type,permanent},
                       {shutdown,infinity},
                       {child_type,supervisor}]

=PROGRESS REPORT==== 23-Apr-2017::02:30:43 ===
          supervisor: {local,sasl_sup}
             started: [{pid,<0.66.0>},
                       {id,release_handler},
                       {mfargs,{release_handler,start_link,[]}},
                       {restart_type,permanent},
                       {shutdown,2000},
                       {child_type,worker}]

=PROGRESS REPORT==== 23-Apr-2017::02:30:43 ===
         application: sasl
          started_at: lorawan@mac

=PROGRESS REPORT==== 23-Apr-2017::02:30:43 ===
         application: syntax_tools
          started_at: lorawan@mac

=PROGRESS REPORT==== 23-Apr-2017::02:30:43 ===
         application: compiler
          started_at: lorawan@mac

=PROGRESS REPORT==== 23-Apr-2017::02:30:43 ===
          supervisor: {local,gr_sup}
             started: [{pid,<0.74.0>},
                       {id,gr_counter_sup},
                       {mfargs,{gr_counter_sup,start_link,[]}},
                       {restart_type,permanent},
                       {shutdown,5000},
                       {child_type,supervisor}]

=PROGRESS REPORT==== 23-Apr-2017::02:30:43 ===
          supervisor: {local,gr_sup}
             started: [{pid,<0.75.0>},
                       {id,gr_param_sup},
                       {mfargs,{gr_param_sup,start_link,[]}},
                       {restart_type,permanent},
                       {shutdown,5000},
                       {child_type,supervisor}]

=PROGRESS REPORT==== 23-Apr-2017::02:30:43 ===
          supervisor: {local,gr_sup}
             started: [{pid,<0.76.0>},
                       {id,gr_manager_sup},
                       {mfargs,{gr_manager_sup,start_link,[]}},
                       {restart_type,permanent},
                       {shutdown,5000},
                       {child_type,supervisor}]

=PROGRESS REPORT==== 23-Apr-2017::02:30:43 ===
         application: goldrush
          started_at: lorawan@mac

=PROGRESS REPORT==== 23-Apr-2017::02:30:43 ===
          supervisor: {local,lager_sup}
             started: [{pid,<0.81.0>},
                       {id,lager},
                       {mfargs,{gen_event,start_link,[{local,lager_event}]}},
                       {restart_type,permanent},
                       {shutdown,5000},
                       {child_type,worker}]

=PROGRESS REPORT==== 23-Apr-2017::02:30:43 ===
          supervisor: {local,lager_sup}
             started: [{pid,<0.82.0>},
                       {id,lager_handler_watcher_sup},
                       {mfargs,{lager_handler_watcher_sup,start_link,[]}},
                       {restart_type,permanent},
                       {shutdown,5000},
                       {child_type,supervisor}]

=PROGRESS REPORT==== 23-Apr-2017::02:30:43 ===
          supervisor: {local,lager_sup}
             started: [{pid,<0.83.0>},
                       {id,lager_crash_log},
                       {mfargs,
                           {lager_crash_log,start_link,
                               ["log/crash.log",65536,10485760,[{hour,0}],5]}},
                       {restart_type,permanent},
                       {shutdown,5000},
                       {child_type,worker}]
{"init terminating in do_boot",{{badmatch,{error,{crypto,{"no such file or directory","crypto.app"}}}},[{lorawan_app,start,0,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_app.erl"},{line,16}]},{init,start_em,1,[]},{init,do_boot,3,[]}]}}
init terminating in do_boot ()

Crash dump is being written to: erl_crash.dump...done

kazi93, 2017-04-23:
By the way, I have installed erlang (Sierra - 10.12) from the below source : 
http://rudix.org/packages/erlang.html

gotthardp, 2017-04-23:
There is an erlang module named 'crypto' that is missing or cannot be found on your system. Could you please check that it was included in the package and is installed. It is usually in /usr/lib/erlang/lib or so.

gotthardp, 2017-04-23:
Would the Erlang package for Mac OS X from https://www.erlang-solutions.com/resources/download.html work for you?

altishchenko, 2017-04-23:
The log output means that you are not on the late master branch, as progress reports (apart from _errors_) were quieted there. And I’d strongly recommend trying erlang-solutions. I am doing my staff on El Capitan with ESL-19.3.1. No problem at all.

Cheers,
Alexander Tishchenko
mob: +7 (916) 704-61-57
atishch@gmail.com

kazi93, 2017-04-23:
Hi My friends! Installing  erlang-solutions solved the problem. I have no time to investigate the rudix problem though. Will take a look when I will have time. Thanks a million. :)

closed 2017-04-23 by kazi93