gotthardp/lorawan-server #63: Network Quality Question

Chill0309, 2017-04-23:
Hi, should running a gateway as a packet forwarder and having the network server in the cloud run ok if the gateway is only connected with a 3G/GPRS SIM connection rather than a 'good', wired internet connection? Thanks.

gotthardp, 2017-04-23:
Hi. First of all, please note that the packet_forwarder protocol is not recommended for public internet as there is no authentication nor integrity. (This is a deficiency of the protocol, not of this server ;-) ) 

Bandwidth is not a big issue. Latency is more critical. The round-trip latency (gateway to server and back MUST be way bellow 2 seconds). Moreover, you need to adjust the `{preprocessing_delay, 300}` parameter t reflect high latency.

Hosting the server on the gateway is not an option for you?

Chill0309, 2017-04-23:
Ok thanks. On the security side, is that not an issue for TTN etc too that allow connection through packet forwarders or do they employ some other technique?

I have been running with a Multitech gateway running Node Red but we want a central server to allow multiple gateways to connect into the same application server.

Any other suggestions are more than welcome, thanks :)

gotthardp, 2017-04-24:
I see. So, please do `ping` your server from the gateway vhile connected via the 3G/GPRS and let me know what the average/min/max latency is. I will recommend some configuration changes and then we shall just try it and see how it works.

gotthardp, 2017-05-02:
Any update? May I close the issue, please?

Chill0309, 2017-05-02:
Hi, I've just actually managed to do this today, previously I've just been
on wired LAN. On 3G I'm getting around a 500ms ping so not fantastic but
everything is working as it should with 1 node connected. Its just whether
it can sustain this with 500 nodes connected!?

Thanks,

Paul

gotthardp, 2017-05-02:
It depends how much will the nodes transmit. The overhead is not big, so it could work. Worth trying.

gotthardp, 2017-05-10:
Any success? May I close the issue, please?

Chill0309, 2017-05-10:
Hi yes, it seems to work ok with 20x nodes on 3G. We'll have to assess as
more join...

gotthardp, 2017-05-10:
Cool! Just let me know if you face another issue.

closed 2017-05-10 by gotthardp