gotthardp/lorawan-server #64: Build a single channel LoRaWAN gateway (RPi3 + Dragino hat)

kazi93, 2017-04-23:
The below gateway is supported by lorawan-server ?
Server address : 192.168.0.101
Gateway address : 192.168.0.100

Use LoRa/GPS HAT + RaspberryPi to set up a LoRa Gw
https://github.com/tftelkamp/single_chan_pkt_fwd

gotthardp, 2017-04-23:
The protocol looks compatible, so it *should* work or it shouldn't be difficult to make it work. I already saw people using the lorawan-server with the single channel gateway using the Lua implementation https://github.com/JaapBraam/LoRaWanGateway, but even your should work.

In the sources I saw the server port to be hard-coded to 1700. By default server listens on 1680, so please change either of these to the same number.

altishchenko, 2017-04-24:
Just change the line in this file: lorawan-server/releases/0.4.0/sys.config
from: {packet_forwarder_listen, [{port, 1680}]}
to: {packet_forwarder_listen, [{port, 1700}]}

Regards,
Alexander Tishchenko
mob: +7 (916) 704-61-57
atishch@gmail.com

kazi93, 2017-04-26:
Do you think the below should work? I am not receiving any frame. :(
//Already changed the port from 1680 to 1700 as I would like to use ttn server for debug/testing.
// define servers
// TODO: use host names and dns
#define SERVER1 "40.114.249.243"    // The Things Network: router.eu.thethings.network
#define SERVER2 "192.168.2.198"      // local
#define PORT 1700                   // The port on which to send data

gotthardp, 2017-04-26:
You combine TTN server (SERVER1) with mine (SERVER2)? That's not a good idea. I recommend setting both SERVER1 and SERVER2 to the same IP.
The best way to debug is to look in the debug.log and into the Received Frames. Do you get something interesting in either of these?

kazi93, 2017-04-26:
@gotthardp Many thanks for quick reply. I am not receiving anything in the "Received Frames" window. The gateway and server configured int he same local network.  I have set both ip to my local ip :
#define SERVER1 "192.168.2.198" // local
#define SERVER2 "192.168.2.198" // local

gotthardp, 2017-04-26:
and could you please post here last 50 lines of the `log/debug.log`?

kazi93, 2017-04-27:
@gotthardp Thanks again. As you can see nothing exists like log/debug.log. And could you please tell me how to setup NetID? I just put 2A2A.
mac:lorawan-server-0.3.0 kazi$ ls
Mnesia.lorawan@mac	erl_crash.dump		log
bin			lib			releases
mac:lorawan-server-0.3.0 kazi$ cd log
mac:log kazi$ ls
console.log.0	crash.log.0	error.log.0
console.log.4	crash.log.4	error.log.4

kazi93, 2017-04-27:
@altishchenko Hi! I just noticed that you are using "lorawan-server/releases/0.4.0/sys.config", How to find ver. 0.4.0 ?

kazi93, 2017-04-27:
@gotthardp If this helps! Thanks. 
mac:log kazi$ cat console.log.4
2017-04-23 02:22:15.379 [info] <0.31.0> Application lager started on node lorawan@unknownACBC32C2A1A1
2017-04-23 02:23:41.178 [info] <0.31.0> Application lager started on node lorawan@unknownACBC32C2A1A1
2017-04-23 02:26:24.771 [info] <0.31.0> Application lager started on node lorawan@unknownACBC32C2A1A1
2017-04-23 02:27:25.805 [info] <0.31.0> Application lager started on node lorawan@mac
2017-04-23 02:30:43.938 [info] <0.31.0> Application lager started on node lorawan@mac
2017-04-23 19:16:03.113 [info] <0.31.0> Application lager started on node lorawan@mac
2017-04-23 19:16:03.120 [info] <0.31.0> Application crypto started on node lorawan@mac
2017-04-23 19:16:03.122 [info] <0.31.0> Application cowlib started on node lorawan@mac
2017-04-23 19:16:03.133 [info] <0.31.0> Application asn1 started on node lorawan@mac
2017-04-23 19:16:03.133 [info] <0.31.0> Application public_key started on node lorawan@mac
2017-04-23 19:16:03.155 [info] <0.31.0> Application ssl started on node lorawan@mac
2017-04-23 19:16:03.157 [info] <0.31.0> Application ranch started on node lorawan@mac
2017-04-23 19:16:03.160 [info] <0.31.0> Application cowboy started on node lorawan@mac
2017-04-23 19:16:03.239 [info] <0.31.0> Application mnesia started on node lorawan@mac
2017-04-23 19:16:03.240 [info] <0.31.0> Application jsx started on node lorawan@mac
2017-04-23 19:16:03.241 [info] <0.31.0> Application iso8601 started on node lorawan@mac
2017-04-23 19:16:03.242 [info] <0.31.0> Application lorawan_server_api started on node lorawan@mac
2017-04-23 19:16:03.247 [info] <0.31.0> Application mnesia exited with reason: stopped
2017-04-23 19:16:03.247 [info] <0.144.0>@lorawan_db:ensure_tables:20 Database create schema
2017-04-23 19:16:03.307 [info] <0.144.0>@lorawan_db:ensure_table:67 Database create users
2017-04-23 19:16:03.308 [info] <0.31.0> Application mnesia started on node lorawan@mac
2017-04-23 19:16:03.335 [info] <0.144.0>@lorawan_db:set_defaults:91 Database create default user:password
2017-04-23 19:16:03.335 [info] <0.144.0>@lorawan_db:ensure_table:67 Database create gateways
2017-04-23 19:16:03.350 [info] <0.144.0>@lorawan_db:ensure_table:67 Database create devices
2017-04-23 19:16:03.363 [info] <0.144.0>@lorawan_db:ensure_table:67 Database create links
2017-04-23 19:16:03.378 [info] <0.144.0>@lorawan_db:ensure_table:67 Database create ignored_links
2017-04-23 19:16:03.392 [info] <0.144.0>@lorawan_db:ensure_table:67 Database create pending
2017-04-23 19:16:03.408 [info] <0.144.0>@lorawan_db:ensure_table:67 Database create txframes
2017-04-23 19:16:03.420 [info] <0.144.0>@lorawan_db:ensure_table:67 Database create rxframes

gotthardp, 2017-04-27:
From what I see the gateway did never connect to the server. I suspect the port configuration is wrong. Please make sure you
 * either have the release 0.3.0 (tagged on github) and then set the listening port by `{forwarder_port, 1700}` in 0.3.0/sys.config
 * or you have the latest stable (branch on github), follow the build and release instrructions and then set the listening port as `{packet_forwarder_listen, [{port, 1700}]}` in 0.4.0/sys.config
 * don't use anything else

kazi93, 2017-04-28:
@gotthardp @altishchenko I have managed to install ver. 0.4.0 on mac. However, still no frames on my dashboard. :(
2017-04-28 17:56:50.998 [info] <0.31.0> Application lager started on node lorawan@mac
2017-04-28 17:56:51.007 [info] <0.31.0> Application crypto started on node lorawan@mac
2017-04-28 17:56:51.008 [info] <0.31.0> Application cowlib started on node lorawan@mac
2017-04-28 17:56:51.028 [info] <0.31.0> Application asn1 started on node lorawan@mac
2017-04-28 17:56:51.028 [info] <0.31.0> Application public_key started on node lorawan@mac
2017-04-28 17:56:51.036 [debug] <0.104.0> Supervisor ssl_admin_sup started ssl_pem_cache:start_link([]) at pid <0.105.0>
2017-04-28 17:56:51.040 [debug] <0.104.0> Supervisor ssl_admin_sup started ssl_manager:start_link([]) at pid <0.106.0>
2017-04-28 17:56:51.040 [debug] <0.103.0> Supervisor ssl_sup started ssl_admin_sup:start_link() at pid <0.104.0>
2017-04-28 17:56:51.043 [debug] <0.107.0> Supervisor ssl_connection_sup started tls_connection_sup:start_link() at pid <0.108.0>
2017-04-28 17:56:51.044 [debug] <0.107.0> Supervisor ssl_connection_sup started ssl_listen_tracker_sup:start_link() at pid <0.109.0>
2017-04-28 17:56:51.046 [debug] <0.107.0> Supervisor ssl_connection_sup started dtls_connection_sup:start_link() at pid <0.110.0>
2017-04-28 17:56:51.047 [debug] <0.107.0> Supervisor ssl_connection_sup started dtls_udp_sup:start_link() at pid <0.111.0>
2017-04-28 17:56:51.047 [debug] <0.103.0> Supervisor ssl_sup started ssl_connection_sup:start_link() at pid <0.107.0>
2017-04-28 17:56:51.047 [info] <0.31.0> Application ssl started on node lorawan@mac
2017-04-28 17:56:51.050 [debug] <0.115.0> Supervisor ranch_sup started ranch_server:start_link() at pid <0.116.0>
2017-04-28 17:56:51.050 [info] <0.31.0> Application ranch started on node lorawan@mac
2017-04-28 17:56:51.053 [debug] <0.120.0> Supervisor cowboy_sup started cowboy_clock:start_link() at pid <0.121.0>
2017-04-28 17:56:51.053 [info] <0.31.0> Application cowboy started on node lorawan@mac
2017-04-28 17:56:51.066 [debug] <0.125.0> Supervisor mnesia_sup started mnesia_sup:start_event() at pid <0.126.0>
2017-04-28 17:56:51.068 [debug] <0.125.0> Supervisor mnesia_sup started mnesia_ext_sup:start() at pid <0.127.0>
2017-04-28 17:56:51.080 [debug] <0.128.0> Supervisor mnesia_kernel_sup started mnesia_monitor:start() at pid <0.129.0>
2017-04-28 17:56:51.082 [debug] <0.128.0> Supervisor mnesia_kernel_sup started mnesia_subscr:start() at pid <0.130.0>
2017-04-28 17:56:51.087 [debug] <0.128.0> Supervisor mnesia_kernel_sup started mnesia_locker:start() at pid <0.131.0>
2017-04-28 17:56:51.090 [debug] <0.128.0> Supervisor mnesia_kernel_sup started mnesia_recover:start() at pid <0.132.0>
2017-04-28 17:56:51.108 [debug] <0.55.0> Supervisor kernel_safe_sup started dets_sup:start_link() at pid <0.134.0>
2017-04-28 17:56:51.109 [debug] <0.55.0> Supervisor kernel_safe_sup started dets_server:start_link() at pid <0.135.0>
2017-04-28 17:56:51.139 [debug] <0.55.0> Supervisor kernel_safe_sup started disk_log_sup:start_link() at pid <0.139.0>
2017-04-28 17:56:51.139 [debug] <0.55.0> Supervisor kernel_safe_sup started disk_log_server:start_link() at pid <0.140.0>
2017-04-28 17:56:51.181 [debug] <0.128.0> Supervisor mnesia_kernel_sup started mnesia_tm:start() at pid <0.133.0>
2017-04-28 17:56:51.183 [debug] <0.128.0> Supervisor mnesia_kernel_sup started mnesia_checkpoint_sup:start() at pid <0.154.0>
2017-04-28 17:56:51.185 [debug] <0.128.0> Supervisor mnesia_kernel_sup started mnesia_snmp_sup:start() at pid <0.155.0>
2017-04-28 17:56:51.185 [debug] <0.128.0> Supervisor mnesia_kernel_sup started mnesia_controller:start() at pid <0.156.0>
2017-04-28 17:56:51.192 [debug] <0.128.0> Supervisor mnesia_kernel_sup started mnesia_late_loader:start() at pid <0.157.0>
2017-04-28 17:56:51.193 [debug] <0.125.0> Supervisor mnesia_sup started mnesia_kernel_sup:start() at pid <0.128.0>
2017-04-28 17:56:51.193 [info] <0.31.0> Application mnesia started on node lorawan@mac
2017-04-28 17:56:51.200 [info] <0.31.0> Application jsx started on node lorawan@mac
2017-04-28 17:56:51.202 [info] <0.31.0> Application iso8601 started on node lorawan@mac
2017-04-28 17:56:51.213 [info] <0.31.0> Application emqttc started on node lorawan@mac
2017-04-28 17:56:51.214 [info] <0.31.0> Application lorawan_server_api started on node lorawan@mac
2017-04-28 17:56:51.229 [debug] <0.55.0> Supervisor kernel_safe_sup started timer:start_link() at pid <0.218.0>
2017-04-28 17:56:51.240 [debug] <0.219.0> Supervisor {<0.219.0>,ranch_listener_sup} started ranch_conns_sup:start_link(http, supervisor, 5000, ranch_tcp, 5000, cowboy_clear) at pid <0.220.0>
2017-04-28 17:56:51.242 [debug] <0.221.0> Supervisor {<0.221.0>,ranch_acceptors_sup} started ranch_acceptor:start_link(#Port<0.5302>, ranch_tcp, <0.220.0>) at pid <0.222.0>
2017-04-28 17:56:51.242 [debug] <0.221.0> Supervisor {<0.221.0>,ranch_acceptors_sup} started ranch_acceptor:start_link(#Port<0.5302>, ranch_tcp, <0.220.0>) at pid <0.223.0>
2017-04-28 17:56:51.242 [debug] <0.221.0> Supervisor {<0.221.0>,ranch_acceptors_sup} started ranch_acceptor:start_link(#Port<0.5302>, ranch_tcp, <0.220.0>) at pid <0.224.0>
2017-04-28 17:56:51.242 [debug] <0.221.0> Supervisor {<0.221.0>,ranch_acceptors_sup} started ranch_acceptor:start_link(#Port<0.5302>, ranch_tcp, <0.220.0>) at pid <0.225.0>
[... cut ...] owboy_rest,call,3,[{file,"/Users/kazi/lorawan-server/_build/default/lib/cowboy/src/cow..."},...]},...]]},...}}
2017-04-28 18:21:45.292 [debug] <0.89.0>@lager_handler_watcher:119 Lager installed handler error_logger_lager_h into error_logger
2017-04-28 18:21:45.329 [debug] <0.89.0>@lager_handler_watcher:119 Lager installed handler error_logger_lager_h into error_logger
2017-04-28 18:21:45.376 [debug] <0.89.0>@lager_handler_watcher:119 Lager installed handler error_logger_lager_h into error_logger

kazi93, 2017-04-28:
Web config 👍 
Gateways List:
	 MAC	                        NetID	
	B827EBFFFF5E0AAE	0123AB					
Devices List:
	 DevEUI	                         Region	       Application	
	1234567890123456	EU863-870	websocket	
Nodes List:
	 DevAddr	Region	        Application	
	26011778	EU863-870	websocket

gotthardp, 2017-04-28:
On the server dashboard (http://server:8080/admin#/dashboard) do you see some 'Last RX' date? Is your gateway indicated as alive?

I don't see any indication in the log the gateway actually did send the server anything. Would you be able to use Wireshark and capture the gateway to server communication?

kazi93, 2017-04-28:
Last RX is empty, nothing. Could you please tell me how to config NetID?, I guess it is a problem with NetID config. I need to setup wireshark, I will update you once it is done. 
Note : version 0.3.0 provided erlang progress report based on date and the 0.4.0 provides the below only. Could it be a problem?
mac:~ kazi$ cd lorawan-server/_build/default/rel/lorawan-server
mac:lorawan-server kazi$ bin/lorawan-server 
......only white space cursor stops it means server is running or what?...

gotthardp, 2017-04-28:
But you set some NetID. It's configured in the Gateways List.

But this is not a NetID problem. This is because the gateway didn't send anything to where the server is listening.

To get some output on the console you should uncomment `{lager_console_backend, debug}` in the config file

kazi93, 2017-04-28:
It is by default uncommented, If I am not wrong.
% error logging settings
{lager, [
    {colored, true},
    {handlers, [
        % {lager_console_backend, debug},
        {lager_file_backend, [{file, "log/debug.log"}, {level, debug}]},
        {lager_file_backend, [{file, "log/error.log"}, {level, error}]}
    ]}
]},

kazi93, 2017-04-28:
@gotthardp                  debug.log seems promising, any suggestion?                              
mac:lorawan-server kazi$ cd log/
mac:log kazi$ ls
crash.log.0	debug.log.0	error.log.0mac:log 

kazi$ cat debug.log

2017-04-29 00:17:55.876 [debug] <0.87.0>@lager_handler_watcher:119 Lager installed handler {lager_file_backend,"log/error.log"} into lager_event
2017-04-29 00:17:55.876 [debug] <0.89.0>@lager_handler_watcher:119 Lager installed handler error_logger_lager_h into error_logger
2017-04-29 00:17:55.884 [debug] <0.74.0> Supervisor gr_param_sup started gr_param:start_link(gr_lager_default_tracer_params) at pid <0.91.0>
2017-04-29 00:17:55.884 [debug] <0.73.0> Supervisor gr_counter_sup started gr_counter:start_link(gr_lager_default_tracer_counters) at pid <0.92.0>
2017-04-29 00:17:55.884 [debug] <0.75.0> Supervisor gr_manager_sup started gr_manager:start_link(gr_lager_default_tracer_params_mgr, gr_lager_default_tracer_params, []) at pid <0.93.0>
2017-04-29 00:17:55.884 [debug] <0.75.0> Supervisor gr_manager_sup started gr_manager:start_link(gr_lager_default_tracer_counters_mgr, gr_lager_default_tracer_counters, [{input,0},{filter,0},{output,0},{job_input,0},{job_run,0},{job_time,0},{job_error,0}]) at pid <0.94.0>
2017-04-29 00:17:56.041 [info] <0.31.0> Application lager started on node lorawan@mac
2017-04-29 00:17:56.050 [info] <0.31.0> Application crypto started on node lorawan@mac
2017-04-29 00:17:56.051 [info] <0.31.0> Application cowlib started on node lorawan@mac
2017-04-29 00:17:56.067 [info] <0.31.0> Application asn1 started on node lorawan@mac
2017-04-29 00:17:56.067 [info] <0.31.0> Application public_key started on node lorawan@mac
2017-04-29 00:17:56.074 [debug] <0.104.0> Supervisor ssl_admin_sup started ssl_pem_cache:start_link([]) at pid <0.105.0>
2017-04-29 00:17:56.076 [debug] <0.104.0> Supervisor ssl_admin_sup started ssl_manager:start_link([]) at pid <0.106.0>
2017-04-29 00:17:56.077 [debug] <0.103.0> Supervisor ssl_sup started ssl_admin_sup:start_link() at pid <0.104.0>
2017-04-29 00:17:56.079 [debug] <0.107.0> Supervisor ssl_connection_sup started tls_connection_sup:start_link() at pid <0.108.0>
2017-04-29 00:17:56.080 [debug] <0.107.0> Supervisor ssl_connection_sup started ssl_listen_tracker_sup:start_link() at pid <0.109.0>
2017-04-29 00:17:56.082 [debug] <0.107.0> Supervisor ssl_connection_sup started dtls_connection_sup:start_link() at pid <0.110.0>
2017-04-29 00:17:56.083 [debug] <0.107.0> Supervisor ssl_connection_sup started dtls_udp_sup:start_link() at pid <0.111.0>
2017-04-29 00:17:56.083 [debug] <0.103.0> Supervisor ssl_sup started ssl_connection_sup:start_link() at pid <0.107.0>
2017-04-29 00:17:56.083 [info] <0.31.0> Application ssl started on node lorawan@mac
2017-04-29 00:17:56.085 [debug] <0.115.0> Supervisor ranch_sup started ranch_server:start_link() at pid <0.116.0>
2017-04-29 00:17:56.085 [info] <0.31.0> Application ranch started on node lorawan@mac
2017-04-29 00:17:56.088 [debug] <0.120.0> Supervisor cowboy_sup started cowboy_clock:start_link() at pid <0.121.0>
2017-04-29 00:17:56.088 [info] <0.31.0> Application cowboy started on node lorawan@mac
2017-04-29 00:17:56.099 [debug] <0.125.0> Supervisor mnesia_sup started mnesia_sup:start_event() at pid <0.126.0>
2017-04-29 00:17:56.101 [debug] <0.125.0> Supervisor mnesia_sup started mnesia_ext_sup:start() at pid <0.127.0>
2017-04-29 00:17:56.114 [debug] <0.128.0> Supervisor mnesia_kernel_sup started mnesia_monitor:start() at pid <0.129.0>
2017-04-29 00:17:56.117 [debug] <0.128.0> Supervisor mnesia_kernel_sup started mnesia_subscr:start() at pid <0.130.0>
2017-04-29 00:17:56.121 [debug] <0.128.0> Supervisor mnesia_kernel_sup started mnesia_locker:start() at pid <0.131.0>
2017-04-29 00:17:56.124 [debug] <0.128.0> Supervisor mnesia_kernel_sup started mnesia_recover:start() at pid <0.132.0>
2017-04-29 00:17:56.143 [debug] <0.55.0> Supervisor kernel_safe_sup started dets_sup:start_link() at pid <0.134.0>
2017-04-29 00:17:56.143 [debug] <0.55.0> Supervisor kernel_safe_sup started dets_server:start_link() at pid <0.135.0>
2017-04-29 00:17:56.172 [debug] <0.55.0> Supervisor kernel_safe_sup started disk_log_sup:start_link() at pid <0.139.0>
2017-04-29 00:17:56.172 [debug] <0.55.0> Supervisor kernel_safe_sup started disk_log_server:start_link() at pid <0.140.0>
2017-04-29 00:17:56.215 [debug] <0.128.0> Supervisor mnesia_kernel_sup started mnesia_tm:start() at pid <0.133.0>
2017-04-29 00:17:56.218 [debug] <0.128.0> Supervisor mnesia_kernel_sup started mnesia_checkpoint_sup:start() at pid <0.154.0>
2017-04-29 00:17:56.221 [debug] <0.128.0> Supervisor mnesia_kernel_sup started mnesia_snmp_sup:start() at pid <0.155.0>
2017-04-29 00:17:56.221 [debug] <0.128.0> Supervisor mnesia_kernel_sup started mnesia_controller:start() at pid <0.156.0>
2017-04-29 00:17:56.228 [debug] <0.128.0> Supervisor mnesia_kernel_sup started mnesia_late_loader:start() at pid <0.157.0>
2017-04-29 00:17:56.228 [debug] <0.125.0> Supervisor mnesia_sup started mnesia_kernel_sup:start() at pid <0.128.0>
2017-04-29 00:17:56.229 [info] <0.31.0> Application mnesia started on node lorawan@mac
2017-04-29 00:17:56.233 [info] <0.31.0> Application jsx started on node lorawan@mac
2017-04-29 00:17:56.235 [info] <0.31.0> Application iso8601 started on node lorawan@mac
2017-04-29 00:17:56.238 [info] <0.31.0> Application emqttc started on node lorawan@mac
2017-04-29 00:17:56.239 [info] <0.31.0> Application lorawan_server_api started on node lorawan@mac
[... cut ...]
2017-04-29 00:17:56.260 [debug] <0.224.0> Supervisor lorawan_sup started lorawan_gw_forwarder:start_link([{port,1700}]) at pid <0.226.0>
2017-04-29 00:17:56.260 [debug] <0.224.0> Supervisor lorawan_sup started lorawan_connector_sup:start_link() at pid <0.227.0>
2017-04-29 00:17:56.262 [debug] <0.224.0> Supervisor lorawan_sup started lorawan_connector_factory:start_link() at pid <0.228.0>
2017-04-29 00:17:56.262 [info] <0.31.0> Application lorawan_server started on node lorawan@mac
2017-04-29 00:17:56.366 [debug] <0.83.0>@lager_handler_watcher:119 Lager installed handler lager_backend_throttle into lager_event

gotthardp, 2017-04-28:
By default it is commented out. See https://github.com/gotthardp/lorawan-server/blob/master/lorawan_server.config#L23

You really, really need to see a line like this
```
23:00:01.792 [info] Gateway <<184,39,235,255,255,176,253,196>> at {global,lorawan_gw_forwarder} {{192,168,1,221},59357,1}
```

I don't see it in your log, which means the gateway didn't send anything. The error is in the gateway, or on the way to the server.

kazi93, 2017-04-28:
@Unfortunately, nothing like this. Could you please take a look in to my admin portal?.
[image]
[image]
[image]

gotthardp, 2017-04-28:
In can try to have a look.

But please, could you try `sudo tcpdump -i eth0 -w file.pcap` and send me the file.pcap? If there is no tcpdump you should `sudo apt-get install tcpdump`.

kazi93, 2017-04-28:
pi@LoRaWANGateway:~ $ sudo tcpdump -i wlan0 -w file.pcap
tcpdump: listening on wlan0, link-type EN10MB (Ethernet), capture size 262144 bytes

Note: Both gw and server configured on Wireless LAN. e.g 192.168.0.0/24.
How to save this file in raspbian?, if you have any advice pls

gotthardp, 2017-04-28:
the file is already saved, you need to download it and/or attach it here

kazi93, 2017-04-28:
@gotthardp Thanks for your patience. :)
[file.pcap.zip](https://github.com/gotthardp/lorawan-server/files/965749/file.pcap.zip)

gotthardp, 2017-04-28:
I can see the gateway talking to 40.114.249.243. That's likely the TTN. You probably forgot to compile, install or configure it properly.

kazi93, 2017-04-29:
@gotthardp I make the file again. I am surprised to know it was sending data to ttn :(
pi@LoRaWANGateway:~/lorawan_gateway/single_chan_pkt_fwd $ make
g++ -c -Wall main.cpp
g++ main.o base64.o -lwiringPi -o single_chan_pkt_fwd

pi@LoRaWANGateway:~/lorawan_gateway/single_chan_pkt_fwd $ ./single_chan_pkt_fwd
SX1276 detected, starting.
Gateway ID: b8:27:eb:ff:ff:5e:0a:ae
Listening at SF7 on 868.100000 Mhz.
------------------
stat update: {"stat":{"time":"2017-04-29 00:26:16 GMT","lati":52.53758,"long":13.39395,"alti":49,"rxnb":0,"rxok":0,"rxfw":0,"ackr":0.0,"dwnb":0,"txnb":0,"pfrm":"Single Channel Gateway","mail":"iotdev.kazi@gmail.com","desc":"NGN LoRaWAN Gateway"}}
stat update: {"stat":{"time":"2017-04-29 00:26:46 GMT","lati":52.53758,"long":13.39395,"alti":49,"rxnb":0,"rxok":0,"rxfw":0,"ackr":0.0,"dwnb":0,"txnb":0,"pfrm":"Single Channel Gateway","mail":"iotdev.kazi@gmail.com","desc":"NGN LoRaWAN Gateway"}}
stat update: {"stat":{"time":"2017-04-29 00:27:16 GMT","lati":52.53758,"long":13.39395,"alti":49,"rxnb":0,"rxok":0,"rxfw":0,"ackr":0.0,"dwnb":0,"txnb":0,"pfrm":"Single Channel Gateway","mail":"iotdev.kazi@gmail.com","desc":"NGN LoRaWAN Gateway"}}
Packet RSSI: -45, RSSI: -98, SNR: 9, Length: 15
rxpk update: {"rxpk":[{"tmst":1095985879,"chan":0,"rfch":0,"freq":868.100000,"stat":1,"modu":"LORA","datr":"SF7BW125","codr":"4/5","lsnr":9,"rssi":-45,"size":15,"data":"QHgXASaAaQAB7j45PGOU"}]}

kazi93, 2017-04-29:
@gotthardp Anything suspicious? I maked the file again. :)
/*******************************************************************************
 *
 * Configure these values!
 *
 *******************************************************************************                                                                                                                     /

// SX1272 - Raspberry connections
int ssPin = 6;
int dio0  = 7;
int RST   = 0;

// Set spreading factor (SF7 - SF12)
sf_t sf = SF7;

// Set center frequency
uint32_t  freq = 868100000; // in Mhz! (868.1)

// Set location
float lat=52.5375839;
float lon=13.3939528;
int   alt=49;

/* Informal status fields */
static char platform[24]    = "Single Channel Gateway";  /* platform definition                                                                                                                      */
static char email[40]       = "iotdev.kazi@gmail.com";  /* used for contact emai                                                                                                                     l */
static char description[64] = "NGN LoRaWAN Gateway";   /* used for free form des                                                                                                                     cription */

// define servers
// TODO: use host names and dns
//#define SERVER1 "40.114.249.243"      // The Things Network: router.eu.thethin                                                                                                                     gs.network

#define SERVER1 "192.168.0.100"       //local home
#define SERVER2 "192.168.0.100"       //local home

#define PORT 1700                    // The port on which to send data

kazi93, 2017-04-29:
@gotthardp Improvement :) Gateway status: Alive 
02:49:59.487 [debug] Supervisor kernel_safe_sup started timer:start_link() at pid <0.218.0>
02:49:59.499 [debug] Supervisor {<0.219.0>,ranch_listener_sup} started ranch_conns_sup:start_link(http, supervisor, 5000, ranch_tcp, 5000, cowboy_clear) at pid <0.220.0>
02:49:59.500 [debug] Supervisor {<0.221.0>,ranch_acceptors_sup} started ranch_acceptor:start_link(#Port<0.5305>, ranch_tcp, <0.220.0>) at pid <0.222.0>
02:49:59.500 [debug] Supervisor {<0.221.0>,ranch_acceptors_sup} started ranch_acceptor:start_link(#Port<0.5305>, ranch_tcp, <0.220.0>) at pid <0.223.0>
02:49:59.500 [debug] Supervisor {<0.221.0>,ranch_acceptors_sup} started ranch_acceptor:start_link(#Port<0.5305>, ranch_tcp, <0.220.0>) at pid <0.224.0>
02:49:59.500 [debug] Supervisor {<0.221.0>,ranch_acceptors_sup} started ranch_acceptor:start_link(#Port<0.5305>, ranch_tcp, <0.220.0>) at pid <0.225.0>
02:49:59.500 [debug] Supervisor {<0.221.0>,ranch_acceptors_sup} started ranch_acceptor:start_link(#Port<0.5305>, ranch_tcp, <0.220.0>) at pid <0.226.0>
02:49:59.607 [debug] Lager installed handler lager_backend_throttle into lager_event
02:50:55.064 [error] ERROR: {bad_mic,<<38,1,23,120>>}
02:52:11.496 [error] ERROR: {bad_mic,<<38,1,23,120>>}
02:53:30.691 [error] ERROR: {bad_mic,<<38,1,23,120>>}
02:53:30.945 [error] ERROR: {bad_mic,<<38,1,23,120>>}
02:54:48.786 [error] ERROR: {bad_mic,<<38,1,23,120>>}
02:56:07.939 [error] ERROR: {bad_mic,<<38,1,23,120>>}

kazi93, 2017-04-29:
@gotthardp My Node has the below code:
https://github.com/SensorsIot/LoRa/blob/master/Nodes/Dragino/HelloWorld/HelloWorld.ino

And configured as Nodes (ABP) in the admin panel.

gotthardp, 2017-04-29:
Cool. The `bad_mic` error means the Keys don't match. How did you set the NWKSKEY1 and APPSKEY1?

kazi93, 2017-04-29:
@gotthardp I guess, I just copied from the *.ino file to the admin Nodes page. It is the same with the last screenshot. :)
#ifdef CREDENTIALS
static const u1_t NWKSKEY[16] = NWKSKEY1;
static const u1_t APPSKEY[16] = APPSKEY1;
static const u4_t DEVADDR = DEVADDR1;
#else
static const u1_t NWKSKEY[16] = "DD67AEA283EE40CE803F3E358D4FF76F";
static const u1_t APPSKEY[16] = "1CF1B11778C0B7019F3C6B4D02F1DBFC";
static const u4_t DEVADDR = 0x26011B64;
#endif

gotthardp, 2017-04-29:
I believe you have to enter it as a struct, not as a string, i.e.:
{0xDD, 0x67, 0xAE, 0xA2, 0x83, 0xEE, 0x40, 0xCE, 0x80, 0x3F, 0x3E, 0x35, 0x8D, 0x4F, 0xF7, 0x6F}

kazi93, 2017-04-29:
@gotthardp Thanks. I changed the strings in the *.ino file as below. However, I cannot paste the same in the Nodes page(admin portal) as it does supports only string format.

static const u1_t NWKSKEY[16] = "{ 0xDD, 0x67, 0xAE, 0xA2, 0x83, 0xEE, 0x40, 0xCE, 0x80, 0x3F, 0x3E, 0x35, 0x8D, 0x4F, 0xF7, 0x6F }";
static const u1_t APPSKEY[16] = "{ 0x1C, 0xF1, 0xB1, 0x17, 0x78, 0xC0, 0xB7, 0x01, 0x9F, 0x3C, 0x6B, 0x4D, 0x02, 0xF1, 0xDB, 0xFC }";
static const u4_t DEVADDR = 0x26011B64;

kazi93, 2017-04-29:
@gotthardp After the change it is still the same error:

15:43:26.014 [debug] Lager installed handler lager_backend_throttle into lager_event
15:43:58.667 [error] ERROR: {bad_mic,<<38,1,27,100>>}
15:46:17.205 [error] ERROR: {bad_mic,<<38,1,27,100>>}
15:46:39.546 [error] ERROR: {bad_mic,<<38,1,27,100>>}
15:47:28.459 [error] ERROR: {bad_mic,<<38,1,27,100>>}
15:48:17.498 [error] ERROR: {bad_mic,<<38,1,27,100>>}
15:49:05.150 [error] ERROR: {bad_mic,<<38,1,27,100>>}
15:49:54.706 [error] ERROR: {bad_mic,<<38,1,27,100>>}
15:50:43.044 [error] ERROR: {bad_mic,<<38,1,27,100>>}

gotthardp, 2017-04-29:
Remove the `"` signs. The keys in your program shall be a structure, not a string.

And yes, you need it write it twice in different formats, one as a hex-string in the web-admin and once as a C-struct in your source.

kazi93, 2017-04-29:
@gotthardp Million thanks! it works like a charm. I am receiving packets on the portal. :)

gotthardp, 2017-04-29:
Cool! We made it :)

closed 2017-04-29 by gotthardp