gotthardp/lorawan-server #65: ERROR: bad_mic

RCOdev, 2017-04-25:
Hi,

I set up the LoRaWAN server with a Lorank8 gateway and a LoRa end-node (ATIM module). With ABP mode, I create the Link manually and everything is good. With another end-node, I would like to try the OTAA mode but I got this error in log : lorawan_handler:rxpk:21 ERROR: bad_mic
So no Link is automatically created...

I don't understand because with OTAA mode there is no session (Link) predefined so it can't be a mic error... Am I right?

Do you know why I get this error?

Thanks!

gotthardp, 2017-04-25:
Check for bad_mic error in the [Troubleshooting Guide](https://github.com/gotthardp/lorawan-server/blob/master/doc/Troubleshooting.md) please.

gotthardp, 2017-04-30:
Were you able to fix this issue?

RCOdev, 2017-05-02:
Hi,
thank you for the link. I found my mistake, my AppKey was wrong. I used LSB first and reversed all bytes but it was not necessary.

Everything works, thanks!

closed 2017-05-02 by RCOdev