gotthardp/lorawan-server #68: Cant access web page

felixgalindo, 2017-04-28:
I have the server running on a raspberry pi but when I try to go to the webpage from a browser in the same network 192.168.2.153:8080 I don't get anything. 192.168.2.153 is my Pis IP address.  I manually ran it using ./bin/lorawan-server. Any ideas why I am not getting anything?

gotthardp, 2017-04-28:
Hello. What HTTP response comes, if any? Is there anything suspicious in the server logs?

felixgalindo, 2017-04-28:
Where are the server logs?

On my browser it returns

*192.168.2.153* didn’t send any data.

ERR_EMPTY_RESPONSE.

gotthardp, 2017-04-28:
The logs are introduced in
https://github.com/gotthardp/lorawan-server/blob/master/doc/Troubleshooting.md

kazi93, 2017-04-28:
@felixgalindo What is the output of the "./bin/lorawan-server"? Simply blank or it provides some output e.g erlang progress report?

felixgalindo, 2017-04-28:
> The logs are introduced in
> https://github.com/gotthardp/lorawan-server/blob/master/
> doc/Troubleshooting.md

I get the following in the error log:

2017-04-28 22:06:48.299 [error] <0.272.0> CRASH REPORT Process <0.272.0>
with 0 neighbours exited with reason: call to undefined function
erlang:cancel_timer(#Ref<0.0.0.637>, [{async,true},{info,false}]) in
cowboy_clear:proc_lib_hack/5 line 34

2017-04-28 22:06:48.299 [error] <0.272.0> Ranch listener http terminated
with reason: call to undefined function
erlang:cancel_timer(#Ref<0.0.0.637>, [{async,true},{info,false}])

2017-04-28 22:06:48.313 [error] <0.273.0> CRASH REPORT Process <0.273.0>
with 0 neighbours exited with reason: call to undefined function
erlang:cancel_timer(#Ref<0.0.0.638>, [{async,true},{info,false}]) in
cowboy_clear:proc_lib_hack/5 line 34

2017-04-28 22:06:48.313 [error] <0.273.0> Ranch listener http terminated
with reason: call to undefined function
erlang:cancel_timer(#Ref<0.0.0.638>, [{async,true},{info,false}])

2017-04-28 22:06:48.325 [error] <0.274.0> CRASH REPORT Process <0.274.0>
with 0 neighbours exited with reason: call to undefined function
erlang:cancel_timer(#Ref<0.0.0.643>, [{async,true},{info,false}]) in
cowboy_clear:proc_lib_hack/5 line 34

2017-04-28 22:06:48.326 [error] <0.274.0> Ranch listener http terminated
with reason: call to undefined function
erlang:cancel_timer(#Ref<0.0.0.643>, [{async,true},{info,false}])

2017-04-28 22:06:48.536 [error] <0.275.0> CRASH REPORT Process <0.275.0>
with 0 neighbours exited with reason: call to undefined function
erlang:cancel_timer(#Ref<0.0.0.652>, [{async,true},{info,false}]) in
cowboy_clear:proc_lib_hack/5 line 34

2017-04-28 22:06:48.538 [error] <0.275.0> Ranch listener http terminated
with reason: call to undefined function
erlang:cancel_timer(#Ref<0.0.0.652>, [{async,true},{info,false}])

2017-04-28 22:06:48.548 [error] <0.276.0> CRASH REPORT Process <0.276.0>
with 0 neighbours exited with reason: call to undefined function
erlang:cancel_timer(#Ref<0.0.0.654>, [{async,true},{info,false}]) in
cowboy_clear:proc_lib_hack/5 line 34

2017-04-28 22:06:48.549 [error] <0.276.0> Ranch listener http terminated
with reason: call to undefined function
erlang:cancel_timer(#Ref<0.0.0.654>, [{async,true},{info,false}])

2017-04-28 22:06:48.559 [error] <0.277.0> CRASH REPORT Process <0.277.0>
with 0 neighbours exited with reason: call to undefined function
erlang:cancel_timer(#Ref<0.0.0.658>, [{async,true},{info,false}]) in
cowboy_clear:proc_lib_hack/5 line 34

2017-04-28 22:06:48.560 [error] <0.277.0> Ranch listener http terminated
with reason: call to undefined function
erlang:cancel_timer(#Ref<0.0.0.658>, [{async,true},{info,false}])

2017-04-28 22:06:48.679 [error] <0.278.0> CRASH REPORT Process <0.278.0>
with 0 neighbours exited with reason: call to undefined function
erlang:cancel_timer(#Ref<0.0.0.663>, [{async,true},{info,false}]) in
cowboy_clear:proc_lib_hack/5 line 34

2017-04-28 22:06:48.681 [error] <0.278.0> Ranch listener http terminated
with reason: call to undefined function
erlang:cancel_timer(#Ref<0.0.0.663>, [{async,true},{info,false}])

2017-04-28 22:06:53.709 [error] <0.279.0> CRASH REPORT Process <0.279.0>
with 0 neighbours exited with reason: call to undefined function
erlang:cancel_timer(#Ref<0.0.0.673>, [{async,true},{info,false}]) in
cowboy_clear:proc_lib_hack/5 line 34

2017-04-28 22:06:53.710 [error] <0.279.0> Ranch listener http terminated
with reason: call to undefined function
erlang:cancel_timer(#Ref<0.0.0.673>, [{async,true},{info,false}])

felixgalindo, 2017-04-28:
I get the progress report:

=PROGRESS REPORT==== 28-Apr-2017::22:05:37 ===

          supervisor: {local,sasl_safe_sup}

             started: [{pid,<0.41.0>},

                       {name,alarm_handler},

                       {mfargs,{alarm_handler,start_link,[]}},

                       {restart_type,permanent},

                       {shutdown,2000},

                       {child_type,worker}]

=PROGRESS REPORT==== 28-Apr-2017::22:05:37 ===

          supervisor: {local,sasl_safe_sup}

             started: [{pid,<0.42.0>},

                       {name,overload},

                       {mfargs,{overload,start_link,[]}},

                       {restart_type,permanent},

                       {shutdown,2000},

                       {child_type,worker}]

=PROGRESS REPORT==== 28-Apr-2017::22:05:37 ===

          supervisor: {local,sasl_sup}

             started: [{pid,<0.40.0>},

                       {name,sasl_safe_sup},

                       {mfargs,

                           {supervisor,start_link,

                               [{local,sasl_safe_sup},sasl,safe]}},

                       {restart_type,permanent},

                       {shutdown,infinity},

                       {child_type,supervisor}]

=PROGRESS REPORT==== 28-Apr-2017::22:05:37 ===

          supervisor: {local,sasl_sup}

             started: [{pid,<0.43.0>},

                       {name,release_handler},

                       {mfargs,{release_handler,start_link,[]}},

                       {restart_type,permanent},

                       {shutdown,2000},

                       {child_type,worker}]

=PROGRESS REPORT==== 28-Apr-2017::22:05:37 ===

         application: sasl

          started_at: lorawan@raspberrypi

=PROGRESS REPORT==== 28-Apr-2017::22:05:38 ===

         application: syntax_tools

          started_at: lorawan@raspberrypi

=PROGRESS REPORT==== 28-Apr-2017::22:05:38 ===

         application: compiler

          started_at: lorawan@raspberrypi

=PROGRESS REPORT==== 28-Apr-2017::22:05:38 ===

          supervisor: {local,gr_sup}

             started: [{pid,<0.51.0>},

                       {name,gr_counter_sup},

                       {mfargs,{gr_counter_sup,start_link,[]}},

                       {restart_type,permanent},

                       {shutdown,5000},

                       {child_type,supervisor}]

=PROGRESS REPORT==== 28-Apr-2017::22:05:38 ===

          supervisor: {local,gr_sup}

             started: [{pid,<0.52.0>},

                       {name,gr_param_sup},

                       {mfargs,{gr_param_sup,start_link,[]}},

                       {restart_type,permanent},

                       {shutdown,5000},

                       {child_type,supervisor}]

=PROGRESS REPORT==== 28-Apr-2017::22:05:38 ===

          supervisor: {local,gr_sup}

             started: [{pid,<0.53.0>},

                       {name,gr_manager_sup},

                       {mfargs,{gr_manager_sup,start_link,[]}},

                       {restart_type,permanent},

                       {shutdown,5000},

                       {child_type,supervisor}]

=PROGRESS REPORT==== 28-Apr-2017::22:05:38 ===

         application: goldrush

          started_at: lorawan@raspberrypi

=PROGRESS REPORT==== 28-Apr-2017::22:05:38 ===

          supervisor: {local,lager_sup}

             started: [{pid,<0.58.0>},

                       {name,lager},

{mfargs,{gen_event,start_link,[{local,lager_event}]}},

                       {restart_type,permanent},

                       {shutdown,5000},

                       {child_type,worker}]

=PROGRESS REPORT==== 28-Apr-2017::22:05:38 ===

          supervisor: {local,lager_sup}

             started: [{pid,<0.59.0>},

                       {name,lager_handler_watcher_sup},

                       {mfargs,{lager_handler_watcher_sup,start_link,[]}},

                       {restart_type,permanent},

                       {shutdown,5000},

                       {child_type,supervisor}]

=PROGRESS REPORT==== 28-Apr-2017::22:05:38 ===

          supervisor: {local,lager_sup}

             started: [{pid,<0.60.0>},

                       {name,lager_crash_log},

                       {mfargs,

                           {lager_crash_log,start_link,

["log/crash.log",65536,10485760,[{hour,0}],5]}},

                       {restart_type,permanent},

                       {shutdown,5000},

                       {child_type,worker}]

gotthardp, 2017-04-28:
This looks like a problem of your Erlang installation. Which version are you running and how did you install it?

felixgalindo, 2017-04-28:
I ran apt-get install erlang

Running "erl" I get:

Erlang/OTP 17 [erts-6.2] [source] [smp:4:4] [async-threads:10]
[kernel-poll:false]

Eshell V6.2  (abort with ^G)

gotthardp, 2017-04-28:
17 is too old. You need at least 19, but it's very easy to install it if you follow the instructions at https://www.erlang-solutions.com/resources/download.html

felixgalindo, 2017-04-28:
> com/resources/download.html

Thanks that worked!

gotthardp, 2017-04-28:
Cool. I am glad it did.

closed 2017-04-28 by gotthardp