gotthardp/lorawan-server #69: Device doesn't OTAA join

felixgalindo, 2017-04-29:
So I have a microchip RN2903 device that is able to join using abp and I am able to view the packets sent up to the server. Problem I am having is the device tries to OTAA join and it seems like the server accepts the request but the device is not receiving the notice that it has joined.

Are my rx window settings wrong? What should they be and how do I set them for my device? I am in the U.S.

Below is my config:
{
	"SX1301_conf": {
		"lorawan_public": true,
		"antenna_gain": 0,
		"clksrc": 0,
		"radio_0": {
			"enable": true,
			"freq": 902600000,
			"rssi_offset": -165,
			"tx_enable": true,
			"tx_freq_max": 928000000,
			"tx_freq_min": 902000000,
			"type": "SX1257"
		},
		"radio_1": {
			"enable": true,
			"freq": 903400000,
			"rssi_offset": -165,
			"tx_enable": false,
			"type": "SX1257"
		},
		"chan_FSK": {
			"enable": false,
			"radio": 0
		},
		"chan_Lora_std": {
			"bandwidth": 500000,
			"enable": true,
			"if": 400000,
			"radio": 0,
			"spread_factor": 8
		},
		"chan_multiSF_0": {
			"enable": true,
			"if": -300000,
			"radio": 0
		},
		"chan_multiSF_1": {
			"enable": true,
			"if": -100000,
			"radio": 0
		},
		"chan_multiSF_2": {
			"enable": true,
			"if": 100000,
			"radio": 0
		},
		"chan_multiSF_3": {
			"enable": true,
			"if": 300000,
			"radio": 0
		},
		"chan_multiSF_4": {
			"enable": true,
			"if": -300000,
			"radio": 1
		},
		"chan_multiSF_5": {
			"enable": true,
			"if": -100000,
			"radio": 1
		},
		"chan_multiSF_6": {
			"enable": true,
			"if": 100000,
			"radio": 1
		},
		"chan_multiSF_7": {
			"enable": true,
			"if": 300000,
			"radio": 1
		},
		"lbt_cfg": {
			"enable": false,
			"nb_channel": 1,
			"rssi_target": 160,
			"scan_time_us": 5000,
			"start_freq": 869525000,
			"tx_delay_1ch_us": 4000000,
			"tx_delay_2ch_us": 4000000
		},
		"tx_lut_0": {
			"dig_gain": 0,
			"mix_gain": 11,
			"pa_gain": 0,
			"rf_power": -6
		},
		"tx_lut_1": {
			"dig_gain": 0,
			"mix_gain": 13,
			"pa_gain": 0,
			"rf_power": -3
		},
		"tx_lut_2": {
			"dig_gain": 0,
			"mix_gain": 9,
			"pa_gain": 1,
			"rf_power": 0
		},
		"tx_lut_3": {
			"dig_gain": 0,
			"mix_gain": 10,
			"pa_gain": 1,
			"rf_power": 3
		},
		"tx_lut_4": {
			"dig_gain": 0,
			"mix_gain": 12,
			"pa_gain": 1,
			"rf_power": 6
		},
		"tx_lut_5": {
			"dig_gain": 0,
			"mix_gain": 10,
			"pa_gain": 2,
			"rf_power": 10
		},
		"tx_lut_6": {
			"dig_gain": 0,
			"mix_gain": 11,
			"pa_gain": 2,
			"rf_power": 11
		},
		"tx_lut_7": {
			"dig_gain": 0,
			"mix_gain": 11,
			"pa_gain": 2,
			"rf_power": 12
		},
		"tx_lut_8": {
			"dig_gain": 0,
			"mix_gain": 12,
			"pa_gain": 2,
			"rf_power": 13
		},
		"tx_lut_9": {
			"dig_gain": 0,
			"mix_gain": 13,
			"pa_gain": 2,
			"rf_power": 14
		},
		"tx_lut_10": {
			"dig_gain": 0,
			"mix_gain": 15,
			"pa_gain": 2,
			"rf_power": 16
		},
		"tx_lut_11": {
			"dig_gain": 0,
			"mix_gain": 10,
			"pa_gain": 3,
			"rf_power": 20
		},
		"tx_lut_12": {
			"dig_gain": 0,
			"mix_gain": 12,
			"pa_gain": 3,
			"rf_power": 23
		},
		"tx_lut_13": {
			"dig_gain": 0,
			"mix_gain": 13,
			"pa_gain": 3,
			"rf_power": 25
		},
		"tx_lut_14": {
			"dig_gain": 0,
			"mix_gain": 15,
			"pa_gain": 3,
			"rf_power": 26
		},
		"tx_lut_15": {
			"dig_gain": 0,
			"mix_gain": 15,
			"pa_gain": 3,
			"rf_power": 27
		}
	},
	"gateway_conf": {
		"gateway_ID": "AA555A0000000000",
		"server_address": "127.0.0.1",
		"serv_port_up": 1680,
		"serv_port_down": 1680,
		"keepalive_interval": 10,
		"stat_interval": 30,
		"push_timeout_ms": 100,
		"forward_crc_valid": true,
		"forward_crc_error": false,
		"forward_crc_disabled": false,
		"synch_word": 52
	}
}

gotthardp, 2017-04-29:
Which server version are you using? Please make sure you set "Region" to "US902-928". There is another US region called either "Private" or "Hybrid", which shouldn't be used in your case.

Also, please check the gateway log to see if the gateway did receive the downlink, did transmit the downlink to the device and didn't complain. Please make sure the "TX Chain" in your gateway configuration is set to zero.

felixgalindo, 2017-04-29:
What is difference between the private region and not private?

gotthardp, 2017-04-29:
The "private" should be called "hybrid". It's the Multitech proprietary way of allocating channels. See the table in http://www.multitech.net/developer/software/lora/introduction-to-lora

gotthardp, 2017-04-29:
Is this problems solved too?

felixgalindo, 2017-04-29:
Yes but experiencing a related issue where my device now otaa joins but doesn't receive acks. Messages are going up to the server fine just not receiving acks.

What could cause that to be?

gotthardp, 2017-04-29:
Could you please check at your gateway whether the downlink was sent by the server and not delivered by the gateway, or not sent by the server? You can check it in the gateway log or by looking at the LED (if your gateway has one) that shines during TX.

gotthardp, 2017-05-10:
Any success? May I close the issue, please?

Please note that I just released version 0.4.0, which may perform better than the version you are currently using.

felixgalindo, 2017-05-13:
This can be closed.

closed 2017-05-13 by gotthardp