gotthardp/lorawan-server #70: Can't create an MQTT connection

felixgalindo, 2017-04-29:
When I read the instructions on how to create an MQTT connection it states to "Open the lorawan-server web-administration and create a Backend Connector". I do not see a way to set this up through the admin page. How do I set up MQTT? Thanks in advance.

gotthardp, 2017-04-29:
You don't see 'backends' in the left menu? Then you use an old version of the server. MQTT is available in the latest 'master' branch only. It's not yet available in any of the stable releases.

kazi93, 2017-04-29:
@felixgalindo 0.4.0 version has this feature.

felixgalindo, 2017-04-29:
I only see 0.3.0.

https://github.com/gotthardp/lorawan-server/releases

gotthardp, 2017-04-29:
0.4.0 is about to be released "soon". It's the https://github.com/gotthardp/lorawan-server/tree/stable code.

felixgalindo, 2017-04-29:
Im using 0.4.0. See picture below:
[image: Inline image 1]

felixgalindo, 2017-04-29:
[image]

gotthardp, 2017-04-29:
What happens when you reload the web-page to clear the browser cache?

felixgalindo, 2017-04-29:
Ok clearing cache worked.

felixgalindo, 2017-04-29:
Now my issue is my MQTT broker doesn't seem to be receiving any data from the lorawan server. I can see the client subscribing but it is not publishing anything.

1493483218: mosquitto version 1.4.10 (build date 2016-09-16 20:26:12+0100) starting
1493483218: Config loaded from /usr/local/etc/mosquitto/mosquitto.conf.
1493483218: Opening ipv6 listen socket on port 1883.
1493483218: Opening ipv4 listen socket on port 1883.
1493483230: New connection from 192.168.2.153 on port 1883.
1493483230: New client connected from 192.168.2.153 as Kp-HiwR0etvGe0y3 (c1, k0).
1493483230: Sending CONNACK to Kp-HiwR0etvGe0y3 (0, 0)
1493483230: Received SUBSCRIBE from Kp-HiwR0etvGe0y3
1493483230: 	in/# (QoS 1)
1493483230: Kp-HiwR0etvGe0y3 1 in/#
1493483230: Sending SUBACK to Kp-HiwR0etvGe0y3

felixgalindo, 2017-04-29:
[image]
[image]

felixgalindo, 2017-04-29:
2017-04-29 16:27:10.410 [debug] <0.118.0> Supervisor ranch_sup started ranch_listener_sup:start_link(http, 10, ranch_tcp, [{connection_type,supervisor},{port,8080}], cowboy_clear, #{connection_type => supervisor,env => #{dispatch => [{'_',[],[{[<<"applications">>],[],lorawan_admin_applications,...},...]}]}}) at pid <0.230.0>
2017-04-29 16:27:10.416 [debug] <0.243.0> Supervisor lorawan_sup started lorawan_gw_router:start_link() at pid <0.244.0>
2017-04-29 16:27:10.436 [debug] <0.243.0> Supervisor lorawan_sup started lorawan_gw_forwarder:start_link([{port,1680}]) at pid <0.245.0>
2017-04-29 16:27:10.440 [debug] <0.243.0> Supervisor lorawan_sup started lorawan_connector_sup:start_link() at pid <0.246.0>
2017-04-29 16:27:10.444 [info] <0.247.0>@lorawan_connector_factory:init_connector:64 Connecting MQTT to mqtt://192.168.2.24:1883
2017-04-29 16:27:10.517 [debug] <0.250.0> Supervisor inet_gethost_native_sup started undefined at pid <0.251.0>
2017-04-29 16:27:10.518 [debug] <0.56.0> Supervisor kernel_safe_sup started inet_gethost_native:start_link() at pid <0.250.0>
2017-04-29 16:27:10.530 [debug] <0.246.0> Supervisor lorawan_connector_sup started lorawan_connector_mqtt:start_link({mqtt,[],"192.168.2.24",1883,"/",[]}, {connector,<<"MQTT">>,true,<<"mqtt://192.168.2.24:1883">>,<<"out/{devaddr}">>,<<"in/#">>,<<"in/{de...">>,...}) at pid <0.248.0>
2017-04-29 16:27:10.530 [debug] <0.243.0> Supervisor lorawan_sup started lorawan_connector_factory:start_link() at pid <0.247.0>
2017-04-29 16:27:10.530 [info] <0.31.0> Application lorawan_server started on node lorawan@raspberrypi
2017-04-29 16:27:11.812 [debug] <0.253.0>@lorawan_mac_commands:auto_adr0:165 DataRate <<86,29,43,142>>: average snr 9 2.5 = 6, dr 3 -> step 2
2017-04-29 16:27:11.812 [debug] <0.253.0>@lorawan_mac_commands:auto_adr0:175 Power <<86,29,43,142>>: average rssi -48, power 5 -> up by 12
2017-04-29 16:27:11.981 [info] <0.244.0>@lorawan_gw_router:handle_cast:52 Gateway <<0,128,0,0,0,0,163,238>> at {global,lorawan_gw_forwarder} {{127,0,0,1},40311,1}
2017-04-29 16:27:16.738 [debug] <0.255.0>@lorawan_mac_commands:auto_adr0:165 DataRate <<86,29,43,142>>: average snr 9 2.5 = 6, dr 3 -> step 2
2017-04-29 16:27:16.738 [debug] <0.255.0>@lorawan_mac_commands:auto_adr0:175 Power <<86,29,43,142>>: average rssi -48, power 5 -> up by 12
2017-04-29 16:27:21.743 [debug] <0.256.0>@lorawan_mac_commands:auto_adr0:165 DataRate <<86,29,43,142>>: average snr 9 2.5 = 7, dr 3 -> step 2
2017-04-29 16:27:21.743 [debug] <0.256.0>@lorawan_mac_commands:auto_adr0:175 Power <<86,29,43,142>>: average rssi -48, power 5 -> up by 12
2017-04-29 16:27:26.755 [debug] <0.257.0>@lorawan_mac_commands:auto_adr0:165 DataRate <<86,29,43,142>>: average snr 9 2.5 = 6, dr 3 -> step 2
2017-04-29 16:27:26.755 [debug] <0.257.0>@lorawan_mac_commands:auto_adr0:175 Power <<86,29,43,142>>: average rssi -48, power 5 -> up by 13
2017-04-29 16:27:31.659 [debug] <0.258.0>@lorawan_mac_commands:auto_adr0:165 DataRate <<86,29,43,142>>: average snr 9 2.5 = 6, dr 3 -> step 2
2017-04-29 16:27:31.659 [debug] <0.258.0>@lorawan_mac_commands:auto_adr0:175 Power <<86,29,43,142>>: average rssi -48, power 5 -> up by 13
2017-04-29 16:27:36.682 [debug] <0.259.0>@lorawan_mac_commands:auto_adr0:165 DataRate <<86,29,43,142>>: average snr 9 2.5 = 6, dr 3 -> step 2
2017-04-29 16:27:36.682 [debug] <0.259.0>@lorawan_mac_commands:auto_adr0:175 Power <<86,29,43,142>>: average rssi -48, power 5 -> up by 13
2017-04-29 16:27:36.683 [debug] <0.259.0>@lorawan_mac_commands:build_fopts:32 <<86,29,43,142>> <- [dev_status_req]
2017-04-29 16:27:41.694 [debug] <0.260.0>@lorawan_mac_commands:handle:17 <<86,29,43,142>> -> [{dev_status_ans,255,7}]
2017-04-29 16:27:41.694 [debug] <0.260.0>@lorawan_mac_commands:handle_status:127 DevStatus: battery 255, margin: -25
2017-04-29 16:27:10.410 [debug] <0.118.0> Supervisor ranch_sup started ranch_listener_sup:start_link(http, 10, ranch_tcp, [{connection_type,supervisor},{port,8080}], cowboy_clear, #{connection_type => supervisor,env => #{dispatch => [{'_',[],[{[<<"applications">>],[],lorawan_admin_applications,...},...]}]}}) at pid <0.230.0>
2017-04-29 16:27:10.416 [debug] <0.243.0> Supervisor lorawan_sup started lorawan_gw_router:start_link() at pid <0.244.0>
2017-04-29 16:27:10.436 [debug] <0.243.0> Supervisor lorawan_sup started lorawan_gw_forwarder:start_link([{port,1680}]) at pid <0.245.0>
2017-04-29 16:27:10.440 [debug] <0.243.0> Supervisor lorawan_sup started lorawan_connector_sup:start_link() at pid <0.246.0>
2017-04-29 16:27:10.444 [info] <0.247.0>@lorawan_connector_factory:init_connector:64 Connecting MQTT to mqtt://192.168.2.24:1883
2017-04-29 16:27:10.517 [debug] <0.250.0> Supervisor inet_gethost_native_sup started undefined at pid <0.251.0>
2017-04-29 16:27:10.518 [debug] <0.56.0> Supervisor kernel_safe_sup started inet_gethost_native:start_link() at pid <0.250.0>
2017-04-29 16:27:10.530 [debug] <0.246.0> Supervisor lorawan_connector_sup started lorawan_connector_mqtt:start_link({mqtt,[],"192.168.2.24",1883,"/",[]}, {connector,<<"MQTT">>,true,<<"mqtt://192.168.2.24:1883">>,<<"out/{devaddr}">>,<<"in/#">>,<<"in/{de...">>,...}) at pid <0.248.0>
2017-04-29 16:27:10.530 [debug] <0.243.0> Supervisor lorawan_sup started lorawan_connector_factory:start_link() at pid <0.247.0>
2017-04-29 16:27:10.530 [info] <0.31.0> Application lorawan_server started on node lorawan@raspberrypi
[... cut ...]
2017-04-29 16:27:36.682 [debug] <0.259.0>@lorawan_mac_commands:auto_adr0:165 DataRate <<86,29,43,142>>: average snr 9 2.5 = 6, dr 3 -> step 2
2017-04-29 16:27:36.682 [debug] <0.259.0>@lorawan_mac_commands:auto_adr0:175 Power <<86,29,43,142>>: average rssi -48, power 5 -> up by 13
2017-04-29 16:27:36.683 [debug] <0.259.0>@lorawan_mac_commands:build_fopts:32 <<86,29,43,142>> <- [dev_status_req]
2017-04-29 16:27:41.694 [debug] <0.260.0>@lorawan_mac_commands:handle:17 <<86,29,43,142>> -> [{dev_status_ans,255,7}]
2017-04-29 16:27:41.694 [debug] <0.260.0>@lorawan_mac_commands:handle_status:127 DevStatus: battery 255, margin: -25

gotthardp, 2017-04-29:
I suppose you have a device sending something ;-)

Did you set the Handler and entered its Name as a group to device/node config?

felixgalindo, 2017-04-29:
No I missed the handler step. Thanks for your help!

gotthardp, 2017-04-29:
Cool. I'm glad it works for you.

closed 2017-04-29 by gotthardp