gotthardp/lorawan-server #71: fcnt_gap_too_large :(

kazi93, 2017-04-29:
@gotthardp With my little knowledge I have cleared the "FCnt Up /	FCnt Down" field in the Nodes page once I received the ERROR message. Now this mess. :(

21:15:43.816 [error] ERROR: {fcnt_gap_too_large,<<38,1,27,100>>,15}
21:16:27.647 [error] ERROR: {fcnt_gap_too_large,<<38,1,23,120>>,21}
21:16:32.521 [error] ERROR: {fcnt_gap_too_large,<<38,1,27,100>>,18}
21:17:14.608 [error] ERROR: {fcnt_gap_too_large,<<38,1,23,120>>,24}
21:17:21.717 [error] ERROR: {fcnt_gap_too_large,<<38,1,27,100>>,21}
21:18:04.374 [error] ERROR: {fcnt_gap_too_large,<<38,1,23,120>>,27}
21:18:08.797 [error] ERROR: {fcnt_gap_too_large,<<38,1,27,100>>,24}
21:18:53.014 [error] Error in process <0.481.0> on node lorawan@mac with exit value:
{badarith,[{lorawan_mac,fcnt_gap16,2,[{file,"/Users/kazi/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac.erl"},{line,249}]},{lorawan_mac,check_link_fcnt,2,[{file,"/Users/kazi/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac.erl"},{line,238}]},{lorawan_mac,process_frame1,4,[{file,"/Users/kazi/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac.erl"},{line,104}]},{lorawan_gw_router,process_frame,3,[{file,"/Users/kazi/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_gw_router.erl"},{line,141}]}]}
21:18:53.014 [error] Hanler terminated: {badarith,[{lorawan_mac,fcnt_gap16,2,[{file,[47,85,115,101,114,115,47,107,97,122,105,47,108,111,114,97,119,97,110,45,115,101,114,118,101,114,47,95,98,117,105,108,100,47,100,101,102,97,117,108,116,47,108,105,98,47,108,111,114,97,119,97,110,95,115,101,114,118,101,114,47,115,114,99,47,108,111,114,97,119,97,110,95,109,97,99,46,101,114,108]},{line,249}]},{lorawan_mac,check_link_fcnt,2,[{file,[47,85,115,101,114,115,47,107,97,122,105,47,108,111,114,97,119,97,110,45,115,101,114,118,101,114,47,95,98,117,105,108,100,47,100,101,102,97,117,108,116,47,108,105,98,47,108,111,114,97,119,97,110,95,115,101,114,118,101,114,47,115,114,99,47,108,111,114,97,119,97,110,95,109,97,99,46,101,114,108]},{line,238}]},{lorawan_mac,process_frame1,4,[{file,[47,85,115,101,114,115,47,107,97,122,105,47,108,111,114,97,119,97,110,45,115,101,114,118,101,114,47,95,98,117,105,108,100,47,100,101,102,97,117,108,116,47,108,105,98,47,108,111,114,97,119,97,110,95,115,101,114,118,101,114,47,115,114,99,47,108,111,114,97,119,97,110,95,109,97,99,46,101,114,108]},{line,104}]},{lorawan_gw_router,process_frame,3,[{file,[47,85,115,101,114,115,47,107,97,122,105,47,108,111,114,97,119,97,110,45,115,101,114,118,101,114,47,95,98,117,105,108,100,47,100,101,102,97,117,108,116,47,108,105,98,47,108,111,114,97,119,97,110,95,115,101,114,118,101,114,47,115,114,99,47,108,111,114,97,119,97,110,95,103,119,95,114,111,117,116,101,114,46,101,114,108]},{line,141}]}]}
21:18:56.189 [error] Error in process <0.482.0> on node lorawan@mac with exit value:
{badarith,[{lorawan_mac,fcnt_gap16,2,[{file,"/Users/kazi/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac.erl"},{line,249}]},{lorawan_mac,check_link_fcnt,2,[{file,"/Users/kazi/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac.erl"},{line,238}]},{lorawan_mac,process_frame1,4,[{file,"/Users/kazi/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac.erl"},{line,104}]},{lorawan_gw_router,process_frame,3,[{file,"/Users/kazi/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_gw_router.erl"},{line,141}]}]}

gotthardp, 2017-04-29:
You're a lord Destructor :) Just put some number back.

You also may want to sent "FCnt Check" to "Reset on zero". This will probably solve the problem you had.

kazi93, 2017-04-29:
@gotthardp Haha... I am damn stupid! I did it already ( "Reset on zero") and watching for last twenty minutes. I am receiving data without problem. However not satisfied with the below :(

22:02:53.674 [debug] DataRate <<38,1,27,100>>: average snr 9 2.5 = 6, dr 5 -> step 2
22:02:53.674 [debug] Power <<38,1,27,100>>: average rssi -43, power 1 -> up by 14
22:02:53.674 [debug] LinkADRReq {5,6,[{0,0}]}
22:02:53.674 [debug] <<38,1,27,100>> <- [{link_adr_req,6,5,1,0,0},dev_status_req]
22:02:53.925 [info] Downlink request ignored. Gateway <<184,39,235,255,255,94,10,174>> not connected.
22:03:07.237 [debug] DataRate <<38,1,23,120>>: average snr 9 2.5 = 6, dr 5 -> step 2
22:03:07.237 [debug] <<38,1,23,120>> <- [dev_status_req]
22:03:07.489 [error] Error in process <0.566.0> on node lorawan@mac with exit value:
{{badmatch,{aborted,{badarith,[{lorawan_mac,'-encode_unicast/5-fun-0-',1,[{file,"/Users/kazi/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac.erl"},{line,414}]},{mnesia_tm,apply_fun,3,[{file,"mnesia_tm.erl"},{line,836}]},{mnesia_tm,execute_transaction,5,[{file,"mnesia_tm.erl"},{line,811}]},{lorawan_mac,encode_unicast,5,[{file,"/Users/kazi/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac.erl"},{line,411}]},{lorawan_mac,send_unicast,5,[{file,"/Users/kazi/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac.erl"},{line,395}]},{lorawan_gw_router,process_frame,3,[{file,"/Users/kazi/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_gw_router.erl"},{line,141}]}]}}},[{lorawan_mac,encode_unicast,5,[{file,"/Users/kazi/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac.erl"},{line,411}]},{lorawan_mac,send_unicast,5,[{file,"/Users/kazi/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac.erl"},{line,395}]},{lorawan_gw_router,process_frame,3,[{file,"/Users/kazi/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_gw_router.erl"},{line,141}]}]}
22:03:07.489 [error] Hanler terminated: {{badmatch,{aborted,{badarith,[{lorawan_mac,'-encode_unicast/5-fun-0-',1,[{file,[47,85,115,101,114,115,47,107,97,122,105,47,108,111,114,97,119,97,110,45,115,101,114,118,101,114,47,95,98,117,105,108,100,47,100,101,102,97,117,108,116,47,108,105,98,47,108,111,114,97,119,97,110,95,115,101,114,118,101,114,47,115,114,99,47,108,111,114,97,119,97,110,95,109,97,99,46,101,114,108]},{line,414}]},{mnesia_tm,apply_fun,3,[{file,[109,110,101,115,105,97,95,116,109,46,101,114,108]},{line,836}]},{mnesia_tm,execute_transaction,5,[{file,[109,110,101,115,105,97,95,116,109,46,101,114,108]},{line,811}]},{lorawan_mac,encode_unicast,5,[{file,[47,85,115,101,114,115,47,107,97,122,105,47,108,111,114,97,119,97,110,45,115,101,114,118,101,114,47,95,98,117,105,108,100,47,100,101,102,97,117,108,116,47,108,105,98,47,108,111,114,97,119,97,110,95,115,101,114,118,101,114,47,115,114,99,47,108,111,114,97,119,97,110,95,109,97,99,46,101,114,108]},{line,411}]},{lorawan_mac,send_unicast,5,[{file,[47,85,115,101,114,115,47,107,97,122,105,47,108,111,114,97,119,97,110,45,115,101,114,118,101,114,47,95,98,117,105,108,100,47,100,101,102,97,117,108,116,47,108,105,98,47,108,111,114,97,119,97,110,95,115,101,114,118,101,114,47,115,114,99,47,108,111,114,97,119,97,110,95,109,97,99,46,101,114,108]},{line,395}]},{lorawan_gw_router,process_frame,3,[{file,[47,85,115,101,114,115,47,107,97,122,105,47,108,111,114,97,119,97,110,45,115,101,114,118,101,114,47,95,98,117,105,108,100,47,100,101,102,97,117,108,116,47,108,105,98,47,108,111,114,97,119,97,110,95,115,101,114,118,101,114,47,115,114,99,47,108,111,114,97,119,97,110,95,103,119,95,114,111,117,116,101,114,46,101,114,108]},{line,141}]}]}}},[{lorawan_mac,encode_unicast,5,[{file,[47,85,115,101,114,115,47,107,97,122,105,47,108,111,114,97,119,97,110,45,115,101,114,118,101,114,47,95,98,117,105,108,100,47,100,101,102,97,117,108,116,47,108,105,98,47,108,111,114,97,119,97,110,95,115,101,114,118,101,114,47,115,114,99,47,108,111,114,97,119,97,110,95,109,97,99,46,101,114,108]},{line,411}]},{lorawan_mac,send_unicast,5,[{file,[47,85,115,101,114,115,47,107,97,122,105,47,108,111,114,97,119,97,110,45,115,101,114,118,101,114,47,95,98,117,105,108,100,47,100,101,102,97,117,108,116,47,108,105,98,47,108,111,114,97,119,97,110,95,115,101,114,118,101,114,47,115,114,99,47,108,111,114,97,119,97,110,95,109,97,99,46,101,114,108]},{line,395}]},{lorawan_gw_router,process_frame,3,[{file,[47,85,115,101,114,115,47,107,97,122,105,47,108,111,114,97,119,97,110,45,115,101,114,118,101,114,47,95,98,117,105,108,100,47,100,101,102,97,117,108,116,47,108,105,98,47,108,111,114,97,119,97,110,95,115,101,114,118,101,114,47,115,114,99,47,108,111,114,97,119,97,110,95,103,119,95,114,111,117,116,101,114,46,101,114,108]},{line,141}]}]}
22:03:41.675 [debug] DataRate <<38,1,27,100>>: average snr 9 2.5 = 6, dr 5 -> step 2
22:03:41.675 [debug] Power <<38,1,27,100>>: average rssi -44, power 1 -> up by 14
22:03:41.675 [debug] LinkADRReq {5,6,[{0,0}]}
22:03:41.675 [debug] <<38,1,27,100>> <- [{link_adr_req,6,5,1,0,0},dev_status_req]
22:03:41.926 [info] Downlink request ignored. Gateway <<184,39,235,255,255,94,10,174>> not connected.
22:03:54.715 [debug] DataRate <<38,1,23,120>>: average snr 9 2.5 = 6, dr 5 -> step 2
22:03:54.715 [debug] <<38,1,23,120>> <- [dev_status_req]
22:03:54.967 [error] Error in process <0.568.0> on node lorawan@mac with exit value:
[... cut ...] 4,99,47,108,111,114,97,119,97,110,95,103,119,95,114,111,117,116,101,114,46,101,114,108]},{line,141}]}]}
22:04:29.616 [debug] DataRate <<38,1,27,100>>: average snr 9 2.5 = 7, dr 5 -> step 2
22:04:29.616 [debug] Power <<38,1,27,100>>: average rssi -44, power 1 -> up by 14
22:04:29.616 [debug] LinkADRReq {5,6,[{0,0}]}
22:04:29.616 [debug] <<38,1,27,100>> <- [{link_adr_req,6,5,1,0,0},dev_status_req]
22:04:29.867 [info] Downlink request ignored. Gateway <<184,39,235,255,255,94,10,174>> not connected.

kazi93, 2017-04-29:
@gotthardp Current two Nodes are sending Hi/Hello payload to the single_ch_pkt_forwarder.
[image]

gotthardp, 2017-04-29:
It it possible that you put a number to *FCnt Up*, but left *FCNt Down* empty?

I am also worried about the "Downlink request ignored." notice. What kind of gateway software you use?

kazi93, 2017-04-29:
@gotthardp  Do not ask :) . A poor student for the moment. It is a Single Channel LoRa gateway with a Raspberry Pi and a Dragino hat :) and firmware : https://github.com/tftelkamp/single_chan_pkt_fwd
However, soon I will have a multitech conduit (full gateway) for testing purpose. 

Note: Nothing to be worried as single-chan-pkt-fwd does not support downlink packet :)

gotthardp, 2017-04-29:
Problem solved? May I close the issue?

kazi93, 2017-04-29:
@gotthardp Not solved! It have restarted the server and the gateway, nothing works! 
23:04:07.284 [error] Error in process <0.1186.0> on node lorawan@mac with exit value:
{{badmatch,{aborted,{badarith,[{lorawan_mac,'-encode_unicast/5-fun-0-',1,[{file,"/Users/kazi/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac.erl"},{line,414}]},{mnesia_tm,apply_fun,3,[{file,"mnesia_tm.erl"},{line,836}]},{mnesia_tm,execute_transaction,5,[{file,"mnesia_tm.erl"},{line,811}]},{lorawan_mac,encode_unicast,5,[{file,"/Users/kazi/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac.erl"},{line,411}]},{lorawan_mac,send_unicast,5,[{file,"/Users/kazi/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac.erl"},{line,395}]},{lorawan_gw_router,process_frame,3,[{file,"/Users/kazi/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_gw_router.erl"},{line,141}]}]}}},[{lorawan_mac,encode_unicast,5,[{file,"/Users/kazi/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac.erl"},{line,411}]},{lorawan_mac,send_unicast,5,[{file,"/Users/kazi/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac.erl"},{line,395}]},{lorawan_gw_router,process_frame,3,[{file,"/Users/kazi/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_gw_router.erl"},{line,141}]}]}
23:04:07.284 [error] Hanler terminated: {{badmatch,{aborted,{badarith,[{lorawan_mac,'-encode_unicast/5-fun-0-',1,[{file,[47,85,115,101,114,115,47,107,97,122,105,47,108,111,114,97,119,97,110,45,115,101,114,118,101,114,47,95,98,117,105,108,100,47,100,101,102,97,117,108,116,47,108,105,98,47,108,111,114,97,119,97,110,95,115,101,114,118,101,114,47,115,114,99,47,108,111,114,97,119,97,110,95,109,97,99,46,101,114,108]},{line,414}]},{mnesia_tm,apply_fun,3,[{file,[109,110,101,115,105,97,95,116,109,46,101,114,108]},{line,836}]},{mnesia_tm,execute_transaction,5,[{file,[109,110,101,115,105,97,95,116,109,46,101,114,108]},{line,811}]},{lorawan_mac,encode_unicast,5,[{file,[47,85,115,101,114,115,47,107,97,122,105,47,108,111,114,97,119,97,110,45,115,101,114,118,101,114,47,95,98,117,105,108,100,47,100,101,102,97,117,108,116,47,108,105,98,47,108,111,114,97,119,97,110,95,115,101,114,118,101,114,47,115,114,99,47,108,111,114,97,119,97,110,95,109,97,99,46,101,114,108]},{line,411}]},{lorawan_mac,send_unicast,5,[{file,[47,85,115,101,114,115,47,107,97,122,105,47,108,111,114,97,119,97,110,45,115,101,114,118,101,114,47,95,98,117,105,108,100,47,100,101,102,97,117,108,116,47,108,105,98,47,108,111,114,97,119,97,110,95,115,101,114,118,101,114,47,115,114,99,47,108,111,114,97,119,97,110,95,109,97,99,46,101,114,108]},{line,395}]},{lorawan_gw_router,process_frame,3,[{file,[47,85,115,101,114,115,47,107,97,122,105,47,108,111,114,97,119,97,110,45,115,101,114,118,101,114,47,95,98,117,105,108,100,47,100,101,102,97,117,108,116,47,108,105,98,47,108,111,114,97,119,97,110,95,115,101,114,118,101,114,47,115,114,99,47,108,111,114,97,119,97,110,95,103,119,95,114,111,117,116,101,114,46,101,114,108]},{line,141}]}]}}},[{lorawan_mac,encode_unicast,5,[{file,[47,85,115,101,114,115,47,107,97,122,105,47,108,111,114,97,119,97,110,45,115,101,114,118,101,114,47,95,98,117,105,108,100,47,100,101,102,97,117,108,116,47,108,105,98,47,108,111,114,97,119,97,110,95,115,101,114,118,101,114,47,115,114,99,47,108,111,114,97,119,97,110,95,109,97,99,46,101,114,108]},{line,411}]},{lorawan_mac,send_unicast,5,[{file,[47,85,115,101,114,115,47,107,97,122,105,47,108,111,114,97,119,97,110,45,115,101,114,118,101,114,47,95,98,117,105,108,100,47,100,101,102,97,117,108,116,47,108,105,98,47,108,111,114,97,119,97,110,95,115,101,114,118,101,114,47,115,114,99,47,108,111,114,97,119,97,110,95,109,97,99,46,101,114,108]},{line,395}]},{lorawan_gw_router,process_frame,3,[{file,[47,85,115,101,114,115,47,107,97,122,105,47,108,111,114,97,119,97,110,45,115,101,114,118,101,114,47,95,98,117,105,108,100,47,100,101,102,97,117,108,116,47,108,105,98,47,108,111,114,97,119,97,110,95,115,101,114,118,101,114,47,115,114,99,47,108,111,114,97,119,97,110,95,103,119,95,114,111,117,116,101,114,46,101,114,108]},{line,141}]}]}
23:04:56.011 [debug] DataRate <<38,1,23,120>>: average snr 9 2.5 = 6, dr 5 -> step 2
23:04:56.011 [debug] <<38,1,23,120>> <- [dev_status_req]
23:04:56.263 [error] Error in process <0.1188.0> on node lorawan@mac with exit value:
{{badmatch,{aborted,{badarith,[{lorawan_mac,'-encode_unicast/5-fun-0-',1,[{file,"/Users/kazi/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac.erl"},{line,414}]},{mnesia_tm,apply_fun,3,[{file,"mnesia_tm.erl"},{line,836}]},{mnesia_tm,execute_transaction,5,[{file,"mnesia_tm.erl"},{line,811}]},{lorawan_mac,encode_unicast,5,[{file,"/Users/kazi/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac.erl"},{line,411}]},{lorawan_mac,send_unicast,5,[{file,"/Users/kazi/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac.erl"},{line,395}]},{lorawan_gw_router,process_frame,3,[{file,"/Users/kazi/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_gw_router.erl"},{line,141}]}]}}},[{lorawan_mac,encode_unicast,5,[{file,"/Users/kazi/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac.erl"},{line,411}]},{lorawan_mac,send_unicast,5,[{file,"/Users/kazi/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac.erl"},{line,395}]},{lorawan_gw_router,process_frame,3,[{file,"/Users/kazi/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_gw_router.erl"},{line,141}]}]}
[... cut ...] 119,97,110,95,115,101,114,118,101,114,47,115,114,99,47,108,111,114,97,119,97,110,95,109,97,99,46,101,114,108]},{line,395}]},{lorawan_gw_router,process_frame,3,[{file,[47,85,115,101,114,115,47,107,97,122,105,47,108,111,114,97,119,97,110,45,115,101,114,118,101,114,47,95,98,117,105,108,100,47,100,101,102,97,117,108,116,47,108,105,98,47,108,111,114,97,119,97,110,95,115,101,114,118,101,114,47,115,114,99,47,108,111,114,97,119,97,110,95,103,119,95,114,111,117,116,101,114,46,101,114,108]},{line,141}]}]}

gotthardp, 2017-04-29:
The same error as before will likely have the same root cause.

kazi93, 2017-04-30:
@gotthardp Many thanks for supporting with patience. Deleting the Nodes and adding them again solved the problem in my opinion. The server is receiving frames since yesterday without interruption. Currently two LoraWAN nodes sending with a 30 second interval. I will try to do some more test with FCnt or FCnt Check features. However, you may close the issue.

gotthardp, 2017-04-30:
Cool. Thanks for letting me know.

BTW, I updated the latest branch, so it doesn't allow people to clear the fields. This may avoid this issue in the future.

closed 2017-04-30 by gotthardp