gotthardp/lorawan-server #72: Gateway Config File for Multitech Conduit

kammyy, 2017-04-30:
Please share the gateway config file for multitech gateway.
The file that i am using is shown below:

`{
    "SX1301_conf": {
        "lorawan_public": true,
        "clksrc": 1,
        "lbt_cfg": {
            "enable": false,
            "rssi_target": -80,
            "chan_cfg":[
                { "freq_hz": 865062500, "scan_time_us": 128 },
                { "freq_hz": 865402500, "scan_time_us": 5000 },
                { "freq_hz": 865985000, "scan_time_us": 128 },
                { "freq_hz": 866550000, "scan_time_us": 128 }
            ],
            "sx127x_rssi_offset": -4
        },
        "antenna_gain": 0,
        "radio_0": {
            "enable": true,
            "type": "SX1257",
            "freq": 865985000,
            "rssi_offset": -166.0,
            "tx_enable": true,
            "tx_notch_freq": 129000,
            "tx_freq_min": 863000000,
            "tx_freq_max": 870000000
        },
        "radio_1": {
            "enable": true,
            "type": "SX1257",
            "freq": 866500000,
            "rssi_offset": -166.0,
            "tx_enable": false
        },
        "chan_multiSF_0": {
            "enable": true,
            "radio": 1,
            "if": -400000
        },
        "chan_multiSF_1": {
            "enable": true,
            "radio": 1,
            "if": -200000
        },
        "chan_multiSF_2": {
            "enable": true,
            "radio": 1,
            "if": 0
        },
        "chan_multiSF_3": {
            "enable": true,
            "radio": 0,
            "if": -400000
        },
        "chan_multiSF_4": {
            "enable": true,
            "radio": 0,
            "if": -200000
        },
        "chan_multiSF_5": {
            "enable": true,
            "radio": 0,
            "if": 0
        },
        "chan_multiSF_6": {
            "enable": true,
            "radio": 0,
            "if": 200000
        },
        "chan_multiSF_7": {
            "enable": true,
            "radio": 0,
            "if": 400000
        },
        "chan_Lora_std": {
            "enable": true,
            "radio": 1,
            "if": -200000,
            "bandwidth": 250000,
            "spread_factor": 7
        },
        "chan_FSK": {
            "enable": true,
            "radio": 1,
            "if": 300000,
            "bandwidth": 125000,
            "datarate": 50000
        }
    },

    "gateway_conf": {
        "gateway_ID": "000800FFFF4A166E",
        "server_address": "localhost",
        "serv_port_up": 1680,
        "serv_port_down": 1680,
        "keepalive_interval": 10,
        "stat_interval": 30,
        "push_timeout_ms": 100,
        "forward_crc_valid": true,
        "forward_crc_error": false,
        "forward_crc_disabled": false
    }
}`

With the above config file, i am getting below mentioned error in "Console.log" text file.
`ERROR: {unknown_mac,<<0,128,0,0,0,0,196,227>>}
2017-04-30 23:25:32.138 [warning] <0.3468.0>@lorawan_mac:process_frame:28 Unknown MAC 008000000000C4E3
2017-04-30 23:25:32.138 [error] <0.3468.0>@lorawan_handler:rxpk:21 ERROR: {unknown_mac,<<0,128,0,0,0,0,196,227>>}
2017-04-30 23:25:36.962 [warning] <0.3469.0>@lorawan_mac:process_frame:28 Unknown MAC 008000000000C4E3
2017-04-30 23:25:36.962 [error] <0.3469.0>@lorawan_handler:rxpk:21`

Please share a working config file for Multitech LoRaWAN Conduit Gateway.

Thanks

gotthardp, 2017-04-30:
The error is caused by server misconfiguration.
See https://github.com/gotthardp/lorawan-server/blob/master/doc/Troubleshooting.md#unknown_mac

kammyy, 2017-05-01:
@gotthardp : Thanks. Got it working.

closed 2017-05-01 by gotthardp