gotthardp/lorawan-server #74: Error while receiving uplink

kammyy, 2017-05-01:
I am receiving below mentioned error when my device sends uplink messages.
`2017-05-01 17:26:51.167 [error] emulator Error in process <0.2966.0> on node 'lorawan@LAPTOP-6BPQHC3R' with exit value:
{{badmatch,<<69,12,0,112,32,12,0>>},[{lorawan_mac,process_frame1,5,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac.erl"},{line,75}]},{lorawan_handler,rxpk,4,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_handler.erl"},{line,16}]}]}`

Does anyone have any idea as to  why this error is occurring ?
I am using Multi-tech gateway and IMST end module.

Thanks

gotthardp, 2017-05-01:
Which server version are you using?

kammyy, 2017-05-01:
v0.3.0.

gotthardp, 2017-05-01:
The 0.3.0 cannot accept empty messages. This is a bug, which has been fixed already. You can either:
 * convince your device to send some port number
 * upgrade to lorawan-server 0.4.0

gotthardp, 2017-05-10:
The 0.4.0 was just release officially. Please let me know if it solved your issue.

closed 2017-05-14 by gotthardp