gotthardp/lorawan-server #75: MQTT guide

Sviests, 2017-05-02:
Hi!
I am currently trying to set up so that the server publishes received messages to MQTT and I am having difficulties getting it working.
What I have done so far:
*Set the application for my node as backend
*Created a Connector with the names of the topics I would like the server to publish on and subscribe to
*Created a Handler for the previously mentioned Connector with the format set to Raw Data

Unfortunately, when subscribing to all topics on my mqtt server ("\#"), I am not receiving anything...

Any help would be appreciated. Thanks!

gotthardp, 2017-05-02:
Hi. Did you set the "Group" of your node to the Handler name (also called group)? You need to link the handler with the device/node.

Sviests, 2017-05-02:
I will try this once I get the server running again, now I am unable to start the server back up after creating a backend, reported in Issue #76

Sviests, 2017-05-02:
Got MQTT working by adding the group, thanks!

gotthardp, 2017-05-02:
I am glad it works.

closed 2017-05-02 by gotthardp