gotthardp/lorawan-server #76: Server startup crash when a connector is added

Sviests, 2017-05-02:
I`ve been trying to set up a Backend so that the received messages would be published to a MQTT topic. After setting everything up, I decided to restart the server, but, unfortunately I was not able to start the server back up. This is the error dump I got:
`{"init terminating in do_boot",{{badmatch,{error,{lorawan_server,{{shutdown,{failed_to_start_child,connector_factory,{{badmatch,{error,no_scheme}},[{lorawan_connector_sup,start_child,1,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_connector_sup.erl"},{line,20}]},{lorawan_connector_factory,init_connector,1,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_connector_factory.erl"},{line,107}]},{lists,foreach,2,[{file,"lists.erl"},{line,1338}]},{lorawan_connector_factory,init,1,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_connector_factory.erl"},{line,34}]},{gen_server,init_it,6,[{file,"gen_server.erl"},{line,328}]},{proc_lib,init_p_do_apply,3,[{file,"proc_lib.erl"},{line,247}]}]}}},{lorawan_app,start,[normal,[]]}}}}},[{lorawan_app,start,0,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_app.erl"},{line,16}]},{init,start_em,1,[]},{init,do_boot,3,[]}]}}
init terminating in do_boot ()

Crash dump is being written to: erl_crash.dump...done

[erl_crash.txt](https://github.com/gotthardp/lorawan-server/files/970017/erl_crash.txt)
`

gotthardp, 2017-05-02:
I confirm this is a bug, but it was caused by a misconfiguration-- you entered URL without the `mqtt://` scheme.

gotthardp, 2017-05-02:
Should be fixed now. Try to `git pull` the latest master and let me know if it helped.

I am sorry for this. I didn't test the error cases well enough.

Sviests, 2017-05-02:
Thank you very much for the fix, it works!

closed 2017-05-02 by gotthardp