gotthardp/lorawan-server #77: Db removes the frames

Kralg, 2017-05-02:
Hello, 

I have noticed that the data gets constantly removed from the server or at least I can't see it in admin window. Would it be possible to get some help on how to save all data and see all the received packages every time?

gotthardp, 2017-05-02:
What admin page are you looking at?

The server keeps last 50 frame for each Node.

Kralg, 2017-05-02:
I mean localhost:8080 and Received frames window. I really don't see 50 packages. Today the most I had was 32 packets, but now only 3 are present without manually deleting. Is there anyway to see what is happening in database, content or commands that were sent?

gotthardp, 2017-05-02:
If you got version 0.3.0 (the older), the received frames get deleted upon device reset. Could this be the case? This is no longer tue for 0.4.0, so you may consider upgrading.

Kralg, 2017-05-02:
Yes, that might be the reason. I will try with 0.4.0.

gotthardp, 2017-05-02:
Perfect. Please let me know if the problem persists.

Kralg, 2017-05-03:
Once I have installed 0.4.0 I can not OTAA. I am attaching the server log. If I try to delete extracted files from /usr/lib/lorawan-server and unzip the 0.3.0 I can see that (Infrastructure and Backends) are still there but now I can OTAA to the serveragain to the server. 

[server_log.txt](https://github.com/gotthardp/lorawan-server/files/973160/server_log.txt)

gotthardp, 2017-05-03:
This is a gateway log. Could you please send me the server log. Also, please make sure you use 0.4.0-pre.1, there was an OTAA bug in pre.0, but I am afraid you have the right version.

Kralg, 2017-05-04:
Hello, I think there was a problem that my window somehow did not update and I could still see devices and gateway but actually they were deleted as new lorawan-server version has been installed. My mistake! 
Is there a way to change the amount of saved packages from a device?

gotthardp, 2017-05-04:
So OTAA works for you now?

The amount of packets stored is still 50, but they don't get removed upon reset or join.

Kralg, 2017-05-04:
Finally I think I got an idea on what is happening. Yes, OTAA seems to be working but only if I start lorawan-server manually.

If I run it as a daemon and systemctl status lorawan-server. I see one red message pam_systemd(su:session): Failed to create session: Message did not received a reply (timeout by message bus)

Another funny thing (with daemon version) is that when I try to access this hostname:8080 in our local network on same RPI I can see new webpage design, on another computer over the wi-fi new design, but on one more compute it is still old version of webpage. It really got refreshed and logged in with admin admin. RPI and computers are on the same Wi-Fi. But when I run server manually webpage got updated on all the computers. Maybe this got something to do with the error?

Kralg, 2017-05-05:
Another question, related to db. Can you advice a way on how can I delete frames from the database?

gotthardp, 2017-05-05:
It seems like you have two installations of the server. For sure the database for the daemon-version may be stored somewhere else than the work-directory version.

And, no, there is no way to delete the received frames. They are used internally for graphs and statistics. Users are not supposed to interfere with these functions. Reeceived frames are read only.

Kralg, 2017-05-07:
I manually run the server from /serv/lib/lorawan-server/bin ./lorawan-server. Is this is not the same place where the daemon would run from? If not and there are some additional files not in /serv/lib/lorawan-server folder can you advice me on how I could remove all the unzipped versions to install only one new version?

gotthardp, 2017-05-07:
And how did you install and how do you run the daemon version?

The original daemon script looks like
```bash
export LORAWAN_HOME=/var/lib/lorawan-server
su lorawan -s /bin/sh -c "/usr/lib/lorawan-server/bin/lorawan-server" &
```
Basically, if LORAWAN_HOME is set, the database is stored there. If LORAWAN_HOME is not set, the database is stored to where the binaries are.

gotthardp, 2017-05-08:
I improved the documentation in https://github.com/gotthardp/lorawan-server/blob/master/doc/Installation.md to clarify this.

Kralg, 2017-05-09:
Thanks, i just have faced the problem that login credentials doesn't update once I change them in sys.config. But as you wrote in documentation once I remove Amnesia folder it seems to update. (it would be nice to update automatically, but probably documentation is enough as this is not changed many times) I will try upgrade procedure and come back with results.

Meanwhile, is there any way on how to get ride of the authentication? I tried to comment credentials line in sys.config, I still get login window.

gotthardp, 2017-05-09:
Hello. The credentials in sys.config are used **only** when creating a fresh database. Otherwise you can change the password from the *Users* page.

Currently there is no way to disable authentication, but it's quite easy to implement.

Kralg, 2017-05-10:
Maybe I will post here the question. Can you give an example on what JSON should I send in order to create a new device. Let's say with REST devices GET I get:
{"adr_flag_set":1,"adr_set":{"chans":null,"datr":null,"power":null},"app":"websocket","appeui":"1341234123544434","appkey":"32423544233452343443343433333432","can_join":true,"deveui":"1234232142342323","fcnt_check":0,"region":"EU863-870","rxwin_set":{"rx1_dr_offset":null,"rx2_dr":null,"rx2_freq":null}}

Do I have to send this JSON and change only EUI, KEY and etc. or should it have more changes?

gotthardp, 2017-05-10:
You can omit the optional fields, i.e. the following should work:
```json
{"app":"websocket","appeui":"1341234123544434","appkey":"32423544233452343443343433333432","deveui":"1234232142342323","fcnt_check":0,"region":"EU863-870"}
```

Kralg, 2017-05-10:
And the json should be sent to http://localhost:8080/devices endpoint? At the moment I am getting "The remote server returned an error: (405) Method Not Allowed." when I send 
{"app":"websocket","appeui":"1288491294124214","appkey":"10249112471894891123121241241244","can_join":true,"deveui":"2342343923849494","fcnt_check":0,"region":"EU863-870"}

Kralg, 2017-05-10:
Thanks, it works.

gotthardp, 2017-05-10:
Is there any issue that persists and needs solving, or I may close this issue?

Kralg, 2017-05-13:
For me the upgrade part still is not very flowless. Daemon does not get updated and running. Do you have some advices on how to update it? I actually deleted lorawan folders in /usr/lib and var/lib extracted .tar into /usr/lib/lorawan-server, copied service into /lib/systemd/system and if I start daemon with systemctl I can't access server in the browser. However if I start it from /usr/lib/lorawan-server/bin directory then it is up and running.

Kralg, 2017-05-13:
I followed the procedure 
remove user (userdel -r user)
systemctl disable lorawan-server
removed /lib/systemd/system/lorawan-server.service
removed /usr/lib/systemd/system/lorawan-server.service
removed /etc/lib/lorawan-server

And followed the install procedure once again I managed to get it working from daemon.

Kralg, 2017-05-13:
Another thought, it would be really nice to somehow be able to change saved frames amount and remove frames from DB with REST commands with filtering and from the web interface with button or so also with filtering.

Kralg, 2017-05-13:
By the way have you tried to connect from windows computer to lorawan-server on the same network with HttWebRequest? For me GET command works fine, but if I use DELETE or PUT I am getting SocketException: No connection could be made because the target machine actively refused it, but I can open the gateway webpage in the browser. If I run lorawan-server on the same windows computer I can use GET, PUT commands (have not tried DELETE yet). I found information that this can be related to firewall, but don't know yet how to allow the connection. On the raspberry side I am not using firewalls, it is default raspbian that is installed, so I believe it should not cause this problem.

gotthardp, 2017-05-13:
Re HTTP: Hard to guess. Would you be able to capture (by `sudo tcpdump -i eth0 -w file.pcap`) the HTTP traffic?

Kralg, 2017-05-13:
I am attaching the file. Scenario: twise repeated sequence: GET device, GET rxframes (actually there are no frames in server right now), PUT device/DevEUI. 

[file.pcap.txt](https://github.com/gotthardp/lorawan-server/files/998608/file.pcap.txt)

gotthardp, 2017-05-13:
But this looks OK. Do Windows report some issue there?

Also, I added some more documentation to https://github.com/gotthardp/lorawan-server/blob/master/doc/Administration.md with REST API examples.

And, the number of retained rxframes is now configurable via the `retained_rxframes` parameter. Simply add `{retained_rxframes, 500},` to your config to increase the number of frames retained to 500.

gotthardp, 2017-05-13:
Also, could you please let me know what upgrade procedure did you follow, which didn't work? What did you do to make the server fail as you describe?

Kralg, 2017-05-13:
I think I found the problem, I was using wrong localhost for only PUT. Sorry for that.
Thanks, I just have looked at that. For me that HTTP part is slightly unclear, probably due to the fact that I don't have enough knowledge for that yet. :) But table with REST commands is still nice.
However, I saw the update user with PUT. I have read somewhere that the main difference between PUT and POST is that PUT - creates, POST - updates?

Kralg, 2017-05-13:
I will try to work on whole REST part for couple of days and then switch back to upgrading. In short if I have 0.3 installation of lorawan-server (I did it according your daemon explanation) and after that I just download binary 0.4 and repeat the procedu (without deleting anything) then daemon fails for me. I thought that this might have something to do that meanwhile I install the new version and copy service, daemon is still running and maybe some files cant get overwrited.

gotthardp, 2017-05-13:
I updated the upgrade instructions in https://github.com/gotthardp/lorawan-server/blob/master/doc/Installation.md#upgrade. Next time you upgrade, please follow these and see if it works.

The POST vs. PUT is always confusing. My philosophy was to create by POST and update by PUT. This is what the ng-admin REST wrapper uses as well.

Kralg, 2017-05-13:
Thanks. I am just confused with what do you mean by update bin, lib, releases. Untar the tar in the folder and it will do the trick? Or should bin, lib, releases untared somewhere else and moved to the /usr/lib/lorawan-server?

Kralg, 2017-05-13:
Regarding retained_rxframes, should I download new binary, or the old should work if I will add retained_rxframes? 
Just to make sure, if I will place it right below the {http_admin_credentials,...}, 
{retained_rxframes, 500},
it should work?

gotthardp, 2017-05-13:
Simple untar does the trick. It shall be incredibly simple, I am not sure what failed on your PC.

New binary is needed for the `retained_rxframes`. Then you place it e.g. bellow http_admin_credentials. If you don't want to build the binary, I just buit it for you at http://swimgate.eu/lorawan-server-0.4.0.tar.gz

Kralg, 2017-05-13:
That is interesting, I think I tried without deleting the sub-directories to untar it and to delete whole /usr/lib/lorawan-server, but not sure about if I stopped the daemon. I will try and post if something will go wrong. 
Thanks a lot!

Kralg, 2017-05-14:
Hi, I just saw, that I am somehow getting JSON with appeui set to deveui and deveui set to apeui. When I create device server recognizes right appeui and deveui.

Kralg, 2017-05-15:
Hi, Just one question is it possible that in the one binary that you posted here you have changed the time when gateway answers to end device? I am trying right now and it seems that in an  old version it was ~5seconds and here it is ~7seconds or so?

gotthardp, 2017-05-15:
And it's a good thing or a bad thing? :) I slightly modified the logic around selecting the TX window for Join-Accept.

Kralg, 2017-05-15:
Can you explain what is happening now? For us wimod device is still working but our programed one does not. :)

Kralg, 2017-05-15:
Ups. Not on purpose.

Kralg, 2017-05-16:
Hi would it be possible to get a binary with old TX frame and number of rxframes choice?

gotthardp, 2017-05-20:
I don't understand the question: what binary? Could you please describe it more thoroughly?

Kralg, 2017-05-22:
Sorry, can you send a compiled version of lorawan-server without the changes that you made with TX frames but with packet number option?

Kralg, 2017-05-22:
And one more question. I am not sure but I think that I saw somewhere a person asking for REST commands to support filtering with sorting and rx number of frames? Are these commands supported in 4.1?

gotthardp, 2017-05-22:
The sorting and filtering is described here: https://github.com/gotthardp/lorawan-server/blob/master/doc/Administration.md

And the binaries of 0.4.1 are the latest and greatest binaries I have.

closed 2017-05-29 by Kralg