gotthardp/lorawan-server #78: how to convert hex data to ascii

YousseMes, 2017-05-04:
please can you explain me how to convert data to ascii
[image]

gotthardp, 2017-05-04:
For example, you can use a tool like http://www.unit-conversion.info/texttools/hexadecimal/

YousseMes, 2017-05-04:
but if i want to see data in application as ascii format ?

gotthardp, 2017-05-04:
You cannot. The data may not be 7-bit ASCII.

kazi93, 2017-05-04:
@YousseMes Hi! Have you managed to read data on a visualization tool e.g freeboard,thingspeak?

YousseMes, 2017-05-05:
No I didn't can you explain how please ?

gotthardp, 2017-05-05:
The list of received frames is not intended to display the payload. I assume that the data decoded by the server will be sent to some backend system and the visualization will be done there,

gotthardp, 2017-05-05:
I made an update for the coming 0.4.0 release, which should meet your needs: If you put your mouse cursor above the *Data* field, it will display printable ASCII characters in that string.

closed 2017-05-05 by gotthardp